openwrt/openwrt #9561: ramips/mt7621: kernel panic under networking load

Vladdrako, 2022-03-27:
Device: Xiaomi Mi Router 3 Pro
While huge networking load I get kernel panic. Possibly after commit https://github.com/openwrt/openwrt/commit/fdbb468c5998d9adb1819751ffaf38013768a8eb
<details>
<pre>
<4>[ 6377.998611] CPU: 0 PID: 700 Comm: napi/mtk_eth-6 Tainted: G           O      5.10.108 #0
<4>[ 6378.006669] $ 0   : 00000000 00000001 0000005c 0000006a
<4>[ 6378.011905] $ 4   : 0067a000 00000a20 80673dc8 00000000
<4>[ 6378.017131] $ 8   : 00000100 00000040 00000039 ffffffff
<4>[ 6378.022361] $12   : 00000681 80b18f2c 80b18f3c 80b18ebc
<4>[ 6378.027593] $16   : 81412a00 00000a20 80a7b464 00000000
<4>[ 6378.032821] $20   : 80673dc8 80a80000 00000001 003fddc0
<4>[ 6378.038050] $24   : 00002039 00000000                  
<4>[ 6378.043277] $28   : 82344000 82345d40 c5ee0000 80673dc8
<4>[ 6378.048507] Hi    : 00000000
<4>[ 6378.051375] Lo    : 00000041
<4>[ 6378.054248] epc   : 801a7f8c 0x801a7f8c
<4>[ 6378.058074] ra    : 80673dc8 0x80673dc8
<4>[ 6378.061893] Status: 11008403	KERNEL EXL IE 
<4>[ 6378.066077] Cause : 40800010 (ExcCode 04)
<4>[ 6378.070072] BadVA : 0000006a
<4>[ 6378.072942] PrId  : 0001992f (MIPS 1004Kc)
<4>[ 6378.077019] Modules linked in: ksmbd(O) nf_nat_amanda nf_conntrack_amanda wireguard snd_usb_audio rndis_host mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length2(O) xt_length xt_ipv4options(O) xt_iprange xt_iface(O) xt_hl xt_helper xt_hashlimit xt_geoip(O) xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_PROTO(O) xt_NFQUEUE xt_NFLOG xt_NETMAP xt_MASQUERADE xt_LUA(O) xt_LOGMARK(O) xt_LOG xt_IPMARK(O) xt_HL xt_DSCP xt_DNETMAP(O) xt_DHCPMAC(O) xt_CT xt_CLASSIFY xt_CHECKSUM vhci_hcd usbnet usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress
<4>[ 6378.077529]  ums_alauda ts_kmp ts_fsm ts_bm snd_usbmidi_lib sch_cake ntfs3(O) nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_inet nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_bridge nf_conncount lz4_decompress lz4_compress libchacha20poly1305 libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_rpfilter ipt_ECN ip6table_raw ip6t_rpfilter ip_tables hwmon hid_generic ebtables ebt_vlan ebt_stp ebt_redirect
<4>[ 6378.164923]  ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat_xtables(O) compat(O) br_netfilter arptable_filter arpt_mangle arp_tables fuse cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact sg hid evdev ledtrig_gpio cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set ip6table_nat ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsd bonding ip6_gre ip_gre gre ifb nat46(O) sit ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp ah4 ip6_tunnel tunnel6 tunnel4
<4>[ 6378.252380]  rpcsec_gss_krb5 auth_rpcgss oid_registry tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore ovpn_dco(O) xfrm_user xfrm_ipcomp af_key lockd sunrpc grace isofs hfsplus hfs cdrom autofs4 dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 nls_koi8_r nls_cp866 nls_cp1251 zram algif_skcipher algif_rng algif_hash algif_aead af_alg arc4 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ahci libahci libata gpio_button_hotplug(O) xfs f2fs ext4 mbcache jbd2 btrfs xor raid6_pq mii tpm rng_core crc32_generic
<4>[ 6378.390058] Process napi/mtk_eth-6 (pid: 700, threadinfo=9f22e92d, task=04795bee, tls=00000000)
<4>[ 6378.398728] Stack : 83634900 0000f800 00000001 816ee750 80c60000 80c60000 0000000e 0000081e
<4>[ 6378.407093]         a35a8d00 844fda80 844fda80 00000960 86ce2be0 844fda80 00003b4c 0000000a
<4>[ 6378.415455]         816ee080 80673dc8 83634900 000005ee 00003b4c 00000009 844fda80 00000960
<4>[ 6378.423818]         a35a8d00 80673e38 816ee7d4 00000001 06ce2be0 816ee5ec 000000d0 816ee5ec
<4>[ 6378.432182]         000000d0 805bd91c 00000000 00000001 40000000 ff7cc630 00000000 00000000
<4>[ 6378.440544]         ...
<4>[ 6378.442989] Call Trace:
<4>[ 6378.443288] [<80673dc8>] 0x80673dc8
<4>[ 6378.449260] [<80673e38>] 0x80673e38
<4>[ 6378.452813] [<805bd91c>] 0x805bd91c
<4>[ 6378.456291] [<80697488>] 0x80697488
<4>[ 6378.459907] [<805bfc84>] 0x805bfc84
<4>[ 6378.463410] [<80697230>] 0x80697230
<4>[ 6378.466984] [<80697078>] 0x80697078
<4>[ 6378.470495] [<8003a328>] 0x8003a328
<4>[ 6378.473998] [<80697230>] 0x80697230
<4>[ 6378.477480] [<8069724c>] 0x8069724c
<4>[ 6378.480989] [<80056420>] 0x80056420
<4>[ 6378.484490] [<808d5588>] 0x808d5588
<4>[ 6378.487980] [<80697198>] 0x80697198
<4>[ 6378.491546] [<80697198>] 0x80697198
<4>[ 6378.495021] [<80056998>] 0x80056998
<4>[ 6378.498504] [<80056820>] 0x80056820
<4>[ 6378.502018] [<80063e0c>] 0x80063e0c
<4>[ 6378.505498] [<80056820>] 0x80056820
<4>[ 6378.509079] [<80056820>] 0x80056820
<4>[ 6378.512563] [<8000376c>] 0x8000376c
<4>[ 6378.516055] 
<4>[ 6378.517542] Code: 02803025  8e02001c  00621821 <8c660000> 41656000  30a50001  000000c0  8f84000c  8e020000 
<4>[ 6378.527308] 
<4>[ 6378.528891] ---[ end trace 74c9bb18e30ca9e6 ]---
<4>[ 6378.571925] ------------[ cut here ]------------
<4>[ 6378.576569] WARNING: CPU: 2 PID: 18888 at lib/refcount.c:28 0x80674bf4
<4>[ 6378.583117] refcount_t: underflow; use-after-free.
<4>[ 6378.587994] Modules linked in: ksmbd(O) nf_nat_amanda nf_conntrack_amanda wireguard snd_usb_audio rndis_host mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length2(O) xt_length xt_ipv4options(O) xt_iprange xt_iface(O) xt_hl xt_helper xt_hashlimit xt_geoip(O) xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_PROTO(O) xt_NFQUEUE xt_NFLOG xt_NETMAP xt_MASQUERADE xt_LUA(O) xt_LOGMARK(O) xt_LOG xt_IPMARK(O) xt_HL xt_DSCP xt_DNETMAP(O) xt_DHCPMAC(O) xt_CT xt_CLASSIFY xt_CHECKSUM vhci_hcd usbnet usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress
<4>[ 6378.588516]  ums_alauda ts_kmp ts_fsm ts_bm snd_usbmidi_lib sch_cake ntfs3(O) nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_inet nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_bridge nf_conncount lz4_decompress lz4_compress libchacha20poly1305 libblake2s_generic iptable_raw iptable_nat [... cut ...]
<4>[ 1085.418048] [<804133b8>] 0x804133b8
<4>[ 1085.421525] [<8003a4b8>] 0x8003a4b8
<4>[ 1085.425011] [<80412c2c>] 0x80412c2c
<4>[ 1085.428501] [<80003c68>] 0x80003c68
<4>[ 1085.431977] 
<4>[ 1085.433461] ---[ end trace f6917992ee262c13 ]---
</pre>
</details>
My build is custom a bit. I applied e53bc43de4a83a205aaccb5d1954860085c6b55b b0734bd0f3794332b01a077e9bf38c1d792abc88 and https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commit;h=df9f06e8182d10e2786de2b83de99034eea92258
for testing purposes.

nbd168, 2022-03-27:
Please build with KALLSYMS enabled and show me traces that contain proper function names

Vladdrako, 2022-03-29:
> Please build with KALLSYMS enabled and show me traces that contain proper function names

With KALLSYMS. Logs collected by pstore.
<details>
<pre>
<1>[33425.731740] CPU 2 Unable to handle kernel paging request at virtual address 0400064c, epc == 8072b128, ra == 807234ac
<4>[33425.742402] Oops[#1]:
<4>[33425.744672] CPU: 2 PID: 24661 Comm: kworker/2:2 Tainted: G           O      5.10.108 #0
<4>[33425.752647] Workqueue: events_power_efficient 0x80723988
<4>[33425.757938] $ 0   : 00000000 00000001 040005fc 8358a080
<4>[33425.763151] $ 4   : 8a368700 00000000 80b10860 00000000
<4>[33425.768363] $ 8   : 00000001 80b18e20 80b18e20 0011fc26
<4>[33425.773575] $12   : 00000000 00000400 00000000 00000000
<4>[33425.778787] $16   : 8a946680 8a368700 80a80000 80a80000
<4>[33425.783999] $20   : fffffff0 80b10860 8a368700 000016ef
<4>[33425.789213] $24   : 00000002 ff970c00                  
<4>[33425.794425] $28   : 84acc000 84acdd80 0041eb00 807234ac
<4>[33425.799637] Hi    : 000015e4
<4>[33425.802499] Lo    : 8d65c000
<4>[33425.805362] epc   : 8072b128 0x8072b128
<4>[33425.809178] ra    : 807234ac 0x807234ac
<4>[33425.812992] Status: 1100fc03	KERNEL EXL IE 
<4>[33425.817164] Cause : 40800008 (ExcCode 02)
<4>[33425.821156] BadVA : 0400064c
<4>[33425.824022] PrId  : 0001992f (MIPS 1004Kc)
<4>[33425.828092] Modules linked in: ksmbd(O) nf_nat_amanda nf_conntrack_amanda wireguard snd_usb_audio rndis_host mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length2(O) xt_length xt_ipv4options(O) xt_iprange xt_iface(O) xt_hl xt_helper xt_hashlimit xt_geoip(O) xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_PROTO(O) xt_NFQUEUE xt_NFLOG xt_NETMAP xt_MASQUERADE xt_LUA(O) xt_LOGMARK(O) xt_LOG xt_IPMARK(O) xt_HL xt_DSCP xt_DNETMAP(O) xt_DHCPMAC(O) xt_CT xt_CLASSIFY xt_CHECKSUM vhci_hcd usbnet usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress
<4>[33425.828409]  ums_alauda ts_kmp ts_fsm ts_bm snd_usbmidi_lib sch_cake ntfs3(O) nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_inet nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_bridge nf_conncount lz4_decompress lz4_compress libchacha20poly1305 libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_rpfilter ipt_ECN ip6table_raw ip6t_rpfilter ip_tables hwmon hid_generic ebtables ebt_vlan ebt_stp ebt_redirect
<4>[33425.915641]  ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat_xtables(O) compat(O) br_netfilter arptable_filter arpt_mangle arp_tables fuse cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact sg hid evdev ledtrig_gpio cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set ip6table_nat ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsd bonding ip6_gre ip_gre gre ifb nat46(O) sit ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp ah4 ip6_tunnel tunnel6 tunnel4
<4>[33426.002987]  rpcsec_gss_krb5 auth_rpcgss oid_registry tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore ovpn_dco(O) xfrm_user xfrm_ipcomp af_key lockd sunrpc grace isofs hfsplus hfs cdrom autofs4 dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 nls_koi8_r nls_cp866 nls_cp1251 zram algif_skcipher algif_rng algif_hash algif_aead af_alg arc4 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ahci libahci libata gpio_button_hotplug(O) xfs f2fs ext4 mbcache jbd2 btrfs xor raid6_pq mii tpm rng_core crc32_generic
<4>[33426.140522] Process kworker/2:2 (pid: 24661, threadinfo=e3a0f1ad, task=010bad09, tls=00000000)
<4>[33426.149096] Stack : 80e144d8 80722ed4 80a7b464 00000001 0000a2a2 8a368700 8a368740 807234ac
<4>[33426.157433]         8a1060e0 8a1060f0 00000001 80722ed4 00000001 80722ec4 8a368700 8a368740
<4>[33426.165770]         80a80000 80a80000 fffffff0 00000000 00002000 000016ef 0041eb00 80723774
<4>[33426.174108]         8a1060e0 8072365c 811ecba0 811ecb60 8a368700 80723914 8a1060f0 8071db50
<4>[33426.182447]         1100fc03 00000377 00000001 811ecb60 8a368700 80723c58 8358a0e0 811ecba0
[... cut ...]
<4>[40683.450963] [<80056998>] 0x80056998
<4>[40683.454448] [<80056820>] 0x80056820
<4>[40683.457962] [<80063e0c>] 0x80063e0c
<4>[40683.461430] [<80056820>] 0x80056820
<4>[40683.464924] [<80056820>] 0x80056820
<4>[40683.468393] [<8000376c>] 0x8000376c
<4>[40683.471872] 
<4>[40683.473352] Code: 02803025  8e02001c  00621821 <8c660000> 41656000  30a50001  000000c0  8f84000c  8e020000 
<4>[40683.483080] 
<4>[40683.484674] ---[ end trace 781f86db5cf39005 ]---
</pre>
</details>
Now I'm building 5.15 :)

Vladdrako, 2022-04-02:
I did a clean build and now looks like kallsyms work now(?).
But kernel panics are caused by software.
nlbwmon
<details>
<pre>
[  340.021902] CPU 0 Unable to handle kernel paging request at virtual address 00000304, epc == 806c909c, ra == 806c9114
[  340.032607] Oops[#1]:
[  340.034903] CPU: 0 PID: 5623 Comm: nlbwmon Tainted: G           O      5.10.109 #0
[  340.042445] $ 0   : 00000000 00000001 00000000 00000008
[  340.047671] $ 4   : 00000000 897abaa0 00000000 897abab0
[  340.052913] $ 8   : 000022f6 00000b86 00000b86 00000000
[  340.058146] $12   : 00000000 80c0322c 80c0323c 80c031bc
[  340.063369] $16   : 8986b800 00000000 80bfaa60 8973c258
[  340.068589] $20   : 00000000 00000000 00000000 83417540
[  340.073810] $24   : 00000000 00000000                  
[  340.079026] $28   : 897aa000 897abae0 0000009f 806c9114
[  340.084255] Hi    : 00000000
[  340.087124] Lo    : 00000000
[  340.090016] epc   : 806c909c neigh_dump_info+0x39c/0x5fc
[  340.095309] ra    : 806c9114 neigh_dump_info+0x414/0x5fc
[  340.100598] Status: 11008403	KERNEL EXL IE 
[  340.104774] Cause : 40800008 (ExcCode 02)
[  340.108758] BadVA : 00000304
[  340.111627] PrId  : 0001992f (MIPS 1004Kc)
[  340.115700] Modules linked in: ksmbd(O) nf_nat_amanda nf_conntrack_amanda wireguard snd_usb_audio rndis_host mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length2(O) xt_length xt_ipv4options(O) xt_iprange xt_iface(O) xt_hl xt_helper xt_hashlimit xt_geoip(O) xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_PROTO(O) xt_NFQUEUE xt_NFLOG xt_NETMAP xt_MASQUERADE xt_LUA(O) xt_LOGMARK(O) xt_LOG xt_IPMARK(O) xt_HL xt_DSCP xt_DNETMAP(O) xt_DHCPMAC(O) xt_CT xt_CLASSIFY xt_CHECKSUM vhci_hcd usbnet usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress
[  340.116019]  ums_alauda ts_kmp ts_fsm ts_bm snd_usbmidi_lib sch_cake nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_inet nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_bridge nf_conncount lz4_decompress lz4_compress libchacha20poly1305 libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_rpfilter ipt_ECN ip6table_raw ip6t_rpfilter ip_tables hwmon hid_generic ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype
[  340.203265]  ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat_xtables(O) compat(O) br_netfilter arptable_filter arpt_mangle arp_tables fuse cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact sg hid evdev ledtrig_gpio cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set ip6table_nat ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsd nfs nfs_ssc bonding ip6_gre ip_gre gre ifb nat46(O) sit ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp ah4 ip6_tunnel tunnel6 tunnel4
[  340.290833]  rpcsec_gss_krb5 auth_rpcgss oid_registry tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore ovpn_dco(O) xfrm_user xfrm_ipcomp af_key lockd sunrpc grace isofs hfsplus hfs cdrom autofs4 dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 nls_koi8_r nls_cp866 nls_cp1251 zram algif_skcipher algif_rng algif_hash algif_aead af_alg arc4 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ahci libahci libata gpio_button_hotplug(O) xfs f2fs ext4 mbcache jbd2 btrfs xor raid6_pq mii tpm rng_core crc32_generic
[  340.428886] Process nlbwmon (pid: 5623, threadinfo=896d3798, task=fbc30fbf, tls=77eb2dcc)
[  340.437040] Stack : 00000001 81412a80 80bfaa60 00000000 0000001c 00000002 897abba0 80b60000
[  340.445387]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  340.453728]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 000128c0
[  340.462070]         00000002 00000001 835ad000 80b668ac 00000000 00000000 0000000a 00000000
[  340.470411]         806c9120 00000000 8973c000 83417540 00004000 8973c258 00004000 8973c258
[  340.478748]         ...
[  340.481190] Call Trace:
[  340.483661] [<806c909c>] neigh_dump_info+0x39c/0x5fc
[... cut ...]
[ 3570.737614] [<8036dd0c>] blk_mq_submit_bio+0x80/0x43c
[ 3570.742661] [<8035f744>] submit_bio_noacct+0x150/0x1d0
[ 3570.748153] [<82c78ca8>] btrfs_map_bio+0x1fc/0x4c8 [btrfs]
[ 3570.753864] [<82c37d10>] btrfs_submit_data_bio+0xa8/0x240 [btrfs]
[ 3570.760173] [<82c5fcc0>] submit_one_bio+0x50/0x534 [btrfs]
[ 3570.765870] [<82c66498>] btrfs_bio_alloc+0x178/0x968 [btrfs]
[ 3570.771614] 
[ 3570.773104] ---[ end trace a6cc23d5c042715a ]---
[ 3571.170563] Rebooting in 3 seconds..
</pre>
</details>
Maybe it helps somehow...

Vladdrako, 2022-04-03:
Hmm, maybe good idea to add this [ba1f8c93](https://github.com/coolsnowwolf/lede/commit/ba1f8c931c9817fdb2e09bacfe988dc77884bc61)
https://www.spinics.net/lists/netdev/msg727603.html

Vladdrako, 2022-04-03:
@nbd168 Finally I caught napi/mtk_eth crash
<pre>
[  370.109000] Unhandled kernel unaligned access[#1]:
[  370.113827] CPU: 1 PID: 699 Comm: napi/mtk_eth-6 Tainted: G           O      5.10.109 #0
[  370.121884] $ 0   : 00000000 00000001 0000005c 0000006a
[  370.127120] $ 4   : 0067f000 00000a20 8069b8e0 00000000
[  370.132352] $ 8   : 00000100 00000040 00000039 0000000b
[  370.137594] $12   : 00000000 80c0322c 80c0323c 80c031bc
[  370.142831] $16   : 81412a80 00000a20 80b6540c 00000000
[  370.148067] $20   : 8069b8e0 80b60000 00000001 0030f251
[  370.153301] $24   : 00001870 00000000                  
[  370.158528] $28   : 82378000 82379d40 c5ee0000 8069b8e0
[  370.163755] Hi    : 00000000
[  370.166621] Lo    : 00000041
[  370.169518] epc   : 801a9db0 kmem_cache_alloc+0x1cc/0x3f8
[  370.174912] ra    : 8069b8e0 __build_skb+0x2c/0x80
[  370.179682] Status: 1100fc03	KERNEL EXL IE 
[  370.183869] Cause : 40800010 (ExcCode 04)
[  370.187860] BadVA : 0000006a
[  370.190728] PrId  : 0001992f (MIPS 1004Kc)
[  370.194806] Modules linked in: ksmbd(O) nf_nat_amanda nf_conntrack_amanda wireguard snd_usb_audio rndis_host mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length2(O) xt_length xt_ipv4options(O) xt_iprange xt_iface(O) xt_hl xt_helper xt_hashlimit xt_geoip(O) xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_PROTO(O) xt_NFQUEUE xt_NFLOG xt_NETMAP xt_MASQUERADE xt_LUA(O) xt_LOGMARK(O) xt_LOG xt_IPMARK(O) xt_HL xt_DSCP xt_DNETMAP(O) xt_DHCPMAC(O) xt_CT xt_CLASSIFY xt_CHECKSUM vhci_hcd usbnet usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress
[  370.195302]  ums_alauda ts_kmp ts_fsm ts_bm snd_usbmidi_lib sch_cake nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_inet nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_bridge nf_conncount lz4_decompress lz4_compress libchacha20poly1305 libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_rpfilter ipt_ECN ip6table_raw ip6t_rpfilter ip_tables hwmon hid_generic ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype
[  370.282689]  ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat_xtables(O) compat(O) br_netfilter arptable_filter arpt_mangle arp_tables fuse cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact sg hid evdev ledtrig_gpio cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set ip6table_nat ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsd nfs nfs_ssc bonding ip6_gre ip_gre gre ifb nat46(O) sit ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp ah4 ip6_tunnel tunnel6 tunnel4
[  370.370411]  rpcsec_gss_krb5 auth_rpcgss oid_registry tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore ovpn_dco(O) xfrm_user xfrm_ipcomp af_key lockd sunrpc grace isofs hfsplus hfs cdrom autofs4 dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 nls_koi8_r nls_cp866 nls_cp1251 zram algif_skcipher algif_rng algif_hash algif_aead af_alg arc4 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ahci libahci libata gpio_button_hotplug(O) xfs f2fs ext4 mbcache jbd2 btrfs xor raid6_pq mii tpm rng_core crc32_generic
[  370.508584] Process napi/mtk_eth-6 (pid: 699, threadinfo=2d768a5a, task=5a243e36, tls=00000000)
[  370.517246] Stack : 000d2a20 00000000 00000001 000110c4 80d40000 80d40000 0000000e 0000081e
[  370.525608]         a972d460 887eaba0 887eaba0 00000960 865476e0 887eaba0 000110c4 0000002e
[  370.533968]         81776080 8069b8e0 812bd2b0 812bd2b0 80e6dd00 00000960 887eaba0 00000960
[  370.542328]         a972d460 8069b950 c5ee0000 805e52c4 065476e0 817765ec 00000146 817765ec
[  370.550693]         00000146 805e5184 816f61e0 812bdb20 28ffb2d0 ff7db5d0 00000000 00000000
[  370.559056]         ...
[  370.561501] Call Trace:
[  370.563966] [<801a9db0>] kmem_cache_alloc+0x1cc/0x3f8
[  370.569018] [<8069b8e0>] __build_skb+0x2c/0x80
[  370.573452] [<8069b950>] build_skb+0x1c/0xa4
[  370.577720] [<805e5184>] mtk_poll_rx+0x29c/0x63c
[... cut ...]
[  370.586851] [<806bedd0>] __napi_poll+0x3c/0x15c
[  370.591368] [<806befa4>] napi_threaded_poll+0xb4/0x11c
[  370.596511] [<80056a2c>] kthread+0x178/0x180
[  370.600775] [<8000380c>] ret_from_kernel_thread+0x14/0x1c
[  370.606159] 
[  370.607644] Code: 02803025  8e02001c  00621821 <8c660000> 41656000  30a50001  000000c0  8f84000c  8e020000 
[  370.617399] 
[  370.618967] ---[ end trace b3632d52b3fa0fc2 ]---
[  370.898808] Kernel panic - not syncing: Fatal exception in interrupt
[  371.123500] Rebooting in 3 seconds..
</pre>

xabolcs, 2022-04-03:
> Logs collected by pstore.

Do you have some manual / how-to somewhere?

When I read about it on the ML for `mt7622`, I tought it would be nice to add to other (sub)targets!

nbd168, 2022-04-03:
So if I understand what you wrote correctly, you get crashes even without the commit that enables threaded NAPI, right?
From what I read in the crash dumps, it happens during memory allocation of new skbs. Based on that, the problem is likely somewhere else entirely in that it might be a skb double-free, or free of an invalid pointer.
Can you show me some KALLSYMS-enabled crashes without threaded NAPI to see if they're related?

Vladdrako, 2022-04-03:
> So if I understand what you wrote correctly, you get crashes even without the commit that enables threaded NAPI, right?

Yes, but caused not by napi.

> Can you show me some KALLSYMS-enabled crashes without threaded NAPI to see if they're related?

Ofc, will rebuild kernel as soon as possible.

Vladdrako, 2022-04-03:
> Do you have some manual / how-to somewhere?

I enabled next symbols in kernel config:
```
CONFIG_PSTORE=y
CONFIG_PSTORE_COMPRESS=y
CONFIG_PSTORE_COMPRESS_DEFAULT="zstd"
CONFIG_PSTORE_CONSOLE=y
CONFIG_PSTORE_PMSG=y
CONFIG_PSTORE_RAM=y
CONFIG_PSTORE_ZSTD_COMPRESS=y
CONFIG_PSTORE_ZSTD_COMPRESS_DEFAULT=y
```
Logs located in `/sys/fs/pstore` folder.

Vladdrako, 2022-04-03:
@nbd168 Done. A quick test by copying a huge file from PC to USB-HDD connected to the router.
<details>
<pre>
[  553.722987] Unhandled kernel unaligned access[#1]:
[  553.727812] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O      5.10.109 #0
[  553.735290] $ 0   : 00000000 00000001 0000005c 0000006a
[  553.740549] $ 4   : 00670000 00000a20 8069b8a0 00000000
[  553.745789] $ 8   : 00000100 00000040 00000039 ffffffff
[  553.751042] $12   : 00000092 80c0322c 80c0323c 80c031bc
[  553.756268] $16   : 81412a80 00000a20 80b6540c 00000000
[  553.761488] $20   : 8069b8a0 80b60000 00000001 026fac10
[  553.766704] $24   : 00001e6e 00000000                  
[  553.771920] $28   : 80b52000 8140bd80 c5ee0000 8069b8a0
[  553.777132] Hi    : 00000000
[  553.779995] Lo    : 00000041
[  553.782894] epc   : 801a9db0 kmem_cache_alloc+0x1cc/0x3f8
[  553.788278] ra    : 8069b8a0 __build_skb+0x2c/0x80
[  553.793045] Status: 11008403	KERNEL EXL IE 
[  553.797217] Cause : 40800010 (ExcCode 04)
[  553.801207] BadVA : 0000006a
[  553.804071] PrId  : 0001992f (MIPS 1004Kc)
[  553.808141] Modules linked in: ksmbd(O) nf_nat_amanda nf_conntrack_amanda wireguard snd_usb_audio rndis_host mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length2(O) xt_length xt_ipv4options(O) xt_iprange xt_iface(O) xt_hl xt_helper xt_hashlimit xt_geoip(O) xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_PROTO(O) xt_NFQUEUE xt_NFLOG xt_NETMAP xt_MASQUERADE xt_LUA(O) xt_LOGMARK(O) xt_LOG xt_IPMARK(O) xt_HL xt_DSCP xt_DNETMAP(O) xt_DHCPMAC(O) xt_CT xt_CLASSIFY xt_CHECKSUM vhci_hcd usbnet usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress
[  553.808461]  ums_alauda ts_kmp ts_fsm ts_bm snd_usbmidi_lib sch_cake nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_inet nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_bridge nf_conncount lz4_decompress lz4_compress libchacha20poly1305 libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_rpfilter ipt_ECN ip6table_raw ip6t_rpfilter ip_tables hwmon hid_generic ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype
[  553.895703]  ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat_xtables(O) compat(O) br_netfilter arptable_filter arpt_mangle arp_tables fuse cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact sg hid evdev ledtrig_gpio cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set ip6table_nat ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsd nfs nfs_ssc bonding ip6_gre ip_gre gre ifb nat46(O) sit ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp ah4 ip6_tunnel tunnel6 tunnel4
[  553.983272]  rpcsec_gss_krb5 auth_rpcgss oid_registry tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore ovpn_dco(O) xfrm_user xfrm_ipcomp af_key lockd sunrpc grace isofs hfsplus hfs cdrom autofs4 dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 nls_koi8_r nls_cp866 nls_cp1251 zram algif_skcipher algif_rng algif_hash algif_aead af_alg arc4 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ahci libahci libata gpio_button_hotplug(O) xfs f2fs ext4 mbcache jbd2 btrfs xor raid6_pq mii tpm rng_core crc32_generic
[  554.121328] Process swapper/0 (pid: 0, threadinfo=8bceb636, task=28b04905, tls=00000000)
[  554.129382] Stack : 8908bcc0 0000f800 00000006 816f2750 80d40000 80d40000 0000000e 0000081e
[  554.137727]         a2ed6130 89ebb500 89ebb500 00000960 87170660 89ebb500 00014622 00000037
[  554.146070]         816f2080 8069b8a0 8908bcc0 000005ee 00014622 00000036 89ebb500 00000960
[  554.154411]         a2ed6130 8069b910 c5ee0000 805e5138 07170660 816f25ec 00000013 816f25ec
[  554.162752]         00000013 805e5144 8917dad4 83abac00 ff7cc638 ff7cc618 00000000 00000000
[  554.171099]         ...
[  554.173542] Call Trace:
[  554.175999] [<801a9db0>] kmem_cache_alloc+0x1cc/0x3f8
[  554.181047] [<8069b8a0>] __build_skb+0x2c/0x80
[  554.185471] [<8069b910>] build_skb+0x1c/0xa4
[... cut ...]
[  554.810926] [<800cb3b8>] smp_call_function_many_cond+0x438/0x454
[  554.816905] [<800cb424>] smp_call_function+0x30/0x64
[  554.821850] [<80035154>] panic+0x12c/0x334
[  554.825929] [<8000ab9c>] die+0x120/0x128
[  554.829833] [<8000d2dc>] emulate_load_store_insn+0x294/0x7b0
[  554.835470] [<8000d880>] do_ade+0x88/0x1b0
[  554.839550] [<80004194>] handle_adel_int+0x30/0x3c
[  554.844341] [<8069b8a0>] __build_skb+0x2c/0x80
[  554.848766] 
[  554.850252] ---[ end trace d6b8fb6b580c7786 ]---
[  555.247841] Rebooting in 3 seconds..
</pre>
</details>

nbd168, 2022-04-03:
Which OpenWrt version are you using. Does your build include the openwrt commit 5e34b316c5b3bf06a1d748cb5569d28b8334820f?

Vladdrako, 2022-04-03:
> Which OpenWrt version are you using. Does your build include the openwrt commit 5e34b316c5b3bf06a1d748cb5569d28b8334820f?

Own build. Yes, the latest version of mt76.

Vladdrako, 2022-04-16:
On kernel 5.15 same.
<details>
<pre>
[ 4397.715963] Unhandled kernel unaligned access[#1]:
[ 4397.720803] CPU: 0 PID: 1685 Comm: napi/phy1-9 Tainted: G           O      5.15.34 #0
[ 4397.728622] $ 0   : 00000000 00000001 0000006a 00670000
[ 4397.733873] $ 4   : 81412e00 00000a20 00000001 00000000
[ 4397.739115] $ 8   : 003fb088 84f986f4 3041277c 8962d8e2
[ 4397.744360] $12   : 7c9b3bb5 1f938c86 17d4419c 00000000
[ 4397.749605] $16   : 00000a20 80d60000 81412e00 00000000
[ 4397.754849] $20   : 0000000e 80b5d434 8067cfa0 80b60000
[ 4397.760097] $24   : 00000000 8039e768                  
[ 4397.765376] $28   : 839cc000 839cd518 89531f28 801b5774
[ 4397.770621] Hi    : 00000002
[ 4397.773498] Lo    : 00000001
[ 4397.776367] epc   : 801b5814 kmem_cache_alloc+0xe8/0x5c8
[ 4397.781710] ra    : 801b5774 kmem_cache_alloc+0x48/0x5c8
[ 4397.787023] Status: 11000403	KERNEL EXL IE 
[ 4397.791220] Cause : 40800010 (ExcCode 04)
[ 4397.795220] BadVA : 0000006a
[ 4397.798093] PrId  : 0001992f (MIPS 1004Kc)
[ 4397.802180] Modules linked in: ksmbd(O) nf_nat_amanda nf_conntrack_amanda pppoe ppp_async wireguard snd_usb_audio rndis_host pptp pppox ppp_mppe ppp_generic mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) cdc_ether xt_time xt_tcpudp xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_LOG vhci_hcd usbnet usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm snd_usbmidi_lib slhc sch_cake nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_inet nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6
[ 4397.802924]  nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_bridge lz4_decompress lz4_compress libchacha20poly1305 libblake2s_generic iptable_mangle iptable_filter ip_tables hwmon hid_generic ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat(O) cls_flower arptable_filter arpt_mangle arp_tables act_vlan fuse ntfs3 cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact sg hid evdev ledtrig_gpio cryptodev(O) ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsd nfs bonding ip6_gre ip_gre gre ifb nat46(O) sit
[ 4397.890374]  ip6_tunnel tunnel6 tunnel4 udp_diag raw_diag rpcsec_gss_krb5 auth_rpcgss tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore ovpn_dco(O) lockd sunrpc grace isofs hfsplus hfs cdrom autofs4 nls_utf8 nls_koi8_r nls_cp866 nls_cp1251 zram algif_skcipher algif_rng algif_hash algif_aead af_alg arc4 leds_gpio ohci_platform ohci_hcd ahci libahci libata fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug(O) xfs f2fs ext4 mbcache jbd2 exfat btrfs xor raid6_pq dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax mii tpm rng_core
[ 4398.030827] Process napi/phy1-9 (pid: 1685, threadinfo=47a22270, task=e16e555e, tls=00000000)
[ 4398.039363] Stack : 00000002 00000001 816c8e40 8067b680 80c2b540 816c8e40 000006f4 00000a20
[ 4398.047757]         80687784 00000000 81412e00 00000000 0000c7d2 8067cfa0 0000000d 80d30000
[ 4398.056144]         80ad6f18 80d60000 816c8e40 816c8e40 838298b8 00000140 000005b4 00000036
[ 4398.064534]         816c8e40 00000000 0000c7d2 80687784 838298b8 0000c21e 84f98140 00000000
[ 4398.072932]         80b60000 84586af4 00000000 00000110 00000036 000005b4 0000010a 00000011
[ 4398.081320]         ...
[ 4398.083771] Call Trace:
[ 4398.086208] [<801b5814>] kmem_cache_alloc+0xe8/0x5c8
[ 4398.091203] [<8067cfa0>] __alloc_skb+0x15c/0x1fc
[ 4398.095833] [<80687784>] skb_segment+0x1e8/0xd18
[ 4398.100465] [<807c07d0>] tcp_gso_segment+0x118/0x5a0
[ 4398.105461] [<807d911c>] inet_gso_segment+0x174/0x3cc
[ 4398.110526] [<8069f1b8>] skb_mac_gso_segment+0xb4/0x130
[ 4398.115780] [<8069f2b4>] __skb_gso_segment+0x80/0x1b8
[ 4398.120832] [<8069f920>] validate_xmit_skb.constprop.0+0x13c/0x334
[ 4398.127010] [<806a0618>] __dev_queue_xmit+0x4ac/0xd04
[ 4398.132058] [<808b0590>] br_dev_queue_push_xmit+0x7c/0x198
[ 4398.137568] [<808acae0>] br_dev_xmit+0x304/0x50c
[ 4398.142187] [<8069ff84>] dev_hard_start_xmit+0x10c/0x1e8
[ 4398.147500] [<806a0cc8>] __dev_queue_xmit+0xb5c/0xd04
[ 4398.152546] [<807871d4>] ip_finish_output2+0x2bc/0x67c
[ 4398.157708] [<807896bc>] __ip_queue_xmit+0x1a8/0x4b0
[ 4398.162675] [<807ad3dc>] __tcp_transmit_skb+0x578/0xd6c
[ 4398.167894] [<807af21c>] tcp_write_xmit+0x3f8/0x11dc
[ 4398.172855] [<807b0044>] __tcp_push_pending_frames+0x44/0x110
[ 4398.178594] [<807a8b8c>] tcp_rcv_established+0x30c/0x7f8
[ 4398.183927] [<807b6ab8>] tcp_v4_do_rcv+0x1f0/0x2b4
[ 4398.188717] [<807b9f38>] tcp_v4_rcv+0xd44/0xe58
[... cut ...]
[ 4398.255681] [<844c51c8>] mt76_rx_poll_complete+0x310/0x5fc [mt76]
[ 4398.261792] [<844c1af4>] mt76_dma_rx_poll+0x3b4/0x7b8 [mt76]
[ 4398.267472] [<8416c8e4>] mt7615_unregister_device+0x270/0x328 [mt7615e]
[ 4398.274211] 
[ 4398.275700] Code: 00000000  8e42001c  02821021 <8c470000> 41656000  30a50001  000000c0  8f84000c  8e420000 
[ 4398.285501] 
[ 4398.287663] ---[ end trace 6f2662d58ee782bd ]---
[ 4398.320199] Kernel panic - not syncing: Fatal exception in interrupt
[ 4398.350776] Rebooting in 3 seconds..
</pre>
</details>

Toon2, 2022-04-21:
Same Issues here. Cudy WR2100
I think it started with OpenWRT 21.02.2
Reboots under heavy load, but sometimes by idle conditions.
`Apr 11 17:44:07 OpenWrt kernel: [1960419.761219] BUG: Bad page state in process swapper/0  pfn:0723b`
Today updated to 21.02.3 - after 4 hours reboot - but nothing in logs.

Vladdrako, 2022-04-29:
<details>
<pre>
[11288.958900] CPU 3 Unable to handle kernel paging request at virtual address 00000104, epc == 801bc350, ra == 801bc078
[11288.969533] Oops[#1]:
[11288.971815] CPU: 3 PID: 346 Comm: napi/mtk_eth-6 Tainted: G           O      5.15.36 #0
[11288.979800] $ 0   : 00000000 00000001 00000122 00000100
[11288.985046] $ 4   : fffffffe 00000000 806a5438 812f0ba0
[11288.990288] $ 8   : 00f436fb 00000040 00000039 00227432
[11288.995529] $12   : 81524120 84548e44 4f357d0d 00000000
[11289.000770] $16   : 80e29ed4 00000000 812f0ba0 81413e00
[11289.006011] $20   : 80150015 00150015 81412e00 00000001
[11289.011252] $24   : 00000010 803a6a94                  
[11289.016491] $28   : 8163e000 8163fcf8 80d90000 801bc078
[11289.021733] Hi    : 00000000
[11289.024603] Lo    : 00000041
[11289.027473] epc   : 801bc350 ___slab_alloc.constprop.0+0x4f4/0x6d8
[11289.033678] ra    : 801bc078 ___slab_alloc.constprop.0+0x21c/0x6d8
[11289.039849] Status: 1100fc02	KERNEL EXL 
[11289.043775] Cause : 4080000c (ExcCode 03)
[11289.047771] BadVA : 00000104
[11289.050641] PrId  : 0001992f (MIPS 1004Kc)
[11289.054723] Modules linked in: ksmbd(O) nf_nat_amanda nf_conntrack_amanda pppoe ppp_async wireguard snd_usb_audio rndis_host pptp pppox ppp_mppe ppp_generic mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) cdc_ether xt_time xt_tcpudp xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_helper xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_CT vhci_hcd usbnet usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm snd_usbmidi_lib slhc sch_cake nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log
[11289.055470]  nft_limit nft_hash nft_fwd_netdev nft_fib_inet nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_bridge nf_conncount mdio_netlink(O) lz4_decompress lz4_compress libchacha20poly1305 libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ip_tables hwmon hid_generic ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat(O) cls_flower arptable_filter arpt_mangle arp_tables act_vlan fuse ntfs3 cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic
[11289.142745]  act_skbedit act_mirred act_gact sg hid evdev ledtrig_gpio cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsd nfs bonding ip6_gre ip_gre gre ifb nat46(O) sit ip6_tunnel tunnel6 tunnel4 udp_diag raw_diag rpcsec_gss_krb5 auth_rpcgss tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore ovpn_dco(O) lockd sunrpc grace isofs hfsplus hfs cdrom autofs4 nls_utf8 nls_koi8_r nls_cp866 nls_cp1251 zram algif_skcipher algif_rng algif_hash algif_aead af_alg arc4 leds_gpio ohci_platform ohci_hcd ahci libahci libata fsl_mph_dr_of ehci_platform ehci_fsl gpio_button_hotplug(O) xfs f2fs ext4
[11289.229943]  mbcache jbd2 exfat btrfs xor raid6_pq dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax mii tpm rng_core
[11289.327498] Process napi/mtk_eth-6 (pid: 346, threadinfo=d2a64ece, task=7ca34ad7, tls=00000000)
[11289.336187] Stack : 892ac400 892ac48c 0000000e 00000010 00000000 808646d0 837db000 00000000
[11289.344574]         80da0000 80150015 812eceac 89d6b780 80d9fa80 80867644 80b87438 80a04360
[11289.352961]         00000a20 80d90000 81412e00 00000000 00000000 80b87438 806a5438 80b80000
[11289.361347]         0000015b 801bc570 80c572e8 00000a20 806a5438 806c95b8 00000a20 801bd690
[11289.369733]         80d90000 00000040 81524010 0000081e 0c8be420 8612e3e0 8612e3e0 00000960
[11289.378119]         ...
[11289.380568] Call Trace:
[11289.383004] [<801bc350>] ___slab_alloc.constprop.0+0x4f4/0x6d8
[11289.388855] [<801bc570>] __slab_alloc.constprop.0+0x3c/0x90
[11289.394424] [<801bd690>] kmem_cache_alloc+0x4cc/0x5c8
[11289.399468] [<806a5438>] __build_skb+0x2c/0x70
[11289.403911] [<806a5498>] build_skb+0x1c/0xb4
[11289.408178] [<805e5c14>] mtk_napi_rx+0x380/0x7f4
[11289.412819] [<806c861c>] __napi_poll+0x3c/0x19c
[11289.417372] [<806c8830>] napi_threaded_poll+0xb4/0x114
[11289.422508] [<80059b0c>] kthread+0x17c/0x1a0
[11289.426788] [<8000378c>] ret_from_kernel_thread+0x14/0x1c
[11289.432189] 
[11289.433677] Code: af820010  8e020008  8e030004 <ac620004> ac430000  24020100  ae020004  24020122  ae020008 
[11289.443470] 
[11289.445346] ---[ end trace b8bd054b852d3bdb ]---
[11289.476671] Kernel panic - not syncing: Fatal exception in interrupt
[11289.507175] Rebooting in 3 seconds..
</pre>
</details>

ryderlee1110, 2022-05-29:
@Vladdrako  Can you enable CONFIG_KASAN and paste the output here?

mgkiller7, 2022-07-21:
i also account the similar kernel fault in openwrt 21.02 with mt7621 when doing iperf3 through ipsec  ethernet.

Kernel Fault Type
1、BUG: Bad page state                                               __do_softirq-> net_rx_action->__napi_poll->mtk_napi_rx->page_frag_alloc->...->get_page_from_freelist
2、CPU 2 Unable to handle kernel paging request        __do_softirq-> net_rx_action->__napi_poll->gro_cell_poll->napi_complete_done-> ... ->kmem_cache_alloc
3、CPU 2 Unable to handle kernel paging request        __do_softirq-> net_rx_action->__napi_poll->gro_cell_poll->napi_gro_receive-> ... ->__kmalloc
4、CPU 2 Unable to handle kernel paging request        __do_softirq-> net_rx_action->__napi_poll->gro_cell_poll->napi_complete_done-> ... ->__kmalloc
5、CPU 2 Unable to handle kernel paging request        mtk_irq_clear->mtk_aead_handle_result->mtk_unmap_dma
6、CPU 0 Unable to handle kernel paging request        gro_normal_list->gro_normal_one
7、CPU 0 Unable to handle kernel paging request      __do_softirq->net_rx_action->__napi_poll->mtk_napi_tx->gro_normal_one

Detail info of each kernel fault：

[ 1353.863451] BUG: Bad page state in process ksoftirqd/0  pfn:00b38
[ 1353.869543] page:80f893e0 refcount:-3 mapcount:0 mapping:00000000 index:0x0
[ 1353.876471] flags: 0x0()
[ 1353.879003] raw: 00000000 00000100 00000122 00000000 00000000 00000000 ffffffff fffffffd
[ 1353.887053] raw: 00000000
[ 1353.889657] page dumped because: nonzero _refcount
[ 1353.894420] Modules linked in: crypto_user pcrypt crypto_hw_eip93 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 cfg80211 compat xt_CT iptable_raw [last unloaded: crypto_user]
[ 1353.910805] CPU: 0 PID: 9 Comm: ksoftirqd/0 Not tainted 5.4.179 #20
[ 1353.917038] Stack : 00000000 00000000 00000195 80068d98 00000000 80650000 80600000 8058dfcc
[ 1353.925362]         8060715c 8fc5db24 80f60000 00000000 80650000 00000001 8fc5dab8 06d760c7
[ 1353.933685]         00000000 00000000 80f70000 80650000 00000000 00000000 00000196 00000000
[ 1353.942005]         00000000 3649c879 80f60000 666f736b 80650000 00000000 00000000 8059f79c
[ 1353.950327]         00000201 8fc5dca8 80f893e0 00000001 00000002 802ab940 00000000 80f30000
[ 1353.958648]         ...
[ 1353.961082] Call Trace:
[ 1353.963537] [<8000b0d8>] show_stack+0x6c/0x12c
[ 1353.967978] [<80500d30>] dump_stack+0xa0/0xcc
[ 1353.972322] [<8011cafc>] bad_page+0x114/0x144
[ 1353.976667] [<8011f9d0>] get_page_from_freelist+0x7c0/0xa44
[ 1353.982218] [<80120afc>] __alloc_pages_nodemask+0x9f4/0xb4c
[ 1353.987767] [<80120d84>] page_frag_alloc+0x48/0x170
[ 1353.992639] [<803050bc>] mtk_napi_rx+0x1f4/0x600
[ 1353.997244] [<803a94f8>] __napi_poll+0x40/0x130
[ 1354.001756] [<803a9830>] net_rx_action+0x10c/0x254
[ 1354.006537] [<8051d99c>] __do_softirq+0x15c/0x2f0
[ 1354.011225] [<80027b6c>] run_ksoftirqd+0x44/0x60
[ 1354.015837] [<8004497c>] smpboot_thread_fn+0x1b8/0x1e8
[ 1354.020967] [<800410f8>] kthread+0x118/0x120
[ 1354.025221] [<80005f98>] ret_from_kernel_thread+0x14/0x1c
[ 1354.030598] Disabling lock debugging due to kernel taint
[ 1354.036390] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 803a8658, ra == 803a8edc
[ 1354.047016] Oops[#1]:
[ 1354.049291] CPU: 0 PID: 9 Comm: ksoftirqd/0 Tainted: G    B             5.4.179 #20
[ 1354.056910] $ 0   : 00000000 00000001 00000000 00000000
[ 1354.062117] $ 4   : 8ffc56d0 8ffc5750 00000001 8ffc5750
[ 1354.067321] $ 8   : 0000c95c 00006617 8fdbfa10 0000061e
[ 1354.072525] $12   : 00000024 ffffff7f 80f60000 80f60000
[ 1354.077729] $16   : 8ffc5750 00000003 8ffc56d0 a0ab28d0
[ 1354.082934] $20   : 00000003 8fc34000 00000040 080abfff
[ 1354.088137] $24   : 00000003 8001c930                  
[ 1354.093341] $28   : 8fc5c000 8fc5dd18 8ffc5750 803a8edc
[ 1354.098546] Hi    : 00000000
[ 1354.101408] Lo    : 00000000
[ 1354.104288] epc   : 803a8658 gro_normal_one+0x14/0x48
[ 1354.109318] ra    : 803a8edc napi_gro_receive+0x74/0xd4
[ 1354.114513] Status: 11008403 KERNEL EXL IE 
[ 1354.118679] Cause : 4080000c (ExcCode 03)
[ 1354.122663] BadVA : 00000000
[ 1354.125526] PrId  : 0001992f (MIPS 1004Kc)
[ 1354.129598] Modules linked in: crypto_user pcrypt crypto_hw_eip93 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 cfg80211 compat xt_CT iptable_raw [last unloaded: crypto_user]
[ 1354.145986] Process ksoftirqd/0 (pid: 9, threadinfo=bf8c27fa, task=4e7db4c9, tls=00000000)
[ 1354.154205] Stack : 8ffc508c 80303c44 8149a970 06d760c7 8ffc56d0 8ffc556c 8ffc5080 80305284
[ 1354.162529]         8ffc5080 06d760c7 00000000 8fce3100 00000000 8fe213c0 806553b4 00000001
[ 1354.170851]         80b7f8a0 0000008d 000017a8 c5ea0000 00000234 000005ea 00b7f8e0 90430855
[ 1354.179172]         deadbef3 00000000 8ffc55ee 805d8a74 00000000 00000000 00000000 00000000
[ 1354.187494]         00000000 00000000 8ffc56d0 00000040 00000000 8fc5de03 80650000 80650000
[ 1354.195816]         ...
[... 443 lines cut ...]
[  142.815468] ra    : 803a8edc napi_gro_receive+0x74/0xd4
[  142.820663] Status: 11008403 KERNEL EXL IE 
[  142.824828] Cause : 4080000c (ExcCode 03)
[  142.828813] BadVA : 00000000
[  142.831676] PrId  : 0001992f (MIPS 1004Kc)
[  142.835747] Modules linked in: crypto_user pcrypt crypto_hw_eip93 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 cfg80211 compat xt_CT iptable_raw
[  142.849628] Process ksoftirqd/0 (pid: 9, threadinfo=0da20add, task=244beed1, tls=00000000)
[  142.857848] Stack : 8fc2e080 803013c4 8149abf0 017ac49d 8fc2e6d0 8fc2e56c 8fc2e080 80305284
[... cut ...]
[  142.913635] [<803a94f8>] __napi_poll+0x40/0x130
[  142.918148] [<803a9830>] net_rx_action+0x10c/0x254
[  142.922931] [<8051d99c>] __do_softirq+0x15c/0x2f0
[  142.927620] [<80027b6c>] run_ksoftirqd+0x44/0x60
[  142.932233] [<8004497c>] smpboot_thread_fn+0x1b8/0x1e8
[  142.937363] [<800410f8>] kthread+0x118/0x120
[  142.941620] [<80005f98>] ret_from_kernel_thread+0x14/0x1c
[  142.946994] Code: ac850084  aca70000  aca20004 <ac450000> 8c820088  00c23021  3c028065  ac860088  8c425970 
[  142.956708]

nledovskikh, 2022-07-22:
It's seems that I have the same issue using ath9k with AMSDU support enabled on mt7621.
The same problems with memory corruption under iperf bi-directional load running tcp mode.

sweroam-mibr, 2022-09-25:
Same issue with unielec u7621-01-ax hardware, so no ath9k. i suspect the mediatek ethernet driver. i will solder a rs232 to catch the crash.

Rexadev, 2022-12-24:
For me problem started after installing SQM
TL-WR850N v2

nledovskikh, 2022-12-26:
Could you please test with testing kernel enabled (5.15)?
It's seems that the problem is resolved there.

Rexadev, 2022-12-26:
> Could you please test with testing kernel enabled (5.15)?
> It's seems that the problem is resolved there.

How to do that? Please tell.

nledovskikh, 2022-12-27:
> > Could you please test with testing kernel enabled (5.15)?
> > It's seems that the problem is resolved there.
> 
> How to do that? Please tell.
For openwrt 22.03, run 'make menuconfig' and choose 'Global build settings -> Use testing kernel version'
(KERNEL_TESTING_PATCHVER variable set to 5.15 in target/linux/ramips/Makefile)

Rexadev, 2022-12-27:
Can I do it in https://firmware-selector.openwrt.org/
@nledovskikh

nledovskikh, 2022-12-28:
> Can I do it in https://firmware-selector.openwrt.org/ @nledovskikh

No, you can't. You need to compile it from scratch.

sweroam-mibr, 2023-01-12:
Have noticed that just before the router hangs, some commands stop working, like ip, wg show etc...  also found allot of processes nf_ft_offload_stats.. 
[wg.txt](https://github.com/openwrt/openwrt/files/10400935/wg.txt)

Vladdrako, 2023-01-12:
Watched a video from USB-HDD through rclone. ramips/mi-r3pro/5.15
<details>
<pre>
<1>[ 4507.509208] CPU 1 Unable to handle kernel paging request at virtual address 00000000, epc == 804ade48, ra == 807d6d74
<4>[ 4507.519899] Oops[#1]:
<4>[ 4507.522191] CPU: 1 PID: 1564 Comm: napi/phy1-10 Not tainted 5.15.86 #0
<4>[ 4507.528716] $ 0   : 00000000 00000001 876ca600 876cb381
<4>[ 4507.533974] $ 4   : 00000000 87f3cc04 876cb380 00000000
<4>[ 4507.539229] $ 8   : 0c800000 00000219 04264e84 4089550c
<4>[ 4507.544478] $12   : 0c49b8c3 00000243 d4f0370b 00000219
<4>[ 4507.549734] $16   : 876ca900 87f3cb00 2279c4f1 2279c4f1
<4>[ 4507.554993] $20   : 00000004 0000001d 876ca900 876cb380
<4>[ 4507.560260] $24   : 00000000 807b28dc                  
<4>[ 4507.565539] $28   : 8437a000 8437ba00 0000001d 807d6d74
<4>[ 4507.570805] Hi    : 00000219
<4>[ 4507.573687] Lo    : 168728bf
<4>[ 4507.576566] epc   : 804ade48 0x804ade48
<4>[ 4507.580401] ra    : 807d6d74 0x807d6d74
<4>[ 4507.584235] Status: 1100fc03	KERNEL EXL IE 
<4>[ 4507.588438] Cause : 4080000c (ExcCode 03)
<4>[ 4507.592441] BadVA : 00000000
<4>[ 4507.595320] PrId  : 0001992f (MIPS 1004Kc)
<4>[ 4507.599408] Modules linked in: nf_nat_amanda nf_conntrack_amanda pppoe ppp_async wireguard sch_fq_pie rndis_host pptp pppox ppp_mppe ppp_generic nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_geoip xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY vhci_hcd usbnet usblp usbip_host usbip_core usbhid ts_kmp ts_fsm ts_bm slhc sch_red sch_prio sch_pie sch_mqprio sch_drr poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_flow_offload
<4>[ 4507.600205]  nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat nf_log_syslog nf_flow_table nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge nf_conncount mdio_netlink libcurve25519_generic iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables hwmon hid_generic ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_itu_t crc_ccitt compat cls_flower cdc_wdm arptable_filter arpt_mangle arp_tables act_vlan fuse ntfs3 cls_bpf act_bpf sch_tbf sch_ingress sch_htb em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_basic act_skbedit
<4>[ 4507.687935]  act_mirred act_gact sg hid evdev ledtrig_usbport ledtrig_gpio cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit jool jool_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos bonding ip6_gre ip_gre gre ifb nat46 sit ip6_tunnel netlink_diag tunnel6 tunnel4 veth tun ovpn_dco nbd minix isofs hfsplus hfs cdrom cifs cifs_md4 cifs_arc4 autofs4 dns_resolver nls_koi8_r nls_iso8859_6 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp936 nls_cp866 nls_cp864 nls_cp862 nls_cp852 nls_cp850 nls_cp775 nls_cp437 nls_cp1251 nls_cp1250 zram zsmalloc algif_skcipher algif_rng algif_hash algif_aead af_alg des_generic arc4 leds_gpio xhci_plat_hcd
<4>[ 4507.775454]  xhci_pci xhci_mtk_hcd xhci_hcd ahci libahci libata loop gpio_button_hotplug xfs vfat fat jfs f2fs ext4 mbcache jbd2 exfat btrfs xor raid6_pq dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax mii mq_deadline kyber_iosched bfq crc32_generic
<4>[ 4507.884374] Process napi/phy1-10 (pid: 1564, threadinfo=ff92fdb4, task=10fc4733, tls=00000000)
<4>[ 4507.893025] Stack : 865ffe78 865ffed8 842ad1d0 842a8484 0cab8840 00000001 0cab8804 00000000
<4>[ 4507.901457]         00000000 0000162d 00000000 0000013b 00000000 00000000 00000000 0000376c
<4>[ 4507.909848]         00000000 00000000 00000000 00000000 00000000 000000fe 227a68f9 00000000
<4>[ 4507.918249]         00000000 842ad190 842ad2e0 8a186100 8437bbd8 842ad190 842ad2e0 84280418
<4>[ 4507.926652]         00007530 807976f0 000005a8 87f3cb00 00000000 00000000 84d63074 85f35180
<4>[ 4507.935052]         ...
<4>[ 4507.937511] Call Trace:
<4>[ 4507.937595] [<842a8484>] 0x842a8484 [nf_tables@24b7d93c+0x2eb50]
<4>[ 4507.946257] [<84280418>] 0x84280418 [nf_tables@24b7d93c+0x2eb50]
<4>[ 4507.952299] [<807976f0>] 0x807976f0
<4>[ 4507.955800] [<807e0a48>] 0x807e0a48
<4>[ 4507.959355] [<807da8c8>] 0x807da8c8
<4>[ 4507.962856] [<807e87f4>] 0x807e87f4
<4>[ 4507.966360] [<807ebeb4>] 0x807ebeb4
<4>[ 4507.970006] [<807a643c>] 0x807a643c
[... cut ...]
<4>[ 4508.657624] [<800391dc>] 0x800391dc
<4>[ 4508.661131] [<843746b4>] 0x843746b4 [mt7615e@05a12c55+0x2500]
<4>[ 4508.666899] [<800392f0>] 0x800392f0
<4>[ 4508.670406] [<806e371c>] 0x806e371c
<4>[ 4508.673891] [<806e3728>] 0x806e3728
<4>[ 4508.677489] [<806e3538>] 0x806e3538
<4>[ 4508.681013] [<800593d4>] 0x800593d4
<4>[ 4508.684493] [<80059210>] 0x80059210
<4>[ 4508.688024] [<80059210>] 0x80059210
<4>[ 4508.691522] [<80003518>] 0x80003518
<4>[ 4508.695016] 
<4>[ 4508.696504] ---[ end trace 0b6348fc395fb148 ]---
</pre>
</details>

nledovskikh, 2023-04-20:
Is there any progress or still panic?

rany2, 2023-06-17:
@nledovskikh panic remains on SNAPSHOT still...

nledovskikh, 2023-07-27:
It seems that the issue may be related to software cache coherency on this cpu.

nledovskikh, 2023-09-26:
> @nledovskikh If it helps, i can't trigger this anymore on SNAPSHOT. Perhaps the recent changes inadvertently fixed it or (more likely) made this harder to trigger.

Still happens, as before. And many interrupt's errors (spurious interrupts) in /proc/interrupts.
But the same modified code of ath9k on other systems work fine. Tested on x86-64 and arm64 (bpi-r64, bpi-r3) that have hard dma coherency, not soft.

dvn0, 2024-03-02:
I'm experiencing this as well, with a gnubee2 device, and what's odd is that it just seemed to crop up one day, and now it happens frequently. Network usage patterns didn't change... Just crashes under moderate sustained network load now.

Yoris1, 2026-02-12:
Hi, @dvn0
Do you still experience this issue? Would you be able to attach some logs and reproduction steps?

dengxinfa, 2026-07-25:
> It seems mt7621 have some cpu cache related errata (for example er35) that leads to such behaviour.

@nledovskikh  Do you have an ER35 patch that works successfully in OpenWRT 22.03 or OpenWRT 21.02? I got it from mtk-openwrt-feed, but it doesn't actually work.

nledovskikh, 2026-08-20:
> > It seems mt7621 have some cpu cache related errata (for example er35) that leads to such behaviour.
> 
> [@nledovskikh](https://github.com/nledovskikh) Do you have an ER35 patch that works successfully in OpenWRT 22.03 or OpenWRT 21.02? I got it from mtk-openwrt-feed, but it doesn't actually work.
Hi, sorry for late reply.
It's not working with 22.03 and later, device is not booting.
I think it was created for older kernels and openwrt 21.02.
It's not working with 21.02 also?