openwrt/openwrt #9566: ramips/mt7621: internal switch acts as a hub

sfan5, 2022-03-27:
OpenWrt snapshot built from 2c26eb4e484fe6d7a8159a22f97b6db7ca6e9221 running on ASUS RT-AC85P

Brief setup description: PC and NAS (`00:1e:06:xx:xx:xx`) each plugged into one port of the router (`d4:5d:64:xx:xx:xx`)

NAS has a lot of outgoing traffic, it is noticeable that all Ethernet LEDs always blink at the same time.
tcpdump on PC confirms issue:
<pre>
username@pc ~ $ sudo tcpdump -nn -i eno1 -ev  
tcpdump: listening on eno1, link-type EN10MB (Ethernet), snapshot length 262144 bytes
17:52:44.073175 00:1e:06:xx:xx:xx > d4:5d:64:xx:xx:xx, ethertype IPv4 (0x0800), length 1444: (tos 0x20, ttl 64, id 19018, offset 0, flags [DF], proto UDP (17), length 1430)
    192.168.2.101.6881 > 47.185.211.212.45940: UDP, length 1402
17:52:44.073210 00:1e:06:xx:xx:xx > d4:5d:64:xx:xx:xx, ethertype IPv4 (0x0800), length 1444: (tos 0x20, ttl 64, id 19019, offset 0, flags [DF], proto UDP (17), length 1430)
    192.168.2.101.6881 > 47.185.211.212.45940: UDP, length 1402
17:52:44.076698 00:1e:06:xx:xx:xx > d4:5d:64:xx:xx:xx, ethertype IPv4 (0x0800), length 1444: (tos 0x20, ttl 64, id 19020, offset 0, flags [DF], proto UDP (17), length 1430)
    192.168.2.101.6881 > 47.185.211.212.45940: UDP, length 1402
17:52:44.076733 00:1e:06:xx:xx:xx > d4:5d:64:xx:xx:xx, ethertype IPv4 (0x0800), length 1444: (tos 0x20, ttl 64, id 19021, offset 0, flags [DF], proto UDP (17), length 1430)
    192.168.2.101.6881 > 47.185.211.212.45940: UDP, length 1402
</pre>

I checked `brctl` on device and found nothing wrong, return traffic is also *not* affected. My guess is that a kernel driver is just configuring something wrong.
(No such issue on OpenWrt 21.02.2, obviously.)

Matthaiks, 2022-03-31:
It seems to be related to this bug: https://github.com/openwrt/openwrt/issues/9420

quarkysg, 2022-04-02:
@sfan5 would you be interested to help test out my patch as explained in this [post](https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/1916?u=quarky)?

I would very much be interested to know how it behaves for mt7621 routers with mt7530 switch.

Vladdrako, 2022-04-03:
@quarkysg I applied it and have no problems. The link speed issue is fixed. But I'm not familiar with switch configuration, can you show an example of how to config it properly?
My config right now:
<details>
<pre>
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fda0:62d6:83aa::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	option stp '0'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '64'

config interface 'wan'
	option device 'wan'
	option proto 'static'
	option ipaddr ''
	option netmask '255.255.224.0'
	option gateway ''
	option force_link '0'
	option metric '1'
	option peerdns '0'
	option ipv6 'off'

config interface 'wan6'
	option proto '6in4'
	option ipaddr ''
	option peeraddr ''
	option ip6addr ''
	list ip6prefix ''

config interface 'wg0'#client
	option proto 'wireguard'
	option listen_port '51820'
	option peerdns '0'
	option private_key ''
	list addresses '10.6.0.2/24'
	list addresses ''

config wireguard_wg0
	option persistent_keepalive '25'
	option endpoint_port '51820'
	option description 'OracleVPS'
	option public_key ''
	option preshared_key ''
	option endpoint_host ''
	list allowed_ips '0.0.0.0/0'
	list allowed_ips '::0/0'

config rule
	option priority '100'
	option lookup 'vpn'
	option mark '0x1'

config rule6
	option priority '100'
	option lookup 'vpn'
	option mark '0x1'

config route6
	option interface 'wan6'
	option target '::/0'

config interface 'wg1'#server
	option proto 'wireguard'
	option private_key ''
	option listen_port '9999'
	list addresses '10.14.0.1/29'

config wireguard_wg1
	option persistent_keepalive '25'
	option description 'Redmi Note 7'
	option public_key ''
	option route_allowed_ips '1'
	list allowed_ips '10.14.0.2/32'

config interface 'TetheringWAN'
	option proto 'dhcp'
	option device 'usb0'
</pre>
</details>
Also, I have no warnings you described in your post in the logs.

quarkysg, 2022-04-04:
@Vladdrako It's great that the patch it working for you.

I guess there's no right or wrong way to configure the switch via UCI.  Here's what was configured for my E8450 as an AP:

```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdc3:0a62:ed18::/48'
        option packet_steering '1'

config interface 'lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option delegate '0'
        option ipaddr '192.168.1.2'
        option gateway '192.168.1.1'
        list dns '192.168.1.1'
        option device 'sw0.1'

config interface 'iptv'
        option proto 'none'
        option delegate '0'
        option device 'sw0.2'

config device
        option type 'bridge'
        option name 'sw0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'tap0'
        list ports 'wan'

config bridge-vlan
        option device 'sw0'
        option vlan '1'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'wan:u*'

config bridge-vlan
        option device 'sw0'
        option vlan '2'
        list ports 'lan4:u*'
        list ports 'tap0:u*'
```
The warning I was referring to only appears if you try to configure a switch port as untagged for multiple VLANs and also set the PVID for more than 1 untagged VLAN.  Your configurations does not satisfy this condition, so it will not show.

sfan5, 2022-06-07:
Finally had a change to test this again (on commit f5a87a0a7b8d5aa83435868eb324601df6c4f4dc) and the issue is fixed.