openwrt/openwrt #9570: urandom-seed seeding "broken" since Linux 4.8

champtar, 2022-03-28:
Thanks to an LWN article (go subscribe now), I stumbled upon [this email](https://lwn.net/ml/linux-kernel/YjoTJFRook+rGyDI@zx2c4.com/) from @zx2c4 and continued to read some LKML threads.

In short `cat seedfile > /dev/urandom` doesn't work as everyone think it does, it doesn't influence immediately `/dev/urandom`. [That might actually be a "regression" since Linux 4.8](https://lwn.net/ml/linux-kernel/CAHmME9p7k2Z2f3aYctHxV9oNwe_GKd62Sghh9Ck1-nRyPaEypA@mail.gmail.com/).

Despite being what is done by all distros since forever,
being what is in the [man page](https://man7.org/linux/man-pages/man4/random.4.html) (but [will be fixed](https://lwn.net/ml/linux-kernel/CAHmME9oMeBqEa_GtGXDDVrVe+N=nJH8zB9APDb-mUxx+3_zouw@mail.gmail.com/)),
and being [what systemd does when it doesn't want to credit](https://github.com/systemd/systemd/blob/ffa047a03e4c5f6bd3af73b7eecb99cd230fe204/src/basic/random-util.c#L273),
It seems `cat seedfile > /dev/urandom` should not be used anymore and we should switch to using RNDADDENTROPY ioctl (or @zx2c4 [seedrng](https://lwn.net/ml/linux-kernel/CAHmME9ovJpdcuuZhNKrOTUc8XvKDDdC+axhAmOD9iESnRR7JqA@mail.gmail.com/))

I'm unclear if using `RNDADDENTROPY` and not crediting is enough to influence immediately `/dev/urandom` output or if we must credit for kernels as old as 5.4 (OpenWrt 21.02) and if in the future we will need to credit or not.

If anyone want to review what urandom-seed is doing right now:
- [code injecting the seed](https://github.com/openwrt/openwrt/blob/master/package/system/urandom-seed/files/lib/preinit/81_urandom_seed)
- [code saving the seed](https://github.com/openwrt/openwrt/blob/master/package/system/urandom-seed/files/sbin/urandom_seed
 )
- [getrandom utils source](https://git.openwrt.org/?p=project/ubox.git;a=blob;f=getrandom.c;h=ad3a7afde117a8098438e89f1f6e38722c3a5096;hb=HEAD)

in short:
- `cat seedfile > /dev/urandom` as early as possible
- `[ -f seedfile ] || getrandom 512 > seedfile.tmp ...` in a service, writing the seed when `getrandom()` unblock

We do not renew the seedfile by default to avoid wearing the flash too much, but you can opt-in with `system.@system[0].urandom_seed` setting

Side note: OpenWrt also has a timing jitter daemon called urngd installed by default on most targets ([Makefile](https://github.com/openwrt/openwrt/blob/master/package/system/urngd/Makefile) / [sources](https://git.openwrt.org/?p=project/urngd.git;a=tree))

zx2c4, 2022-03-28:
I'm preparing a patch for this. Standby.

zx2c4, 2022-03-28:
I posted a patch to fix this issue here: 
http://lists.openwrt.org/pipermail/openwrt-devel/2022-March/038338.html

closed 2022-03-28 by jow-

ynezz, 2022-03-28:
> I posted a patch to fix this issue here: http://lists.openwrt.org/pipermail/openwrt-devel/2022-March/038338.html

Thank you for the patch!