openwrt/openwrt #9576: block mount is unable to mount btrfs subvolumes

M95D, 2022-03-28:
Block mount is unable to mount more than one btrfs subvolume.

To reproduce this bug:
- install btfs and tools
- format a USB stick with btrfs
- create two subvolumes (my example uses "log" and "db")
- try to mount them from /etc/config/fstab:
```config mount
        option label 'USBStick'
        option target '/var/log'
        option fstype 'btrfs'
        option options 'subvol=log'
        option enabled '1'

config mount
        option label 'USBStick'
        option target '/var/db'
        option fstype 'btrfs'
        option options 'subvol=db'
        option enabled '1'
```
The result is that one subvolume is mounted, but the second subvolume fails with this error:
```daemon.err block: /dev/sda is already mounted on /tmp/db```

The error seems to be in block.c
```
        m = find_block(pr->uuid, pr->label, device, NULL);
        if (m && m->extroot)
                return -1;

        mp = find_mount_point(pr->dev);
        if (mp) {
                if (m && m->type == TYPE_MOUNT && m->target && strcmp(m->target, mp)) {     # <--- HERE
                        ULOG_ERR("%s is already mounted on %s\n", pr->dev, mp);
                        err = -1;
                } else
                        err = 0;
                free(mp);
                return err;
        }
```

neheb, 2022-03-29:
openwrt tools accept patches on the mailing list.

M95D, 2022-03-29:
I can't fix that. My C++ is not that good. I can remove it, but I don't know if that is the correct thing to do.

As I understand the code, that line checks if the same device (/dev/xxx) is already mounted somewhere.
- Why not allow mounting the same device again?
- Why check anything instead of letting mount try and possibly fail and reporting that error?

neheb, 2022-03-29:
Because that’s not something smart to do with most file systems. Byrd’s is different though.

crass, 2022-11-06:
> Because that’s not something smart to do with most file systems. btrfs is different though.

I'm questioning this logic, it may at one time have been true, but I don't think it is now. I presume its "not something smart to do" because most linux filesystems aren't written to handle isolated simultaneous mounts. So yes that shouldn't be done, eg. mount an EXT4 block device on two virtual machines simultaneously. However, mounting the same block device on different locations using the same running kernel, at least in theory should be safe, and if not it would be a kernel bug if the second mount was allowed by the kernel but caused corruption.

In fact, I've tested on OpenWRT 22.03.2 that mounting the same block device to two different locations at the same time is allowed by the kernel. I tested this with vfat, xfs, and ubifs filesystems. I also have yet to encounter any kind of corruption. I suspect the kernel recognizes that the filesystem is already opened and internally converts the second mount to a kind of bind mount.

So I think the question of whether this check is still needed should be revisited.

Of course, if my initial presumption was not what you were thinking. Please elaborate further.

closed 2024-12-04 by M95D

ryjelsum, 2025-12-19:
This issue shouldn't be closed, by my reckoning - it is still present as of OpenWRT 24.10.4 and I do not believe it is a duplicate of the other btrfs subvolume related issue mentioned (#12308). See [my comment on that issue](https://github.com/openwrt/openwrt/issues/12308#issuecomment-3673739978) for more about how these two issues present and inter-relate.

M95D, 2025-12-19:
As you may have noticed, this issue was open for 2.5 **years**. You either fix it yourself in your own fork of OpenWrt or abandon it entirely, 'cause nobody's gonna fix here, that's for sure.