openwrt/openwrt #9583: kernel 5.15: mt76 driver is broken

Vladdrako, 2022-03-29:
Device: Xiaomi Mi Router 3 Pro (mt7621+mt7615e)
```
[   40.579993] mt7615e 0000:01:00.0: of_irq_parse_pci: failed with rc=-14
[   40.598192] mt7615e 0000:01:00.0: assign IRQ: got 0
[   40.598308] pci 0000:00:00.0: enabling bus mastering
[   40.598355] mt7615e 0000:01:00.0: enabling device (0000 -> 0002)
[   40.604558] mt7615e 0000:01:00.0: enabling bus mastering
[   40.604640] mt7615e: probe of 0000:01:00.0 failed with error -22
[   40.610848] mt7615e 0000:02:00.0: of_irq_parse_pci: failed with rc=-14
[   40.617387] mt7615e 0000:02:00.0: assign IRQ: got 0
[   40.617427] pci 0000:00:01.0: enabling bus mastering
[   40.617441] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
[   40.623517] mt7615e 0000:02:00.0: enabling bus mastering
[   40.623553] mt7615e: probe of 0000:02:00.0 failed with error -22
```
Complete dmesg
<details>
<pre>
[    0.000000] Linux version 5.15.31 (vladdrako@Vladdrako-PC) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20628-6f6ca59a6e) 11.2.0, GNU ld (GNU Binutils) 2.38) #0 SMP PREEMPT Tue Mar 29 04:24:27 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3 Pro
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s16368 r8192 d20496 u45056
[    0.000000] pcpu-alloc: s16368 r8192 d20496 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=0001c000
[    0.000000] Readback ErrCtl register=0001c000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 504096K/524288K available (9380K kernel code, 707K rwdata, 2364K rodata, 1292K init, 278K bss, 20192K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.007984] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066214] pid_max: default: 32768 minimum: 301
[    0.070963] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078172] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.088736] rcu: Hierarchical SRCU implementation.
[    0.094158] smp: Bringing up secondary CPUs ...
[    0.099506] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.099531] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.099546] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.099592] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.158906] Synchronize counters for CPU 1: done.
[    0.190983] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.191003] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.191014] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.191043] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.250141] Synchronize counters for CPU 2: done.
[... 667 lines cut ...]
[   99.069022] usb 1-1.1.1: SerialNumber: 004-0D9F-000
[   99.170760] hid-generic 0003:0D9F:0004.0009: hiddev0,hidraw0: USB HID v1.00 Device [POWERCOM Co.,LTD HID UPS Battery] on usb-1e1c0000.xhci-1.1.1/input0
[  100.675343] I/O scheduler: non-rotational device 'mtdblock6' was '[mq-deadline] kyber bfq none', now 'mq-deadline [kyber] bfq none'
[  102.739194] usb 1-1.1.1: USB disconnect, device number 18
[  102.767005] xhci-mtk 1e1c0000.xhci: drop_ep_quirk low-speed ep1in intr, mpkt:8, interval:100/100ms
[  103.858746] I/O scheduler: non-rotational device 'loop3' was '[none] mq-deadline kyber bfq ', now 'mq-deadline [kyber] bfq none'
[  105.721826] usb 1-1.1.1: new low-speed USB device number 19 using xhci-mtk
[  106.009740] usb 1-1.1.1: New USB device found, idVendor=0d9f, idProduct=0004, bcdDevice= 0.02
[  106.018605] usb 1-1.1.1: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  106.026746] usb 1-1.1.1: Product: HID UPS Battery
[  106.031594] usb 1-1.1.1: Manufacturer: POWERCOM Co.,LTD
[  106.042661] usb 1-1.1.1: SerialNumber: 004-0D9F-000
[  106.148334] hid-generic 0003:0D9F:0004.000A: hiddev0,hidraw0: USB HID v1.00 Device [POWERCOM Co.,LTD HID UPS Battery] on usb-1e1c0000.xhci-1.1.1/input0
[  108.628156] I/O scheduler: non-rotational device 'nbd5' was '[mq-deadline] kyber bfq none', now 'mq-deadline [kyber] bfq none'
[  109.641143] usb 1-1.1.1: USB disconnect, device number 19
[  109.662709] xhci-mtk 1e1c0000.xhci: drop_ep_quirk low-speed ep1in intr, mpkt:8, interval:100/100ms
[  112.711974] usb 1-1.1.1: new low-speed USB device number 20 using xhci-mtk
[  112.952968] usb 1-1.1.1: New USB device found, idVendor=0d9f, idProduct=0004, bcdDevice= 0.02
[  112.963587] usb 1-1.1.1: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  112.973871] usb 1-1.1.1: Product: HID UPS Battery
[  112.978709] usb 1-1.1.1: Manufacturer: POWERCOM Co.,LTD
[  112.996015] usb 1-1.1.1: SerialNumber: 004-0D9F-000
[  113.210633] hid-generic 0003:0D9F:0004.000B: hiddev0,hidraw0: USB HID v1.00 Device [POWERCOM Co.,LTD HID UPS Battery] on usb-1e1c0000.xhci-1.1.1/input0
</pre>
</details>

Vladdrako, 2022-03-30:
Ok, looks like I enabled too much in the kernel config. Now PCI works fine. dmesg from default config:
<details>
<pre>
[    0.000000] Linux version 5.15.31 (vladdrako@Vladdrako-PC) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20628-6f6ca59a6e) 11.2.0, GNU ld (GNU Binutils) 2.38) #0 SMP Wed Mar 30 04:43:15 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3 Pro
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s15568 r8192 d21296 u45056
[    0.000000] pcpu-alloc: s15568 r8192 d21296 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=0001c000
[    0.000000] Readback ErrCtl register=0001c000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 507268K/524288K available (7281K kernel code, 632K rwdata, 1468K rodata, 1228K init, 244K bss, 17020K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000005] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.007988] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066217] pid_max: default: 32768 minimum: 301
[    0.070964] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078173] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.088605] rcu: Hierarchical SRCU implementation.
[    0.093615] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.101648] smp: Bringing up secondary CPUs ...
[    0.106832] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.106856] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.106871] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.106918] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.161486] Synchronize counters for CPU 1: done.
[    0.193926] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.193948] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.193959] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.193986] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.252860] Synchronize counters for CPU 2: done.
[    0.283337] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.283357] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.283368] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.283400] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.338052] Synchronize counters for CPU 3: done.
[    0.367905] smp: Brought up 1 node, 4 CPUs
[    0.375522] random: get_random_u32 called from bucket_table_alloc.isra.0+0x70/0x180 with crng_init=0
[    0.376425] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.395252] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.402292] pinctrl core: initialized pinctrl subsystem
[    0.408306] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.415235] cpuidle: using governor teo
[    0.430369] FPU Affinity set after 11720 emulations
[    0.441892] clocksource: Switched to clocksource GIC
[... 484 lines cut ...]
           CPU0       CPU1       CPU2       CPU3       
  8:      21670      21665      21655      21648  MIPS GIC Local   1  timer
  9:     176161          0          0          0  MIPS GIC  63  IPI call
 10:          0      82241          0          0  MIPS GIC  64  IPI call
 11:          0          0     188580          0  MIPS GIC  65  IPI call
 12:          0          0          0      87277  MIPS GIC  66  IPI call
 13:      12777          0          0          0  MIPS GIC  67  IPI resched
 14:          0      15723          0          0  MIPS GIC  68  IPI resched
[... cut ...]
 23:         14          0          0          0  MIPS GIC  30  mt7530
 24:          3          0          0          0    mt7530   1  mt7530-0:01
 25:          3          0          0          0    mt7530   2  mt7530-0:02
 26:          5          0          0          0    mt7530   3  mt7530-0:03
 27:          3          0          0          0    mt7530   4  mt7530-0:04
 28:          0          0          0          0  1e000600.gpio  18  keys
ERR:          0
</pre>
</details>
But mt76 driver is still broken.
@nbd168 any suggestions?

Vladdrako, 2022-04-01:
> mt7530 driver is not properly assigning IRQs.

Same on 5.10. Maybe @LGA1150 can tell more.

> Which device tree are you using? I am using the one which was on staging and now is properly mainlined since interrupts have totally changed in 5.15.

5.15 added to the openwrt. a3764ee29dd09394fa750c78160e300d86532561

paraka, 2022-04-01:
I have just compiled openwrt's master branch to test in one of my devices and I get a "firmware failed to start" message from the mt76 driver:

[    0.000000] Linux version 5.15.31 (sergio@camaron) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19336-483fe539c4) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Mar 31 22:51:40 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is HiLink HLK-7621A evaluation board
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2} total 2
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s15504 r8192 d21360 u45056
[    0.000000] pcpu-alloc: s15504 r8192 d21360 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=0001dac0
[    0.000000] Readback ErrCtl register=0001dac0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 248872K/262144K available (7058K kernel code, 628K rwdata, 1436K rodata, 1232K init, 242K bss, 13272K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x554/0x76c with crng_init=0
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000003] sched_clock: 64 bits at 1000MHz, resolution 1ns, wraps every 4398046511103ns
[    0.016106] Calibrating delay loop... 666.41 BogoMIPS (lpj=3332096)
[    0.088409] pid_max: default: 32768 minimum: 301
[    0.097682] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.112093] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.129717] rcu: Hierarchical SRCU implementation.
[    0.139424] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.154754] smp: Bringing up secondary CPUs ...
[    0.164163] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.164185] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.164198] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.164240] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.214679] Synchronize counters for CPU 1: done.
[    0.274352] smp: Brought up 1 node, 2 CPUs
[    0.285820] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.305303] futex hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.318900] pinctrl core: initialized pinctrl subsystem
[    0.329976] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.342563] cpuidle: using governor teo
[    0.359555] FPU Affinity set after 13320 emulations
[    0.369517] clocksource: Switched to clocksource GIC
[    0.380083] NET: Registered PF_INET protocol family
[    0.389818] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.404688] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.421223] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.436411] TCP bind hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.450587] TCP: Hash tables configured (established 2048 bind 2048)
[    0.463291] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.476180] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.490208] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.501377] PCI: CLS 0 bytes, default 32
[... 153 lines cut ...]
[   42.407480] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[   42.475059] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   42.491143] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   42.509421] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   42.526377] mt7530 mdio-bus:1f lan1: configuring for phy/gmii link mode
[   42.546660] br-lan: port 1(lan1) entered blocking state
[   42.557166] br-lan: port 1(lan1) entered disabled state
[... cut ...]
[   42.760266] device lan3 entered promiscuous mode
[   42.793978] mt7530 mdio-bus:1f lan4: configuring for phy/gmii link mode
[   42.823436] br-lan: port 4(lan4) entered blocking state
[   42.833956] br-lan: port 4(lan4) entered disabled state
[   42.856333] device lan4 entered promiscuous mode
[   42.908556] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   46.045274] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   46.074820] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[  108.870053] done.

Black6spdZ, 2022-04-08:
[   13.839991] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
[   21.530773] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[   21.562049] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   21.570367] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   21.574255] device eth0 entered promiscuous mode
[   21.584147] mt7530 mdio-bus:1f lan1: configuring for phy/gmii link mode
[   21.596428] 8021q: adding VLAN 0 to HW filter on device lan1
[   21.606152] br-lan: port 1(lan1) entered blocking state
[   21.611431] br-lan: port 1(lan1) entered disabled state
[   21.619446] device lan1 entered promiscuous mode
[   21.646352] mt7530 mdio-bus:1f lan2: configuring for phy/gmii link mode
[   21.655607] 8021q: adding VLAN 0 to HW filter on device lan2
[   21.668379] br-lan: port 2(lan2) entered blocking state
[   21.673670] br-lan: port 2(lan2) entered disabled state
[   21.680734] device lan2 entered promiscuous mode
[   21.694029] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[   21.701989] 8021q: adding VLAN 0 to HW filter on device lan3
[   21.712243] br-lan: port 3(lan3) entered blocking state
[   21.717594] br-lan: port 3(lan3) entered disabled state
[   21.725631] device lan3 entered promiscuous mode
[   21.739588] mt7530 mdio-bus:1f lan4: configuring for phy/gmii link mode
[   21.746816] 8021q: adding VLAN 0 to HW filter on device lan4
[   21.757205] br-lan: port 4(lan4) entered blocking state
[   21.762482] br-lan: port 4(lan4) entered disabled state
[   21.770822] device lan4 entered promiscuous mode
[   21.787345] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   21.794661] 8021q: adding VLAN 0 to HW filter on device wan
[   21.808520] br-lan: port 5(wan) entered blocking state
[   21.813722] br-lan: port 5(wan) entered disabled state
[   21.822190] device wan entered promiscuous mode
[   23.253454] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[   24.550420] br-lan: port 2(lan2) entered blocking state
[   24.555744] br-lan: port 2(lan2) entered forwarding state
[   24.563647] br-lan: port 6(wlan0) entered blocking state
[   24.569039] br-lan: port 6(wlan0) entered disabled state
[   24.574966] device wlan0 entered promiscuous mode
[   24.609350] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[   24.616992] br-lan: port 3(lan3) entered blocking state
[   24.622233] br-lan: port 3(lan3) entered forwarding state
[   24.738541] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   24.746043] br-lan: port 5(wan) entered blocking state
[   24.751166] br-lan: port 5(wan) entered forwarding state
[   25.506046] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   25.512744] br-lan: port 6(wlan0) entered blocking state
[   25.518345] br-lan: port 6(wlan0) entered forwarding state
[   27.076882] br-lan: port 7(wlan1) entered blocking state
[   27.082218] br-lan: port 7(wlan1) entered disabled state
[   27.088383] device wlan1 entered promiscuous mode
[   29.311317] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   29.318004] br-lan: port 7(wlan1) entered blocking state
[   29.323320] br-lan: port 7(wlan1) entered forwarding state
[ 2734.309186] mt7530 mdio-bus:1f lan2: Link is Down
[ 2734.314374] br-lan: port 2(lan2) entered disabled state
[43035.688348] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[43035.696102] br-lan: port 2(lan2) entered blocking state
[43035.701346] br-lan: port 2(lan2) entered forwarding state
[43039.002610] mt7530 mdio-bus:1f lan2: Link is Down
[43039.007851] br-lan: port 2(lan2) entered disabled state
[43040.678161] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[43040.686123] br-lan: port 2(lan2) entered blocking state
[43040.691361] br-lan: port 2(lan2) entered forwarding state
[93047.300528] mt7530 mdio-bus:1f lan2: Link is Down
[93047.305687] br-lan: port 2(lan2) entered disabled state
[128819.971114] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[128819.978975] br-lan: port 2(lan2) entered blocking state
[128819.984376] br-lan: port 2(lan2) entered forwarding state
[128823.407330] mt7530 mdio-bus:1f lan2: Link is Down
[128823.412856] br-lan: port 2(lan2) entered disabled state
[128825.080793] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[128825.088413] br-lan: port 2(lan2) entered blocking state
[... 206 lines cut ...]
[630650.053972] mt7530 mdio-bus:1f lan2: Link is Down
[630650.059390] br-lan: port 2(lan2) entered disabled state
[630651.661631] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[630651.669594] br-lan: port 2(lan2) entered blocking state
[630651.675337] br-lan: port 2(lan2) entered forwarding state
[646692.633615] mt7530 mdio-bus:1f lan2: Link is Down
[646692.639446] br-lan: port 2(lan2) entered disabled state
[669714.900922] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[669714.908713] br-lan: port 2(lan2) entered blocking state
[669714.914052] br-lan: port 2(lan2) entered forwarding state
[669717.224725] mt7530 mdio-bus:1f lan2: Link is Down
[669717.230526] br-lan: port 2(lan2) entered disabled state
[669718.895853] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[669718.903730] br-lan: port 2(lan2) entered blocking state
[669718.909565] br-lan: port 2(lan2) entered forwarding state
[... cut ...]
[731081.630301] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[731081.638232] br-lan: port 2(lan2) entered blocking state
[731081.643985] br-lan: port 2(lan2) entered forwarding state
[731084.929792] mt7530 mdio-bus:1f lan2: Link is Down
[731084.935375] br-lan: port 2(lan2) entered disabled state
[731086.745157] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[731086.752921] br-lan: port 2(lan2) entered blocking state
[731086.758710] br-lan: port 2(lan2) entered forwarding state

Vladdrako, 2022-04-09:
I wonder if devices on kernel 5.15 should assign the same IRQs as on 5.10?
5.10
```
mt7530 mdio-bus:1f lan3 (uninitialized): PHY [mt7530-0:01] driver [MediaTek MT7530 PHY] (irq=27)
mt7530 mdio-bus:1f lan2 (uninitialized): PHY [mt7530-0:02] driver [MediaTek MT7530 PHY] (irq=28)
mt7530 mdio-bus:1f lan1 (uninitialized): PHY [mt7530-0:03] driver [MediaTek MT7530 PHY] (irq=29)
mt7530 mdio-bus:1f wan (uninitialized): PHY [mt7530-0:04] driver [MediaTek MT7530 PHY] (irq=30)
Wi-Fi
mt7621-pci 1e140000.pcie: bus=0 slot=0 irq=23
mt7621-pci 1e140000.pcie: bus=0 slot=1 irq=24
```
5.15
```
mt7530 mdio-bus:1f lan3 (uninitialized): PHY [mt7530-0:01] driver [MediaTek MT7530 PHY] (irq=24)
mt7530 mdio-bus:1f lan2 (uninitialized): PHY [mt7530-0:02] driver [MediaTek MT7530 PHY] (irq=25)
mt7530 mdio-bus:1f lan1 (uninitialized): PHY [mt7530-0:03] driver [MediaTek MT7530 PHY] (irq=26)
mt7530 mdio-bus:1f wan (uninitialized): PHY [mt7530-0:04] driver [MediaTek MT7530 PHY] (irq=27)
```
The rest IRQs match.

closed 2022-04-09 by Vladdrako

justinkb, 2022-04-16:
is this fixed? if not, why was this issue closed?

Vladdrako, 2022-04-16:
@justinkb yes, it was my local problem (corrupted mt7621.dtsi).