openwrt/openwrt #9584: [21.02] octeon/generic: network silently fails

PolynomialDivision, 2022-03-29:
The network freezed silently on a EdgeRouter 4 with OpenWrt:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02-SNAPSHOT'
DISTRIB_REVISION='r16485-59e7ae8d65'
DISTRIB_TARGET='octeon/generic'
DISTRIB_ARCH='mips64_octeonplus'
DISTRIB_DESCRIPTION='OpenWrt 21.02-SNAPSHOT r16485-59e7ae8d65'
DISTRIB_TAINTS=''
```

Dmesg shows this:
```
[24996.513121] Port 0 receive error code 4, packet dropped
[450778.378736] Port 0 receive error code 4, packet dropped
[560582.300270] Port 0 receive error code 4, packet dropped
[599425.012343] Port 0 receive error code 4, packet dropped
[832688.044884] Port 0 receive error code 4, packet dropped
```

Documentation says:
> 4 = oversize error: the RGMII packet is longer than allowed.

The device will not notice anything. `ip a` will not show anything. Ping will produce this output:
```
PING xxx (xxx): 56 data bytes

--- xxx ping statistics ---
3 packets transmitted, 0 packets received, 100% packet loss
```
Any idea?

markmn123, 2022-08-05:
Any update on this issue? Was looking at info regarding flashing my ER-4 with openWRT but cant find any success stories or how people like it

shvetsnikita, 2026-11-05:
Still reproducible on 25.12.5, triggered by link saturation via not-very-reliable cable.

Workaround: install luci-app-sqm and set ingress, egress 900000, discipline fq_codel, script simplest_tbf.qos for affected port.