openwrt/openwrt #9585: FW4 (nft) logging zone option log '1' has no affect

clutch2sft, 2022-03-29:
OpenWrt SNAPSHOT, r19293-1c7e146f7e

Setting zone log via the LUCI interface inserts option log '1' into /etc/config/firewall but no firewall logs are generated.

I read some of the nft manual and nothing in the nft output would convince me that the setting in /etc/config/firewall has inserted log instructions to the actual fw4 configuration.

Can happily be proven wrong if you are better informed than I.

dave14305, 2022-03-31:
Also discussed on the forums:

https://forum.openwrt.org/t/nftables-logging/123995

clutch2sft, 2022-03-31:
Thank you for the reference.  

I had found that thread at one point.  Though for my knowledge level it wasn't clearly related since I was executing the zone logging feature from within LUCI and they were discussing manipulation at the command line via nft.

I'll re-read all that and apply the suggestions to see if it address the challenge presented to me.

clutch2sft, 2022-03-31:
Not so much for me.  I'll grab today's snapshot see if it is different.  @jow- can you share the output of your banner?

root@OpenWrt:~# uci set firewall.@zone[1].log=1
root@OpenWrt:~# fw4 reload
/proc/self/fd/0:119:25-27: Error: Could not process rule: No such file or directory

/proc/self/fd/0:123:25-27: Error: Could not process rule: No such file or directory

root@OpenWrt:~#

jow-, 2022-03-31:
Yes, I do have kmod-ipt-core and kmod-ip6tables installed.

dave14305, 2022-03-31:
> kmod-ip6tables

Wouldn’t this be non-default in a base image with firewall4? I only have kmod-ipt-core, for example.

jow-, 2022-03-31:
Yeah likely, the kmod packaging has to be revised

clutch2sft, 2022-04-05:
Hi - Just wanted to come back to this for closure on my comments.  Thank you for the confirmation of a packaging bug.  I wanted to go into 22.03 and verify same... took me a couple days to learn enough to build that.   Sure enough same.  Base install has no logging capability and after installing four packages (two installed as dependencies if you just install  kmod-ipt-core and kmod-ip6tables) then logging works as expected.