openwrt/openwrt #9594: RTC modules not enabled for Raspberry Pi

build2stone, 2022-03-30:
Is there a reason for this?
They appear to be enabled in the official kernel config:
https://github.com/raspberrypi/linux/blob/5830759fc2591a3c6aad1783f14ad6efaa904a1a/arch/arm/configs/bcmrpi_defconfig#L1212
``` diff
diff --git a/target/linux/bcm27xx/bcm2708/config-5.10 b/target/linux/bcm27xx/bcm2708/config-5.10
index 56e076ba95..133e08bf5d 100644
--- a/target/linux/bcm27xx/bcm2708/config-5.10
+++ b/target/linux/bcm27xx/bcm2708/config-5.10
@@ -301,6 +301,9 @@ CONFIG_RESET_SIMPLE=y
 # CONFIG_RPIVID_MEM is not set
 # CONFIG_RPI_AXIPERF is not set
 # CONFIG_RPI_POE_POWER is not set
+CONFIG_RTC_CLASS=y
+CONFIG_RTC_I2C_AND_SPI=y
+CONFIG_RTC_MC146818_LIB=y
 CONFIG_SCSI=y
 # CONFIG_SCSI_LOWLEVEL is not set
 # CONFIG_SCSI_PROC_FS is not set
```

pkgadd, 2022-03-30:
Why would you enable them?

The RPi might have an RTC, but not a battery backed RTC - which is worse than not having any at all (so removing the drivers makes sense). 

Why is it worse, simply because the RTC will claim with authority its time, although it doesn't have the slightest clue about the correct time after a power-cycle (that's where the battery backup is severely needed) - breaking the normal mitigations to step up the time quickly.

build2stone, 2022-03-31:
I'm trying to use an external battery-backed RTC module, similar to the one in [this](https://thepihut.com/blogs/raspberry-pi-tutorials/17209332-adding-a-real-time-clock-to-your-raspberry-pi) article, but the drivers are missing.
I was not aware of the RPi-internal RTC, but apparently it does not interfere with the function of these RTC modules.
Just building the modules wouldn't change anything if they're not installed, right?

pepe2k, 2022-04-03:
@build2stone @pkgadd I think the problem here is that the `bcm27xx` target doesn't include `rtc` within its `FEATURES`, here:
https://github.com/openwrt/openwrt/blob/master/target/linux/bcm27xx/Makefile#L11

This results in missing `CONFIG_RTC_CLASS=y` in kernel config for pre-built images and no support for RTC at all (thus no kmod-rtc-* for download).

build2stone, 2022-04-04:
Ah, that makes much more sense!
Sorry about the misleading issue, as you can tell I am still quite unfamiliar with OpenWrt internals.

closed 2022-04-07 by jow-

takumiando, 2022-07-27:
Will this fix be backported to ``21.02`` series?

commit ff09905a (2022-04-07): bcm27xx: include 'rtc' in target's 'FEATURES'

There are many ways to add external RTC to Raspberry Pi boards. Let's
include support for this for the whole target and while at it, sort
features alphabetically.

Fixes: #9594
commit 6120a66e (2022-05-13): bcm27xx: include 'rtc' in target's 'FEATURES'

There are many ways to add external RTC to Raspberry Pi boards. Let's
include support for this for the whole target and while at it, sort
features alphabetically.

Fixes: #9594

(cherry picked from commit ff09905a468b4cc44f039a76568e8fe6cbaea8d9)