openwrt/openwrt #9596: Openwrt: Subtarget MT7628 based boards ???

Organarh, 2022-03-30:
Archer MR-200 V4 Mediatek - MT7628DAN RAM-64 MB Flash-8 MB CPU-580 MHz .
git clone https://github.com/openwrt/openwrt.git ...........
Target System Mediatek Ralink MIPS
Subtarget MT7628 based boards ???? https://ibb.co/wSdfmq9

981213, 2022-03-31:
The specific router you mentioned isn't supported in OpenWrt yet, which means there isn't an out-of-box option to select for your router.

MT7628DAN is just a MT7628AN with 64M integrated DDR2 ram. If you plan to add support for the router it should be added under mt76x8 subtarget.

Anyway this isn't a bug in OpenWrt. I'll close this issue.

closed 2022-03-31 by 981213

Organarh, 2022-03-31:
Hello, can I build openwrt myself? So: Target System Mediatek Ralink MIPS + Subtarget MT76x8 based boards. I can install Archer C50 v3 or Archer C20 AC750 v4 instead of TP-Link Archer MR-200 V 4, they have MT7628AN processor as MR-200 V 4. Then add "telephony" package to support SIM card.
Can I do this, is it possible or is it not possible to do this???

981213, 2022-04-01:
No. In most cases you can't just flash a firmware for a different router.

Anyway this isn't the correct place for these questions. You should use the forum instead: https://forum.openwrt.org/