openwrt/openwrt #9598: odhcpd: Static leases with multiple MAC addresses get skipped when configuring odhcpd

ktetzlaff, 2022-03-30:
In https://openwrt.org/docs/guide-user/base-system/dhcp#static_leases, the /mac/ option of a DHCP _Static Lease_ is defined as:
```
mac | string | no | (none) | The hardware address(es) of this host, separated by spaces.
```

This explicitly allows to create leases with multiple MAC addresses. However, /odhcpd/ skips such /leases/ during configuration.

Here's the code fragment which handles MAC addresses of static leases in `set_lease(..)` (found in `config.c`):
``` c
	if ((c = tb[LEASE_ATTR_MAC]))
		if (!ether_aton_r(blobmsg_get_string(c), &l->mac))
			goto err;
```
As can be seen, the code doesn't have any special handling for the multiple MAC case. So `ether_aton_r(..)` returns `NULL` which triggers `goto err;` and the lease doesn't make it into /leases/.

alucryd, 2024-07-05:
Is this the same bug as this? https://forum.openwrt.org/t/dhcp-static-leases-stopped-working-properly/197380/2

ktetzlaff, 2024-07-05:
It's probably related, but in the case described above, there lease is really using multiple MAC addresses whereas in the linked issue, there is only one MAC address (which is then converted from a single item option to a list of one item). The original code for multiple MACs used a simple option value with spaces to separate multiple MAC addresses. Now, Luci seems to store MAC addresses in a list (which is probably the right idea) and if odhcpd has not been updated, it will not handle those list entries correctly.

YodaDaCoda, 2024-10-08:
> Now, if LuCi has been updated to store even single MAC addresses in a list (which is probably the right idea), the code in odhcpd probably needs an update, too.

I recently switched from dnsmasq to odhcpd and this tripped me up until i changed all the `list mac` to `option mac` in `/etc/config/dhcp`. LuCI does indeed store single mac addresses as a list when entries are created/updated.

larskotthoff, 2025-03-22:
This also just tripped me up...

closed 2025-09-18 by thess