openwrt/openwrt #9600: 2.4 GHz radio doesn't work on Raspberry Pi 4B

Aviators01, 2022-03-31:
I flashed the openwrt image to the microSD card and booted it up on the raspberry pi 4B. I didn't change anything other than set the admin password and change the region to CA for Canada. I next setup the 2.4GHz radio by entering in encryption password and leaving everything else at default, however, the 2.4 GHz wifi doesn't work and will not show up. It doesn't matter if I put it in N mode, legacy mode, or mess around with the channel width. I've peeked into the config files and tried the other tips people have suggested as well. This issue was in 21.02.01 as well.

In contrast, the 5GHz wifi radio works without any problems in both N mode and AC mode.

pepe2k, 2022-04-01:
@Aviators01 your report is too general, please provide more information, e.g. OpenWrt version, config files, dmesg output, etc.

Aviators01, 2022-04-03:
Sorry about that. Version is 21.02.2. This is the filename: openwrt-21.02.2-bcm27xx-bcm2711-rpi-4-ext4-factory.img . config files are as follows: 

-----------------------------------------
/etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd33:d326:f3ef::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
-----------------------------------------
/etc/config/system

config system
        option hostname 'OpenWrt'
        option ttylogin '0'
        option log_size '64'
        option urandom_seed '0'
        option zonename 'America/Edmonton'
        option timezone 'MST7MDT,M3.2.0,M11.1.0'
        option log_proto 'udp'
        option conloglevel '8'
        option cronloglevel '5'

config timeserver 'ntp'
        list server '0.openwrt.pool.ntp.org'
        list server '1.openwrt.pool.ntp.org'
        list server '2.openwrt.pool.ntp.org'
        list server '3.openwrt.pool.ntp.org'
-----------------------------------------
/etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option cell_density '0'
        option country 'CA'
        option channel 'auto'
        option htmode 'HT20'
        option band '2g'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key 'raspberrypi'
        option network 'lan'

Hopefully this helps. Sorry if I make a mistake. I'm still learning. I've used htmode HT40 as well but that doesn't do anything to fix the problem with the 2.4 GHz radio.

Aviators01, 2022-04-03:
Here's the kernel log:

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.179 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16495-bf0c965af0)) #0 SMP Wed Feb 16 20:29:10 2022
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 2061312
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12288 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000]   Normal zone: 16384 pages used for memmap
[    0.000000]   Normal zone: 1048576 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45336 r8192 d32488 u86016
[    0.000000] pcpu-alloc: s45336 r8192 d32488 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:E7:F3:C1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 7943840K/8245248K available (7102K kernel code, 432K rwdata, 1144K rodata, 512K init, 541K bss, 235872K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from 0xffffffc0108a0b98 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000006] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000154] Console: colour dummy device 80x25
[    0.000384] printk: console [tty1] enabled
[    0.000419] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000440] pid_max: default: 32768 minimum: 301
[    0.000700] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.000844] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.001426] cgroup: Disabling memory control group subsystem
[    0.002194] ASID allocator initialised with 32768 entries
[    0.002284] rcu: Hierarchical SRCU implementation.
[    0.002909] smp: Bringing up secondary CPUs ...
[    0.003297] Detected PIPT I-cache on CPU1
[    0.003345] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003790] Detected PIPT I-cache on CPU2
[    0.003823] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004295] Detected PIPT I-cache on CPU3
[    0.004329] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004400] smp: Brought up 1 node, 4 CPUs
[    0.004426] SMP: Total of 4 processors activated.
[    0.004439] CPU features: detected: 32-bit EL0 Support
[    0.004452] CPU features: detected: CRC32 instructions
[    0.004508] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004525] CPU: All CPU(s) started at EL2
[    0.004553] alternatives: patching kernel code
[    0.012669] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.012707] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.014440] pinctrl core: initialized pinctrl subsystem
[    0.015268] NET: Registered protocol family 16
[    0.016682] DMA: preallocated 1024 KiB pool for atomic allocations
[    0.017226] cpuidle: using governor ladder
[    0.017256] cpuidle: using governor menu
[    0.017515] Serial: AMBA PL011 UART driver
[    0.018956] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[... 160 lines cut ...]
[    6.910358] NET: Registered protocol family 24
[    6.930922] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.100606] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.119308] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:20:01 version 7.45.206 (r725000 CY) FWID 01-febaba43
[    7.159339] usbcore: registered new interface driver brcmfmac
[... cut ...]
[   10.533417] br-lan: port 2(wlan0) entered disabled state
[   10.539120] device wlan0 entered promiscuous mode
[   17.339714] device wlan0 left promiscuous mode
[   17.344250] br-lan: port 2(wlan0) entered disabled state
[ 1124.040545] br-lan: port 2(wlan0) entered blocking state
[ 1124.045897] br-lan: port 2(wlan0) entered disabled state
[ 1124.051339] device wlan0 entered promiscuous mode
[ 1130.867702] device wlan0 left promiscuous mode
[ 1130.872234] br-lan: port 2(wlan0) entered disabled state

Aviators01, 2022-04-03:
current setup is the raspberry pi connected to my laptop via ethernet. No other connections. I ssh'd in to get the config files, tried to setup logging via ssh and then just went the easier route and got the kernel log from the web UI.

pepe2k, 2022-04-03:
@Aviators01 please, try static channel number instead of `auto` and provide output from `logread` afterwards.

Aviators01, 2022-04-07:
Here it is. Did the change, then disabled, then enabled to no effect:
'''
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
Wed Mar 30 19:43:30 2022 kern.notice kernel: [    0.000000] Linux version 5.4.179 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16495-bf0c965af0)) #0 SMP Wed Feb 16 20:29:10 2022
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 2061312
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000]   DMA zone: 3792 pages used for memmap
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000]   DMA zone: 242688 pages, LIFO batch:63
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000]   DMA32 zone: 12288 pages used for memmap
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages used for memmap
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000]   Normal zone: 1048576 pages, LIFO batch:63
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s45336 r8192 d32488 u86016
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s45336 r8192 d32488 u86016 alloc=21*4096
Wed Mar 30 19:43:30 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] Detected PIPT I-cache on CPU0
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] CPU features: detected: EL2 vector hardening
Wed Mar 30 19:43:30 2022 kern.warn kernel: [    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
Wed Mar 30 19:43:30 2022 kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:E7:F3:C1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] Memory: 7943840K/8245248K available (7102K kernel code, 432K rwdata, 1144K rodata, 512K init, 541K bss, 235872K reserved, 65536K cma-reserved)
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] rcu:  CONFIG_RCU_FANOUT set to non-default value of 32.
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Wed Mar 30 19:43:30 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from 0xffffffc0108a0b98 with crng_init=0
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000154] Console: colour dummy device 80x25
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000381] printk: console [tty1] enabled
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000416] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000439] pid_max: default: 32768 minimum: 301
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000702] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.000845] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
Wed Mar 30 19:43:30 2022 kern.info kernel: [    0.001430] cgroup: Disabling memory control group subsystem
[... cut ...]
Wed Mar 30 19:44:54 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Wed Mar 30 19:44:54 2022 daemon.notice netifd: radio0 (2079): Command failed: Invalid argument
Wed Mar 30 19:44:54 2022 daemon.notice netifd: radio0 (2079): Device setup failed: HOSTAPD_START_FAILED
Wed Mar 30 19:44:54 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
Wed Mar 30 19:44:54 2022 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Wed Mar 30 19:44:54 2022 daemon.notice netifd: Wireless device 'radio0' is now down
'''

Aviators01, 2022-04-11:
Is there anything else I can help with, @pepe2k?

Svistunoff, 2022-05-13:
Well, is the problem already solved?  I have the same situation.  Nothing helps.

Aviators01, 2022-05-13:
```
Well, is the problem already solved?
```
Not as far as I can tell. I haven't heard anything from pepe2k or anyone else, so it seems this issue got buried under the other ones-at least until someone gets back around to it.

pepe2k, 2022-05-15:
@Aviators01 have you read and followed this: https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi#wireless_country_code_issue?

Cheers,
Piotr

Svistunoff, 2022-05-15:
Yes, I am familiar with this method. In my case it didn't work. Everything remains the same.

Aviators01, 2022-05-15:
Yes, I've already tried that. It didn't work.

pepe2k, 2022-05-16:
Just for confirmation, does 2.4 GHz work when you are running official Raspbian/RPi OS?

Aviators01, 2022-05-16:
Correct. 2.4 Ghz does work in the official Raspbian/RPi OS as well as the 5Ghz.

In OpenWrt, only the 5Ghz also works. But in a previous release version, the 5Ghz channel also didn't work. With this current version, I believe there is also a way to make it fail. I forget exactly how to make it fail, but I believe it happens if one tries to pick a channel number instead of leaving it on automatic. Do you want a logread from the 5GHz channel as well?

DagBjorndal, 2022-06-13:
I'm having the exact same problem.

rosahay-silabs, 2022-06-21:
I had a similar issue, 5GHZ works in AP mode, while 2.4GHZ won't work at all. The only solution for me now was to run in LEGACY mode(g), N mode breaks the AP.

DagBjorndal, 2022-06-22:
Thanks for the tip, Rosahay. I will try to run the onboard radio in legacy mode until the problem is fixed.

Aviators01, 2022-07-04:
Yeah, I tried legacy mode as well, but it didn't work for me at all.

Aviators01, 2022-09-19:
I checked the new release 22.03, and it still has the same issue with 2.4 Ghz radio. Only AC mode and N mode at 5 Ghz works. Everything else doesn't.

amisix, 2022-09-22:
@Aviators01 I too am unable to get 2.4Ghz to function in OpenWrt 22.03 on the Raspberry Pi 4B. Unfortunately Legacy mode does not work for me either as suggested by others. Country Code is properly set (US). 20Mhz or 40Mhz channel width does not make a difference.

inconnu299, 2022-10-15:
this works for me on Rpi3B+ 
(I think you should add     ===    option short_gi_40 '0'    and if it does not work  add also option htmode 'HT420   instead of HT40)  Use both options.

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option channel '1'
        option band '2g'
        option disabled '0'
        option short_gi_40 '0'
        option cell_density '0'
        option htmode 'HT40'

config wifi-iface 'wifinet1'
        option device 'radio0'
        option network 'lan'
	option mode 'ap'
	option ssid 'Owrt_pi'
        option encryption 'psk2'
        option key 'secretpass12'

Vonderhecke, 2022-10-28:
Thanks a lot inconnu299. it worked for me

Aviators01, 2022-11-01:
Thanks so much @inconnu299. It looks like adding the option short_gi_40 '0' and picking a fixed channel worked for me.

Aviators01, 2022-11-01:
@pepe2k . Here are my logs and config files from the fix inconnu299 suggested. It now works because of the fix. Only remaining problem is that option channel 'auto' doesn't work. It must be a number in order for the wifi to work. Hope this can help with future updates.

Wireless config file:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option cell_density '0'
        option country 'CA'
        option band '2g'
        option short_gi_40 '0'
        option channel '4'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key 'roottesting'
```

Logread
```
root@OpenWrt:/etc/config# logread
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
Mon Sep 19 16:32:14 2022 kern.notice kernel: [    0.000000] Linux version 5.10.138 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19685-512e76967f) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Sat Sep 3 02:55:34 2022
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Zone ranges:
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Early memory node ranges
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 2061312
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000]   DMA zone: 3792 pages used for memmap
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000]   DMA zone: 242688 pages, LIFO batch:63
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000]   DMA32 zone: 12288 pages used for memmap
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages used for memmap
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000]   Normal zone: 1048576 pages, LIFO batch:63
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s45528 r8192 d32296 u86016
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s45528 r8192 d32296 u86016 alloc=21*4096
Mon Sep 19 16:32:14 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Detected PIPT I-cache on CPU0
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] CPU features: detected: Spectre-v2
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] CPU features: detected: Spectre-v4
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] CPU features: detected: Spectre-BHB
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
Mon Sep 19 16:32:14 2022 kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:E7:F3:C1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] Memory: 7942444K/8245248K available (7934K kernel code, 902K rwdata, 1280K rodata, 448K init, 545K bss, 237268K reserved, 65536K cma-reserved)
Mon Sep 19 16:32:14 2022 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[... cut ...]
Mon Sep 19 16:38:00 2022 kern.info kernel: [  353.636105] br-lan: port 2(wlan0) entered blocking state
Mon Sep 19 16:38:00 2022 kern.info kernel: [  353.641424] br-lan: port 2(wlan0) entered forwarding state
Mon Sep 19 16:38:00 2022 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Mon Sep 19 16:38:00 2022 daemon.notice hostapd: wlan0: AP-ENABLED
Mon Sep 19 16:38:01 2022 daemon.notice netifd: Wireless device 'radio0' is now up
Mon Sep 19 16:38:01 2022 daemon.notice netifd: Network device 'wlan0' link is up
```

cohen-sydney, 2024-01-12:
> @pepe2k . Here are my logs and config files from the fix inconnu299 suggested. It now works because of the fix. Only remaining problem is that option channel 'auto' doesn't work. It must be a number [...]

> ```

where are the "Wireless config file:" ??? thanks

cohen-sydney, 2024-01-12:
are they on /etc/config/wireless
??