openwrt/openwrt #9619: Corruption of compressed firmware .gz files due to metadata in KDE Ark / Should metadata be appended before or after gzip?

ValdikSS, 2022-04-01:
OpenWRT build system appends signature metadata to the firmware file, at least for x86 builds.

https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/include/image-commands.mk#L82-L91

This also applies for x86 builds which are published as compressed .gz images:

https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/x86/image/Makefile#L108-L109

Upon **extracting with KDE Ark** archive manager, the resulting unpacked .img file is 126 337 024 bytes in size, and it **doesn't boot** due to the fact that the latest partition in partition table exceeds disk image size (see #9032).
However, if this file is unpacked with `gunzip`, the resulting file is 126 353 408 bytes in size — 16384 bytes larger. It boots fine and works as intended.

Gunzip reports 'garbage' in the end of the compressed data:
`gzip: openwrt-21.02.2-x86-generic-generic-ext4-combined.img.gz: decompression OK, trailing garbage ignored`

This is undoubtedly a bug of Ark, however I also **not sure if appending metadata to the .gz file** (not to the original firmware image) **makes any sense**.

This is inconsistent in OpenWRT.
Metadata is appended to the **original image** before compressing in:
* [sunxi](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/sunxi/image/Makefile#L37)
* [omap](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/omap/image/Makefile#L38)
* [ipq40xx](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/ipq40xx/image/generic.mk#L581)

Metadata is appended to the **compressed .gz image** in:
* [apm821xx](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/apm821xx/image/sata.mk#L22)
* [bcm27xx](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/bcm27xx/image/Makefile#L62)
* [mediatek](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/mediatek/image/mt7623.mk#L141)
* [mvebu](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/mvebu/image/Makefile#L173)
* [qoriq](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/qoriq/image/generic.mk#L9)
* [rockchip](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/rockchip/image/armv8.mk#L10)
* [tegra](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/tegra/image/Makefile#L35)
* [x86](https://github.com/openwrt/openwrt/blob/1fa8780056a8c7a2e26c8b4d5e6979232f117349/target/linux/x86/image/Makefile#L108-L109)

tidux, 2022-04-19:
Appending data to a gzipped archive is almost never the right thing to do.  This should probably be considered a bug in OpenWRT.

joshenders, 2023-03-03:
Excellent bug report @ValdikSS!

I noticed this today as well decompressing `openwrt-22.03.3-x86-64-generic-ext4-combined-efi.img.gz`:

```sh
$ gunzip openwrt-22.03.3-x86-64-generic-ext4-combined-efi.img.gz
gzip: openwrt-22.03.3-x86-64-generic-ext4-combined-efi.img.gz: decompression OK, trailing garbage ignored
```

Used this to get the trailing data:

```pl
use warnings;
use strict;
use IO::Uncompress::Gunzip qw(:all) ;
use IO::File ;

my $file =  "./openwrt-22.03.3-x86-64-generic-ext4-combined-efi.img.gz";
my $in = new IO::File "<$file";

gunzip $in => "/dev/null",
    TrailingData => my $trailing;

print "$trailing\n";
```

Which returns:
```
$ perl script.pl 
# fake certificateFWx0>d}"
```

ulope, 2023-07-31:
Appending the metadata also breaks decompressing the archive on macOS. The built in utility in Finder fails decompressing the file with a very unhelpful `Unable to expand "<image>" into <Folder> (Error 2 - No such file or directory)`

Presumably it uses `gunzip` under the hood which exits with status `2` when warnings occur (and the Finder integration can't deal with that).

MrSteezyMcSteez, 2023-09-16:
I am seeing this with `openwrt-22.03.5-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz`

here's the trailing data:

```bash
$ cat trailing.raw 
{  "metadata_version": "1.1", "compat_version": "1.0",   "supported_devices":["friendlyarm,nanopi-r4s"], "version": { "dist": "OpenWrt", "version": "22.03.5", "revision": "r20134-5f15225c1e", "target": "rockchip/armv8", "board": "friendlyarm_nanopi-r4s" } }
FWx0慇?# fake certificateFWx08eI"
```

and here it is with special characters printed (e.g. `^@` is null char):

```bash
$ cat -vet trailing.raw 
^@^@^@^@^@^@^@^@{  "metadata_version": "1.1", "compat_version": "1.0",   "supported_devices":["friendlyarm,nanopi-r4s"], "version": { "dist": "OpenWrt", "version": "22.03.5", "revision": "r20134-5f15225c1e", "target": "rockchip/armv8", "board": "friendlyarm_nanopi-r4s" } }$
FWx0M-fM-^EM-^GM-"^A^@^@^@^@^@^A^Z# fake certificateFWx0^T8eI^@^@^@^@^@^@^@"
```

FYI, here is a bash script to extract the trailing data from a gzip, adopted from [this stackoverflow answer](https://unix.stackexchange.com/a/706331):

```bash
#!/usr/bin/env bash

test -z "$1" && { echo "usage: $0 file_with_garbage.gz" ; exit 1 ; }

# Adopted from: https://unix.stackexchange.com/a/706331

find_gzip_size() {
  local gzip_file="$1"
  test -z "$gzip_file" && { echo "missing gzip file to get size" ; exit 1 ; }

  set -e
  gzip_file=${1:?Inform a gzip file}
  size=$(stat -c%s "$gzip_file")
  min=0
  max=$size
  while true; do
          if head -c "$size" "$gzip_file" | gzip -v -t - &>/dev/null; then
                  echo $size
                  break
          else
                  case "$?" in
                          1) min=$size ;;
                          2) max=$size ;;
                  esac
                  size=$(((max-min)/2 + min))
          fi
  done
  set +e
}

find_trailing_data() {
  local gzip_file="$1"
  test -z "$gzip_file" && { echo "missing gzip file to get size" ; exit 1 ; }
  gzip_size=$(find_gzip_size "$gzip_file")
  head -c "$gzip_size" "$gzip_file" > data.gz
  tail -c +$((1+gzip_size)) "$gzip_file" > trailing.raw
}

find_trailing_data "$@"

echo "See files data.gz and trailing.raw"

```

joshenders, 2026-09-04:
No longer seeing any errors from `gunzip` when extracting `openwrt-25.12.5-x86-64-generic-ext4-combined-efi.img.gz`