openwrt/openwrt #9620: "btrfs device scan" runs after trying to mount extroot

M95D, 2022-04-01:
Trying to mount extroot overlay from a btrfs raid1 on two usb memory sticks. Mount fails due to missing second device. At this time, the second device is in fact connected, detected by usb, detected by usb-storage and device file /dev/sdb created, but not scanned by btrfs.
```
[   15.883125] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   15.897278] BTRFS: device label GRAPHRT devid 1 transid 822 /dev/sda scanned by block (941)
...
[   15.946046] BTRFS error (device sda): devid 2 uuid 8cddeba0-2e1f-4057-bd05-13fa1a590c42 is missing
[   15.955072] BTRFS error (device sda): failed to read the system array: -2
[   15.962616] BTRFS error (device sda): open_ctree failed
[   15.968330] block: extroot: mounting /dev/sda (btrfs) on /tmp/extroot/overlay failed: 22 (Invalid argument)
```

A boot without the two sticks will show that "btrfs device scan" (/lib/preinit/85_btrfs_scan) is run *after* trying to mount extroot.
```
[   21.785141] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   21.796904] block: extroot: device not present, retrying in 5 seconds
[   27.209919] block: extroot: cannot find device with label GRAPHRT
[   27.223607] mount_root: switching to ubifs overlay
[   27.321773] ubi1: attaching mtd8
[   27.522631] ubi1: scanning is finished
[   27.534697] ubi1: attached mtd8 (name "syscfg", size 38 MiB)
...
[   27.721534] random: fast init done
[   27.820045] UBIFS (ubi1:0): UBIFS: mounted UBI device 1, volume 0, name "syscfg"
...
[   27.895238] random: crng init done
[   27.898665] random: 5 urandom warning(s) missed due to ratelimiting
[   27.905307] urandom-seed: Seeding 4096 bits and crediting
[   27.911083] urandom-seed: Saving 4096 bits of creditable seed for next boot
Scanning for Btrfs filesystems
[   28.659123] procd: - early -
```

M95D, 2022-04-19:
I don't understand OpenWrt boot process. It seems that **btrfs device scan** won't run twice from two different files in **/lib/preinit/**.
If run (once) before root mount, it won't find the drives anyway, not even with **-d** parameter.
This btrfs stuff seems hopeless. Better use classic md raid. I'm closing this issue.

closed 2022-04-19 by M95D

g6094199, 2026-02-01:
btrfs support is simply still broken in 2026 in openwrt.
see:

https://github.com/openwrt/openwrt/issues/7176
https://github.com/openwrt/openwrt/issues/9620

https://forum.openwrt.org/t/possible-bug-btrfs-raid-mdraid-uuid-bug-on-kirkwood/243828/