openwrt/openwrt #9624: firewall4 ipv6 address parsing error when using ipv6 port forwarding (ipv6 nat)

wackejohn, 2022-04-02:
When using ipv6 port forwarding, the ipv6 address was parsed without square brackets ( `[` and `]` )  before the ipv6 port, and this cause the ipv6 port forwading failing.

My ipv6 port forwarding config:

```
config redirect
        option target 'DNAT'
        option src 'wan'
        option dest 'lan'
        option proto 'tcp'
        option family 'ipv6'
        option src_dport '591'
        option dest_port '5900'
        option name 'HOME-Server-VNC_6'
        option dest_ip 'dd5d:189b:b5f9:10::2'
```
The nft ruleset after firewall4 parsed, the port forwarding failed:
```
        chain dstnat_wan {
              .....
                meta nfproto ipv6 tcp dport 591 counter packets 0 bytes 0 dnat ip6 to dd5d:189b:b5f9:10::2:5900 comment "!fw4: HOME-Server-VNC_6"
             .....
        }

```
But if i add square brackets ( `[` and `]` ) to the ipv6 address manually, the port forwarding do working:
```
chain dstnat_wan {
        meta nfproto ipv6 tcp dport 591 counter dnat [dd5d:189b:b5f9:10::2]:5900 comment "!fw4: HOME-Server-VNC_6"
}

```
The nft ruleset after add square bractets manually:
```
        chain dstnat_wan {
                ......
                meta nfproto ipv6 tcp dport 591 counter packets 0 bytes 0 dnat ip6 to [dd5d:189b:b5f9:10::2]:5900 comment "!fw4: HOME-Server-VNC_6"
                ......
        }

```

jow-, 2022-04-02:
Upstream fix at https://git.openwrt.org/?p=project/firewall4.git;a=commitdiff;h=11feddff95a47f292d84784112a71b69cfa951b6

wackejohn, 2022-04-03:
> Upstream fix at https://git.openwrt.org/?p=project/firewall4.git;a=commitdiff;h=11feddff95a47f292d84784112a71b69cfa951b6

Many thanks.

closed 2022-04-03 by wackejohn