openwrt/openwrt #9630: Nftables Security escalation

GeertVisser, 2022-04-03:
See link for background and link to kernel patch. This issue is a FYI since we are working on nftable support at the moment. https://blog.dbouman.nl/2022/04/02/How-The-Tables-Have-Turned-CVE-2022-1015-1016/

SirDMZ, 2022-04-03:
So the second CVE is interesting. 
"5.1. Affected versions and patch
Kernels starting from version 3.13-rc1"

This will affect anyone running a non-patched system running NF_Tables

oldium, 2022-04-16:
Both CVEs are fixed in 5.15.32.
5.10.109 contains fix for only CVE-2022-1016.

SirDMZ, 2022-04-16:
Yeah, vulnerabilty for -1015 wasn’t introduced until 5.12 branch iirc, so all good.

Ansuel, 2022-06-04:
We are at 5.15.40 and 5.10.115
Guess i can close this with fixed.

closed 2022-06-04 by Ansuel