openwrt/openwrt #9632: iptables-ntf prints warning because of dependencies with old iptable table modules

oldium, 2022-04-03:
The switch to nftables still keeps installing and loading old iptable table modules. The most visible are (they are [direct reason](https://git.netfilter.org/iptables/tree/iptables/nft-shared.c#n1356) for the warning):

* `iptables-nft` requires `kmod-ipt-core` package, which installs and _auto-loads_ `iptable_mangle` and `iptable_filter`
* `ip6tables-nft` requires `kmod-ip6tables` package, which installs and _auto-loads_ `ip6table_mangle` and `iptable_filter`
* `arptables-nft` requires `kmod-arptables` package, which installs and _auto-loads_ `arptable_filter`

When installed, the `iptables-nft -L` and `iptables-nft-save` prints warning that `# Warning: iptables-legacy tables present, use iptables-legacy to see them`.

oldium, 2022-04-09:
Another issue with nftables is that `nft_log` module (part of `kmod-nft-core`) is actually unusable without the loggers, which are currently part of `kmod-ipt-core`, which you, in the current state, do not want to have installed.

Doridian, 2022-07-19:
Is there any work around for this issue? Is the issue even a problem besides the warning messages?

Should one just remove those modules from autoload (those modules = `iptable_*`, `arptable_*` and `ip6table_*`)?

Doridian, 2022-07-19:
From reading through the whole dependency chain of things, is the dependency itself wrong? Or just the loading of iptable_* modules?

Because it seems there is a big chain of dependencies on the kernel modules like: `ip_tables` `xt_*` `ipt_*` (which are all provided by ipt-core), just not `iptable_*`.
So, as I understand, the correct solution here would be to split off packages again, like in the log case?
As in, split out the `iptable_*`, and also untangle the dependency of `kmod-ipt-raw`, as this also seems to be unnecessary, but modules that `mwan3` for example requires `iptables-mod-conntrack-extra` which depends on `kmod-ipt-raw`.
Is this dependency also wrong? Or do we need the kmods from the `iptables-mod-conntrac-extra` and just not `iptable_raw`?

oldium, 2022-07-19:
When the iptables modules are loaded, you cannot use `iptables-nft` at all - the execution ends with a warning message shown above (it is a fatal error for the program, nothing is being done). The only long-term workaround which I found is simple - do not install `iptables-nft` on nft-only system.

The solution is really to split the dependencies to nft-only, iptables-only and shared ones (I have not investigated what depends on what). I have not checked if anything improved with regards to iptables dependencies any recently.

Doridian, 2022-07-19:
It seems keepalived (through dependency on libipset13) also depends indirectly on iptables kmods.
However, I was able to just force removal of the ipset kmods and leave libipset intact.
This makes keepalived work fine (I assume because I don't use any features that need ipset to function), and now I no longer have any of these old things loaded or even present.
So this seems like something else to check for.

champtar, 2022-07-19:
@oldium those are warnings, things still work (I'm running 22.03 + mwan3 which uses iptables-nft and it works for me)

Yes many dependencies are wrong and need fixing, just need to find some time

oldium, 2022-07-19:
> @oldium those are warnings, things still work (I'm running 22.03 + mwan3 which uses iptables-nft and it works for me)

You are right, I probably had different issue and I mixed that up. The [code](https://git.netfilter.org/iptables/tree/iptables/nft-shared.c#n1394) only prints warning and continues.

oldium, 2025-07-30:
Closing, probably not up-to-date any more. Using nftables happily now.

closed 2025-07-30 by oldium