openwrt/openwrt #9636: Reboot does not work with Hilink HLK-7688a / short removal of power nedded

balou1974, 2022-04-04:
Hello,

it seems reboot does not work with Hilink HLK-7688a in combination with HLK-RM08a development board.
Openwrt simply stops but no rebootis issued. U-Boot does not start after reboot command (checked with serial cable).
Maybe this has something to do with the DTB?

ptpt52, 2022-04-04:
32M flash 4B addr issue

balou1974, 2022-04-06:
@ptpt52: Will it help to add broken-flash-reset to the dts-source / would be no problem to clone the openwrt-repository, add the missing line and compile everything - maybe I have time to try it out a little later.

balou1974, 2022-04-08:
I added the broken-flash-reset to mt7628an_hilink_hlk-7688a.dts and  mt7628an_hilink_hlk-7628n.dts and reboot worked like a charm.

Adding broken-flash-reset to the dts files will fix the issue!

pepe2k, 2022-04-11:
Should be fixed in 169c9e3a88b7d4521a797864540657ef33554663.

closed 2022-04-11 by pepe2k