openwrt/openwrt #9639: KSMBD version mismatch.

tapper82, 2022-04-04:
After trying to run ksmbd you get. Tue Mar 29 16:36:01 2022 kern.err kernel: [ 959.844616] ksmbd: No IPC daemon response for 20s 
You can find more info here:
https://github.com/openwrt/openwrt/commit/fbb8e1ef45fa3ebb10e653cdb14465508541350d#commitcomment-70364862

tapper82, 2022-04-04:
I am all so told that it's broken in OpenWrt 22.03 and OpenWrt 21.0.2.3 How ever I have not tested this.

neheb, 2022-04-07:
Should already be fixed.

tapper82, 2022-04-07:
Hi I can't find where on git the fix is. Could you point me to a link pleas?

neheb, 2022-04-07:
https://github.com/openwrt/packages/commit/1f949f545624ede60a21f304fa19078dc89f7f72

tapper82, 2022-04-08:
Hi thanks for the link. Wat do you meen by this "This does not work with the kmod currently in tree." If I do a new build from master will ksmbd work or not?

tapper82, 2022-04-08:
Still broken. 
Fri Apr  8 19:21:19 2022 kern.err kernel: [  505.369816] ksmbd: No IPC daemon response for 20s
This is on:
OpenWrt SNAPSHOT r19393-0392644083 / LuCI Master git-22.089.43958-7110635
Kernel Version
5.15.31

egorenar, 2022-04-09:
Still borked.

```
OpenWrt SNAPSHOT r19416-bda992684dac 
```

```
[  183.117459] ksmbd: No IPC daemon response for 20s
[  183.156445] ksmbd: sock_read failed: -108
```

egorenar, 2022-04-09:
It seems that the fix was applied only on a branch and not master ?

neheb, 2022-04-09:
If you're using ksnbd-tools 3.4.4, downgrade to 3.4.2.

Dr-Escher, 2022-04-21:
This issue persists on OpenWrt 21.02.3 r16554-1d4dea6d4f

```
root@OpenWrt:~# opkg list | grep ksmbd
kmod-fs-ksmbd - 5.4.188+3.4.4-1
ksmbd-server - 3.4.2-2
luci-app-ksmbd - git-21.295.66827-c592d0a
```

```
root@OpenWrt:~# ksmbd.mountd -V
ksmbd-tools version : 3.4.2
```

```
root@OpenWrt:~# ksmbd.mountd --n
[ksmbd-worker/6539]: ERROR: Recv() error Input data out of range [-8]
[ksmbd-manager/6538]: ERROR: WARNING: child process exited abnormally: 6539
[ksmbd-manager/6538]: ERROR: Fatal IPC error. Terminating. Check dmesg.
[ksmbd-manager/6538]: ERROR: can't execute kill 6539: No such process
[ksmbd-manager/6538]: INFO: Exiting. Bye!
```

neheb, 2022-04-21:
This is strange.

One group of people says ksmbd-tools 3.4.4 doesn't work properly and another says 3.4.2 does not.

Dr-Escher, 2022-04-22:
This is caused by a version mismatch between kmod-fs-ksmbd and ksmbd-server (contains ksmbd-tools). Previously the tools version was ahead, which caused issues (fixed by downgrade). Now as the kmod got upgraded, we need the corresponding tools version as well.

Before: kmod-fs-ksmbd 3.4.2 <-> ksmbd-tools 3.4.4
Current: kmod-fs-ksmbd 3.4.4 <-> ksmbd-tools 3.4.2

Both combinations are incompatible. Workaround as of now is using ksmbd-tools 3.4.4 directly from the snapshot repository.

petiepooo, 2022-05-08:
Would it be possible to post more than one version to the 21.02.3 repo so we can try various combinations by specifying the version for opkg to install?

Barring a quick solution, this should probably also be listed as a regression in the 21.02.3 release page at https://openwrt.org/releases/21.02/notes-21.02.3.  Had I known, I would have held off on my upgrade.

clauded, 2022-05-12:
> This is caused by a version mismatch between kmod-fs-ksmbd and ksmbd-server (contains ksmbd-tools). Previously the tools version was ahead, which caused issues (fixed by downgrade). Now as the kmod got upgraded, we need the corresponding tools version as well.
> 
> Before: kmod-fs-ksmbd 3.4.2 <-> ksmbd-tools 3.4.4 Current: kmod-fs-ksmbd 3.4.4 <-> ksmbd-tools 3.4.2
> 
> Both combinations are incompatible. Workaround as of now is using ksmbd-server 3.4.4 directly from the snapshot repository.

Since the new driver is compiled for kernel 5.10, have you tried it with the current 5.4 kernel? Or else, can you point where to find it for 5.4?

M95D, 2022-05-15:
I just built an image from git master. What was built is v3.4.4, not 3.4.2.
It's not working. Log shows "ksmbd: No IPC daemon response for 20s" -  the daemon crashed on first smb access by a client.
I run a strace on ksmbd.mount to find out why: [ksmbd-worker/8928]: ERROR: Recv() error Input data out of range [-8]
It might be this bug: https://github.com/cifsd-team/ksmbd-tools/issues/227

M95D, 2022-05-15:
I tried to test the patch, but kernel module is v3.4.3, not 3.4.4 ?! (Building for mvebu / Linksys mamba.)
After some adjustments, the patch works, daemon no longer crashes, but it still doesn't work. Now I get:
... FUTEX_WAIT_PRIVATE, 5, {tv_sec=0, tv_nsec=0}) = -1 ETIMEDOUT (Operation timed out)
and
... FUTEX_WAIT_PRIVATE, 2, NULL) = -1 EAGAIN (Resource temporarily unavailable)

tapper82, 2022-05-15:
If you have the space on your router you could just use samba like i did. I got sick of asking about this. When OpenWrt makes the jump to Kernel 5.15 KSMBD is built in to the Kernel.

M95D, 2022-05-16:
5.15 - I can't wait!
I have plenty of space, but I can't use Samba. I get "illegal character" for filenames with accents. I tried "full language support" in menuconfig.
----
With the patch, ETIMEDOUT errors occur when opening //server/, but opening //server/share directly works without any problems! It might be just a permissions problem on my system... It's a mess with all the tests.
I'll try to clean the patch for v3.4.3 and make a pull request when I have the time, but I would prefer an update to kernel module v3.4.4, if possible, and the original patch.

yoursunny, 2022-06-05:
I have GL-AR750 device, and ksmbd stopped working since I upgraded to 21.02.3.
Today I found the solution:

1. Go to https://archive.openwrt.org/releases/packages-21.02/mips_24kc/packages/ and download `ksmbd-server_3.4.4-1_mips_24kc.ipk` package.
2. Manually upload and install this package.

For some reason `opkg` selects the 3.4.2 version despite 3.4.4 is available, but manual installation fixes the problem.

neheb, 2022-06-05:
Latest version should be fine. I'll put up backport PRs.

AapoTahkola, 2022-06-07:
No go with OpenWrt 22.03.0-rc3 with MT7621.
ksmbd-server 3.4.5-1
ksmbd-utils 3.4.5-1
kmod-fs-ksmbd 5.10.116+3.4.3-1

I would not rather go on a crusade to get 5.15 kernel on this thing. I was able to get writable ntfs with kmod-fs-antfs which is kinda nice. With user space samba4 big file read speed is about 7MB/s which i guess is not that bad and I was able to mount samba share on linux without "vers=2.0" option. 7MB/s is still pretty far from what the SSD I have connected would be able to deliver.

ldiebel, 2022-06-27:
> I have GL-AR750 device, and ksmbd stopped working since I upgraded to 21.02.3. Today I found the solution:
>
> 1. Go to https://archive.openwrt.org/releases/packages-21.02/mips_24kc/packages/ and [...]

This fixed my problem. Thanks yoursunny

AapoTahkola, 2022-06-28:
Works fine with OpenWrt 22.03.0-rc4 . kmod-fs-antfs was under-performing so switched back to ntfs-3g. 35MB/s big file read speed with this combo as I recall. Not much benefit from using ext4 over ntfs.

Neustradamus, 2022-12-27:
I think that this ticket has been solved, right?

tapper82, 2022-12-27:
Yes fixt.

closed 2022-12-27 by tapper82