openwrt/openwrt #9644: Mikrotik Hap Ac3 bootloop

djwujek, 2022-04-04:
Mikrotik Hap Ac3 router. The first image (Kernel) uploads without any problem. When I update, the router starts to restart every now and then.

indviral3, 2022-05-02:
> Mikrotik Hap Ac3 router. The first image (Kernel) uploads without any problem. When I update, the router starts to restart every now and then.
Hi, you repair router, how?) Thanx.

mmacura311, 2023-10-17:
Just yesterday I attempted to flash openwrt on mikrotik hap ac3 since the 23.05 release is out and I experienced the same problem - netboot to initramfs image went ok, but after flashing the sysupgrade the device went into endless reboot loop.
I kinda thought this issue is solved as the device was included in the release. Anyway - I did netinstall the routeros back and the device works.

few questions: this commands posted above, do they need to be executed at every flash attempt? Assuming I will be running 23.05 and later 23.05.1 is release, later 23.05.2 and so on... ? or is this one-time needed only when converting from routeros to openwrt? Can routeros be restored after re-formatting the mtd1 device or is this irreversible? Just asking to be clear what I'm getting into for next time I will try to flash the firmware.

Also - can this be somehow incorporated in the image flashing procedure? is there a eg. place to run a scriptlet? If yes this should be done so users do not fall into this trap? if not, this should be documented. The device page for hap ac3 [does not exist yet](https://openwrt.org/toh/start?dataflt%5BBrand*%7E%5D=mikrotik&dataflt%5BModel*%7E%5D=hap), so it should be created and the flash procedure documented. (I can write it if I have access)

thx

exzombie, 2024-07-23:
> ```
> ubidetach -m 1
> ubiformat /dev/mtd1 -y
> ubiattach -m 1
> ```

I confirm these steps work.

BenjiWiebe, 2025-02-15:
I just had the same issue as mmacura311 of the device bootlooping after attempting to install the sysupgrade through the initramfs mode. I netbooted it back to the initramfs, and did the ubiformat steps, and now installing the sysupgrade works. I am using 24.10.0.