openwrt/openwrt #9645: Atheros OpenWrt csi tool build failed

shuchi-seki, 2022-04-05:
Hi,
I am new in openwrt. I am using two TL-WDR4300 routers with my laptop for my configuration. I have installed the openwrt firmwares in routers. And now I need to install the atheros openwrt csi tool in my ubuntu. whenever i am trying to install the atheros csi tool for openwrt and run the make, the build says 'failed'. Everytime it is showing errors whenever I am trying to run the feeds\packages fro the scripts. here is the following issues i am facing. Can someone help me solve this issue? I really need the tool for my work. Thanks.

shuchi@shuchi:~$ cd ~/Atheros_CSI_tool_OpenWRT_src
shuchi@shuchi:~/Atheros_CSI_tool_OpenWRT_src$ ./scripts/feeds update -a 
Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git' ...
Cloning into './feeds/packages'...
remote: Enumerating objects: 6468, done.
remote: Counting objects: 100% (6468/6468), done.
remote: Compressing objects: 100% (5412/5412), done.
remote: Total 6468 (delta 205), reused 5022 (delta 127), pack-reused 0
Receiving objects: 100% (6468/6468), 13.78 MiB | 1.54 MiB/s, done.
Resolving deltas: 100% (205/205), done.
Updating files: 100% (4135/4135), done.
Create index file './feeds/packages.index' 
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'time'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'ldconfig-stub'... ok.
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
ERROR: please fix feeds/packages/devel/pkgconf/Makefile - see logs/feeds/packages/devel/pkgconf/dump.txt for details
Collecting package info: feeds/packages/lang/python/python3-django-etesync-journ
ERROR: please fix feeds/packages/libs/afalg_engine/Makefile - see logs/feeds/packages/libs/afalg_engine/dump.txt for details
ERROR: please fix feeds/packages/libs/freetype/Makefile - see logs/feeds/packages/libs/freetype/dump.txt for details
ERROR: please fix feeds/packages/libs/glib2/Makefile - see logs/feeds/packages/libs/glib2/dump.txt for details
ERROR: please fix feeds/packages/libs/gost_engine/Makefile - see logs/feeds/packages/libs/gost_engine/dump.txt for details
ERROR: please fix feeds/packages/libs/jose/Makefile - see logs/feeds/packages/libs/jose/dump.txt for details
ERROR: please fix feeds/packages/libs/json-glib/Makefile - see logs/feeds/packages/libs/json-glib/dump.txt for details
ERROR: please fix feeds/packages/libs/jsoncpp/Makefile - see logs/feeds/packages/libs/jsoncpp/dump.txt for details
ERROR: please fix feeds/packages/libs/libdrm/Makefile - see logs/feeds/packages/libs/libdrm/dump.txt for details
ERROR: please fix feeds/packages/libs/libesmtp/Makefile - see logs/feeds/packages/libs/libesmtp/dump.txt for details
ERROR: please fix feeds/packages/libs/libevdev/Makefile - see logs/feeds/packages/libs/libevdev/dump.txt for details
ERROR: please fix feeds/packages/libs/libinput/Makefile - see logs/feeds/packages/libs/libinput/dump.txt for details
ERROR: please fix feeds/packages/libs/liblz4/Makefile - see logs/feeds/packages/libs/liblz4/dump.txt for details
ERROR: please fix feeds/packages/libs/libmbim/Makefile - see logs/feeds/packages/libs/libmbim/dump.txt for details
ERROR: please fix feeds/packages/libs/libmpdclient/Makefile - see logs/feeds/packages/libs/libmpdclient/dump.txt for details
ERROR: please fix feeds/packages/libs/libpsl/Makefile - see logs/feeds/packages/libs/libpsl/dump.txt for details
ERROR: please fix feeds/packages/libs/libqmi/Makefile - see logs/feeds/packages/libs/libqmi/dump.txt for details
ERROR: please fix feeds/packages/libs/libqrtr-glib/Makefile - see logs/feeds/packages/libs/libqrtr-glib/dump.txt for details
ERROR: please fix feeds/packages/libs/libslirp/Makefile - see logs/feeds/packages/libs/libslirp/dump.txt for details
ERROR: please fix feeds/packages/libs/libsoup/Makefile - see logs/feeds/packages/libs/libsoup/dump.txt for details
ERROR: please fix feeds/packages/libs/nlohmannjson/Makefile - see logs/feeds/packages/libs/nlohmannjson/dump.txt for details
[... 65 lines cut ...]
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Collecting package info: done
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Collecting target info: done
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
Cloning into './feeds/telephony'...
remote: Enumerating objects: 287, done.
remote: Counting objects: 100% (287/287), done.
remote: Compressing objects: 100% (274/274), done.
remote: Total 287 (delta 7), reused 70 (delta 1), pack-reused 0
Receiving objects: 100% (287/287), 194.59 KiB | 257.00 KiB/s, done.
Resolving deltas: 100% (7/7), done.
Create index file './feeds/telephony.index' 
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Collecting package info: done
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Collecting target info: done

pkgadd, 2022-04-05:
At what level is this an OpenWrt issue?

This tool isn't available in OpenWrt at all, so you're trying to compile something obtained from another source, talk to them for support.

shuchi-seki, 2022-04-05:
Well, I think my understanding was wrong that it is something to do with
the openwrt firmware that I installed. Thanks for pointing that out.

mans0n, 2022-04-06:
Yeah, ath csi tool is a downstream project so here is not the best place to ask for the support. But I've tried it before so I might be of little help...

The OpenWrt version of the ath csi tool is very old (IIRC it's CC or DD) so it will not be able to fetch the current feeds correctly. Probably you can just use it without the feeds (I mean skipping `./scripts/feeds update && ./scripts/feeds install`). Or you can fix the feeds version if you really need it.