openwrt/openwrt #9655: openwrt stable [21.02.2] - console cmd "reboot" not working - ZBT-3526 (MT7621)

camelcamro, 2022-04-05:
Hello
tested today the ZBT-3526 (MT7621)
with version: 21.02.2 -> [openwrt-21.02 branch (git-22.046.85957-59c3392)](https://github.com/openwrt/luci) / OpenWrt 21.02.2 r16495-bf0c965af0
and tested if reboot is working via console (ssh)
and it is hanging, don't know what is the problem
i need to reset the device via removing the power
that is really a show stopper, and without working reboot, is senseless for me.
any ideas ?
cu

pepe2k, 2022-04-06:
Possibly related: #9636, #9547

pepe2k, 2022-04-11:
Should be fixed in 169c9e3a88b7d4521a797864540657ef33554663.

closed 2022-04-11 by pepe2k

pepe2k, 2022-04-12:
@camelcamro the fix was pushed to 21.02 branch, so it will be included in the next service release (21.02.03) and soon in the 21.02-snapshot images (https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/). There is no way to have that included in already released 21.02.2 version.

camelcamro, 2022-04-12:
wau !

FYI: if i would have known that, i wouldn'T had ask for it ...

BUt, if you open the directory:
https://downloads.openwrt.org/releases
, then there is the folder: 
https://downloads.openwrt.org/releases/21.02-SNAPSHOT
not visible ...
so, i never did know, that such kind of folder is existing at all.

Thx a lot.

-
thx a lot  !!!!
i will wait few days, and then i will try the "https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/" and will report back. if fixes solved the issue.. THx

camelcamro, 2022-04-14:
this special SNAPSHOT are having a password ?
furthermore i can't login to luci, as there comes only "
"Bad Gateway 
The process did not produce any response"

so, do i need to hack into via failsafe mode or what is that version special, that neiter a login is working ?
also googled for the password, but didn'T found anything.
that "https://downloads.openwrt.org/releases/21.02-SNAPSHOT" is really something special hidden to the world ?
any hints pls to test ?
tried it on 3 routers, and none of them, are working after a flash (also used the failsafe ROM to reset before
i was flashing around 12 times ... i don'T think, that it is a user issue. 
pls assist if you want that i test the 21.01.02 snapsjhot version
cu

pepe2k, 2022-04-15:
@camelcamro no password, nothing special, just built from a release branch (instead of `master` in case of the "standard" snapshots). The fastest way to find out what's going on after upgrade on your board would be gaining access over a serial console.

camelcamro, 2022-04-15:
well, then these images are bricked ...

as i was doing a failsafe, and it is not possible to do the mount_root, so, they are corrupted
as also the jffs is not ready and can't be mounted ..
tested wg3526 16mb and 32mb version. so they are having a bigger problem
also the luci is broken and can't be used -> "Bad Gateway
The process did not produce any response"

hmm, in worst case, i need to wait for 21.1.3 version which will be maybe soon built up ;-(

camelcamro, 2022-04-15:
ok, i tested meanwhile the mt7620 - we 826 -. 16mb + 32 versioin. they are seems to be OK
so, defintive only bricked images for the mt7621 - zbt3526 (16+32MB version)
as I'm not having a serial interface, i can'T find out what is wrong, that thiese images are broken
maybe they are since a longer while broken, and noone checked them out.

camelcamro, 2022-04-15:
well, it was possible to come up after several flashes .... don'T ask me, why it was now possible but, i got the logs ..
maybe someone can see whats going on with the jffs ?
`Mon Apr 11 20:53:02 2022 kern.notice kernel: [    0.000000] Linux version 5.4.188 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16549-c5ef62a218)) #0 SMP Mon Apr 11 20:52:57 2022
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] MIPS: machine is Zbtlink ZBT-WG3526 (32M)
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Mon Apr 11 20:53:02 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Mon Apr 11 20:53:02 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mon Apr 11 20:53:02 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Zone ranges:
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Early memory node ranges
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
Mon Apr 11 20:53:02 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Mon Apr 11 20:53:02 2022 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Mon Apr 11 20:53:02 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon Apr 11 20:53:02 2022 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Mon Apr 11 20:53:02 2022 kern.debug kernel: [    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26768 r8192 d22384 u57344
Mon Apr 11 20:53:02 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s26768 r8192 d22384 u57344 alloc=14*4096
Mon Apr 11 20:53:02 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
Mon Apr 11 20:53:02 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00020000
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00020000
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] Memory: 510056K/524288K available (6097K kernel code, 210K rwdata, 748K rodata, 1252K init, 238K bss, 14232K reserved, 0K cma-reserved, 262144K highmem)
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Mon Apr 11 20:53:02 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from 0x806e7a3c with crng_init=0
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.007832] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.042018] pid_max: default: 32768 minimum: 301
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.046800] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.054027] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... cut ...]
Mon Apr 11 20:58:23 2022 authpriv.info dropbear[2702]: Exit before auth from <192.168.1.11:37700>: Timeout before auth
Mon Apr 11 20:58:49 2022 authpriv.info dropbear[2863]: Child connection from 192.168.1.11:38296
Mon Apr 11 20:58:50 2022 authpriv.notice dropbear[2863]: Password auth succeeded for 'root' from 192.168.1.11:38296
Mon Apr 11 20:58:56 2022 authpriv.info dropbear[2881]: Child connection from 192.168.1.11:38307
Mon Apr 11 20:58:59 2022 authpriv.notice dropbear[2881]: Password auth succeeded for 'root' from 192.168.1.11:38307
`

pepe2k, 2022-04-17:
@camelcamro You are using image for 32M board version:

`Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.000000] MIPS: machine is Zbtlink ZBT-WG3526 (32M)`

On a device with 16M fash:

```
Mon Apr 11 20:53:02 2022 kern.info kernel: [    0.624400] spi-nor spi0.0: w25q128 (16384 Kbytes)
...
Mon Apr 11 20:53:02 2022 kern.warn kernel: [    0.665194] mtd: partition "firmware" extends beyond the end of device "spi0.0" -- size truncated to 0xfb0000
```

This is wrong and that's the reason for tour problems.
You should use image for 16M.

Cheers,
Piotr

camelcamro, 2022-04-20:
well, i only wanted to tell you, that i tried 16 + 32MB hardware.
the 32MB version is defintiv not coming up and brings device only in a "bricked state", then only ROM reset working to set it back.

pepe2k, 2022-04-21:
> the 32MB version is defintiv not coming up and brings device only in a "bricked state", then only ROM reset working to set it back.

Show me please full dump from serial console, from the 32M version.
Testing image for 32M device on a 16M device doesn't make any sense.