openwrt/openwrt #9659: Mikrotik Hap Ac3 2.4 GHz radio does not work

djwujek, 2022-04-06:
The installation package is installed after it, the update also, but the 2.4 GHz radio is missing, an error pops up that the interface cannot be started.
5 Ghz radio works.

djwujek, 2022-04-06:
in the "Overview of the wireless networks status" tab in luci next to the "Qualcomm Atheros IPQ4019 802.11bgn" interface it says "Wireless network is not connected".
I do not know what information is required, I am doing it for the first time so sorry if there are any errors in the description.

neilsan1366, 2022-05-10:
hi what is your version 22.03 rc 1 ? people with 22.03 rc1 has a bug with wireless 2.4 in version 

please make cat /etc/config/wireless 

thanks