openwrt/openwrt #9663: MWAN3 [feature request] support to get IPSET also used for source or/and destination

camelcamro, 2022-04-06:
as name is saying, it would be great, if ipset can be used for source IP as well.
currently only what i read in manual, it is only for destination address.
but specailly to handle the local traffic, it is needed to use the IPSET also for source address ..
i guess it would be not that big deal to add this into, and adding another new option in settings file, to use:
eg: on an ipset rule ...

option ipset="SOURCE | DESTINATION" and if not existing, then destination as default)

camelcamro, 2022-04-06:
for now i made a hack for myself ...
so, that i can use for SRC ipset a ipset rule, which is starting with "IPS_I_" like: "**IPS_I_**BLOCKINTERNAL"
in /lib/mwan3/mwan3.sh

	#if [ -n "$ipset" ]; then
	#	ipset="-m set --match-set $ipset dst"
	#fi
	#camel - source or destination selection
	if [ -n "$ipset" ]; then
		if [[ "$ipset" =~ ^IPS_I_.* ]]; then
		#if [ if [ "${ipset:0:6}" = IPS_I_ ]; then
			ipset="-m set --match-set $ipset src"
			#logger "mwan3 - CAMEL ipset=$ipset" 
		else
			ipset="-m set --match-set $ipset dst"
		fi
	fi

pepe2k, 2022-04-06:
@camelcamro this package is not part of the core packages, it's part of the community maintained packages. Please report the problem here: https://github.com/openwrt/packages

closed 2022-04-06 by pepe2k