openwrt/openwrt #9666: dropbear failed to bind on multiple interfaces

kouhj, 2022-04-06:
The dropbear has a nice config option to support multiple interfaces, such as:

```
config dropbear
	option PasswordAuth 'on'
	option RootPasswordAuth 'on'
	list Interface 'lan'
	list Interface 'lan2'
```

    
The  `service_trigger()` function of `/etc/init.d/dropbear` contains the following snippet to monitor mulitple interfaces parsed from config:

```
	[ -n "${interfaces}" ] && {
		for n in $interfaces ; do
			procd_add_interface_trigger "interface.*" $n /etc/init.d/dropbear reload
		done
	}
```

And the code below in `/etc/init.d/dropbear` seems to try to load all the IP addresses of all the interfaces, however it will fail with error message  `interface lan lan2 has no physdev or physdev has no suitable ip`

```
dropbear_instance()
{
	local ipaddrs

	[ "$2" = 0 ] || {
		echo "validation failed"
		return 1
	}

	[ -n "${Interface}" ] && {
		[ -n "${BOOT}" ] && return 0

		network_get_ipaddrs_all ipaddrs "${Interface}" || {
			echo "interface ${Interface} has no physdev or physdev has no suitable ip"
			return 1
		}
	}
	...
```

It seems that `network_get_ipaddrs_all ()` only supports one interface at a time.  
So the patch to this issue could loop through all the interfaces:

```
--- /etc/init.d/dropbear.orig
+++ /etc/init.d/dropbear.new
@@ -134,7 +134,7 @@
 
 dropbear_instance()
 {
-	local ipaddrs
+	local ipaddrs if_ipaddrs
 
 	[ "$2" = 0 ] || {
 		echo "validation failed"
@@ -144,10 +144,13 @@
 	[ -n "${Interface}" ] && {
 		[ -n "${BOOT}" ] && return 0
 
-		network_get_ipaddrs_all ipaddrs "${Interface}" || {
-			echo "interface ${Interface} has no physdev or physdev has no suitable ip"
-			return 1
-		}
+		for n in ${Interface}; do
+			network_get_ipaddrs_all if_ipaddrs $n || {
+				echo "interface $n has no physdev or physdev has no suitable ip"
+				return 1
+			}
+			ipaddrs="$ipaddrs $if_ipaddrs"
+		done
 	}
 
 	[ "${enable}" = "0" ] && return 1

```

rockdrilla, 2022-11-14:
Hi there!

https://github.com/openwrt/openwrt/blob/c984fc762412f206ebeb3e873742988ff760fb90/package/network/services/dropbear/files/dropbear.init#L120

I suppose that "`Interface`" is type of string and holds name of a (single) interface hovewer your patch looks good.

dhewg, 2022-11-15:
As an alternative it supports multiple instances, and that's what luci supports too:

```
config dropbear 'lan'
    option Interface 'lan'
    option Port '22'
    option PasswordAuth 'off'
    option RootPasswordAuth 'off'

config dropbear 'wg0'
    option Interface 'wg0'
    option Port '22'
    option PasswordAuth 'off'
    option RootPasswordAuth 'off'
```

I didn't test if your approach works just as well, but maybe there's a reason there're multiple instances instead of multiple interfaces per instance?

rockdrilla, 2022-11-18:
@kouhj, @pepe2k, please take a look at https://github.com/openwrt/openwrt/pull/11223.

rockdrilla, 2024-01-09:
#11223 introduces new option `DirectInterface` which takes precedence over `Interface`. Specifying both is likely an error because `Interface` value will be ignored completely; however there's no error while network interface specified by `DirectInterface` is up. Does it fit in your case?

commit 3f96246e (2024-02-09): dropbear: better handle interfaces

- introduce 'DirectInterface' option to bind exactly to specified interface;
  fixes #9666 and late IPv4/IPv6 address assignment
- option 'DirectInterface' takes precedence over 'Interface'
- improve interface/address handling,
  e.g. verify count of listening endpoints due to dropbear limit (10 for now)
closed 2024-02-09 by openwrt-bot