openwrt/openwrt #9676: [nftables] Very poor performance with nftables under an ASUS RT-AC85P

howl, 2022-04-07:
I have seen this sometime ago, but left unreported because I thought development was too early, but now the issue is not only in master but also in the 22.03 branch. With ASUS RT-AC85P router, ramips/mt7621, nftables makes the connection to work erratically and also very very slow.

Using iptables I can get the total bandwidth I have and the connection is stable. With nftables the speed slows down to over 80/15 and the connection behaves erratically partially loading things, sometime hanging...

I'm pretty sure is because nftables because I used the master branch because some mt76 issues present with mt7615 to see if latest development fixes it, so when I first notice this I tested the last iptables and the first nftables snapshots.

SirDMZ, 2022-04-07:
Curious to see if you still observe this under the 5.15 kernel.

howl, 2022-04-21:
Me too but I don't have actually enough free space to compile myself, have to wait to promote from testing kernel in snapshots.

I have just test 22.03.0-rc1 I have reached 100/15 speed with lockups. I think this release is going to be buggy because fw4 issues, at least with mt7621, or at least with this router specifically.

abajk, 2022-05-07:
@howl @SirDMZ 
Take a look at [this comment](https://github.com/openwrt/openwrt/issues/9848#issuecomment-1120198182) and let us know if it improves things for you.

howl, 2022-05-12:
I had no flowtable ft defined so I created one with wan, lan1, lan2, lan3, lan4, and the wifi ifaces I have. Still the same, no improve at all.

As you can see the device is using DSA switch and the internet connection is PPPoE.

howl, 2022-05-12:
Have tried with wan vlan and pppoe-wan instead of wan but same result.

howl, 2022-05-12:
Ok, I now see why I didn't had previously defined the flowtable. This bug report is without any type of offloading. I have problems with flow_offload even software and ipv6 connectivity, where periodically it get freeze while ipv4 continues to work with fw3.

I have set up flowoffload only software and restarted the firewall and I get (without adding or deleting anything):
```
table inet fw4 {
	flowtable ft {
		hook ingress priority filter
		devices = { br-lan, pppoe-wan }
	}
}
```

And if I set also hw offloading and restar the firewall I get (without adding or deleting anything):
```
table inet fw4 {
	flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, lan3, lan4, wan }
		flags offload
	}
}
```

It doesn't work right no matter what I set.

I don't used offload with fw3 and can get 300/300 while with fw4 also no flow offload only 80-90/15-20.

howl, 2022-05-14:
For the record, I have drop invalid packets enabled, but despite I'm not 100% sure, I think the first time I tested nftables, I haven't.

howl, 2022-05-16:
I have acquired a mediatek/mt7622 device, no flow offloading and it works, have to test download speed halting other devices but I could get 200/300 stable.

It seems that is device dependant, perhaps the dsa modules?

howl, 2022-05-26:
I have tested actual 22.03-snapshot and snapshot, still there.

howl, 2022-05-27:
Forget to mention, this last tests I have done was with ethernet cable from ASUS RT-AC85P WAN to another roter that work right LAN and this router is the one that is with PPPoE. The first test I did with this setup was with 21.02 and the speed was all the bandwitdh available. The next tests with 22.03 and snapshots over 90/15, so the issue is also with no PPPoE connection.

howl, 2022-05-27:
I have tested with 2.4 GHz radio, it seems to work a lot better, don't know what is the maximum bandwidth that could be obtained with 21.02 but I can get 145/175 with snapshot.

It seems to be something with the 5 GHz radio because also I have set up phy0tpt and phy1tpt that are not by default set in the led triggers and both phy0tpt (2.4 GHz) and phy1tpt (5 GHz) are blink are blinking even with no stations connected, the 5 GHz is almost all the time blinking. Don't know if this could be important.

Will try to test with 21.02.

howl, 2022-05-28:
I have done tests with 21.02 and 22.03-snapshot. The issue seems to be in fact in mt76 with the 5 GHz radio in a MT7615E as with 21.02.3 works like it should, ethernet, wifi 2.4 GHz and wifi 5 GHz. In 22.03 ethernet and wifi 2.4 GHz works, but wifi 5 GHz fails.

I'm going to close this and open a bug in mt76.

closed 2022-05-28 by howl