openwrt/openwrt #9678: Linksys WRT1900ACS v2 - 21.02 - Unable to activate VLAN

Berbe, 2022-04-08:
_Originally posted on the [forum](https://forum.openwrt.org/t/linksys-wrt1900acs-v2-swconfig-to-dsa-unable-to-reimplement-configuration/124375)
I am submitting this here as it feels it might be a bug and not a mere configuration error_

I switched from v19.07 to v21.02 on my Linksys WRT1900ACS v2, and I am now trying to reimplement VLANs I was using on the old system on the new one.

I must point out I read:
* [DSA Mini-Tutorial](https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial)
* [Converting to DSA](https://openwrt.org/docs/guide-user/network/dsa/converting-to-dsa)

The conversion page takes the example of the basic VLAN layout in swconfig and [explains how to implement it in DSA](https://openwrt.org/docs/guide-user/network/dsa/converting-to-dsa#swconfig_dsa_vlan_configuration_comparison).
The steps are:
1. Adding the `wan` `Network device` to the `br-lan` `Bridge device`
2. Adding `lan[1-3]` ports (`lan4` is excluded, belonging to another `Bridge device`) to VLAN №1 as `untagged`, `Primary VLAND ID`
3. Adding `wan` port to VLAN №2 as `untagged`, `Primary VLAND ID`
4. Assigning `br-lan.1` VLAN `Device` to `lan` `Interface`

Here are the changes recorded before attempting to apply them:
```
uci add network bridge-vlan # =cfg0da1b0
uci set network.@bridge-vlan[-1].device='br-lan'
uci set network.@bridge-vlan[-1].vlan='1'
uci add_list network.@bridge-vlan[-1].ports='lan1:u*'
uci add_list network.@bridge-vlan[-1].ports='lan2:u*'
uci add_list network.@bridge-vlan[-1].ports='lan3:u*'
uci add network bridge-vlan # =cfg0ea1b0
uci set network.@bridge-vlan[-1].device='br-lan'
uci set network.@bridge-vlan[-1].vlan='2'
uci add_list network.@bridge-vlan[-1].ports='wan:u*'
uci del network.cfg030f15.ports
uci add_list network.cfg030f15.ports='lan1'
uci add_list network.cfg030f15.ports='lan2'
uci add_list network.cfg030f15.ports='lan3'
uci add_list network.cfg030f15.ports='wan'
uci set network.lan.device='br-lan.1'
```
However, whenever I try to set this up through LuCI, the configuration fails to apply.

Loads of kernel log messages (`dmesg`) coming from the `mv88e6085 f1072004.mdio-mii:00` facility.
Here are the ones catching my attention:
```
[34215.573306] mv88e6085 f1072004.mdio-mii:00: p0: hw VLAN 1 already used by port 1 in br-lan
[34215.614448] mv88e6085 f1072004.mdio-mii:00 lan4: failed to initialize vlan filtering on this port
```
IIRC, with `swconfig`, IIRC ports 0 & 1 were the 2 CPUs.
I got 2 devices:
* `br-lan` on ports `lan[1-3]` (+ `wan` for the needs of VLAN, following the conversion instructions from `swconfig`), used by a `lan` interface
* `br-lanp` on port `lan4` used by a `lanp` interface

I did not set VLAN up on the `br-lanp`, so why is anything attempted with `lan4`?

The conflict on the CPU confuses me too.
The `lan` interface is being used by 2 wireless ESSID, one on each radio (`radio0` 802.11nac & `radio1` 802.11bgn).
Might be where the struggle comes from?

neoxic, 2022-04-14:
_Originally posted to the same [forum thread](https://forum.openwrt.org/t/linksys-wrt1900acs-v2-swconfig-to-dsa-unable-to-reimplement-configuration/124375/5)._

I find myself in a similar situation as I've just updated my Linksys WRT3200ACM to OpenWrt 21.02.2 and have been playing around with the DSA configuration.

> br-lanp on port lan4 used by a lanp interface

Based on what I've seen so far, this is the core of the problem. It seems there's no way to get two bridges running at the same time for some weird reason despite the fact that it's documented in the Mini DSA Tutorial. This is what happens when I add another bridge named `br-test`:

```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option device 'br-lan'

config bridge-vlan
	option device 'br-lan'
	list ports 'lan1'
	option vlan '1'

config bridge-vlan
	option device 'br-lan'
	option vlan '100'
	list ports 'lan3'
	list ports 'lan4'

config device
	option type 'bridge'
	option name 'br-test'
	list ports 'lan2'

config interface 'test'
	option proto 'static'
	option ipaddr '192.168.2.1'
	option netmask '255.255.255.0'
	option device 'br-test'
```

The second bridge is always down:
```
# ip addr show dev br-test
24: br-test: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN qlen 1000
    link/ether ae:98:88:e5:de:c8 brd ff:ff:ff:ff:ff:ff
```
dmesg:
```
[68519.261382] mv88e6085 f1072004.mdio-mii:00 lan2: configuring for phy/gmii link mode
[68519.274945] 8021q: adding VLAN 0 to HW filter on device lan2
[68519.328836] br-test: port 1(lan2) entered blocking state
[68519.334175] br-test: port 1(lan2) entered disabled state
[68519.340115] device lan2 entered promiscuous mode
[68519.345885] mv88e6085 f1072004.mdio-mii:00: p2: hw VLAN 1 already used by port 3 in br-lan
[68519.387664] mv88e6085 f1072004.mdio-mii:00 lan2: failed to initialize vlan filtering on this port
[68519.408097] br-test: port 1(lan2) entered blocking state
[68519.413435] br-test: port 1(lan2) entered disabled state
[68519.420404] mv88e6085 f1072004.mdio-mii:00: p2: hw VLAN 1 already used by port 3 in br-lan
[68519.467691] mv88e6085 f1072004.mdio-mii:00 lan2: failed to initialize vlan filtering on this port
[68519.488110] br-test: port 1(lan2) entered blocking state
[68519.493448] br-test: port 1(lan2) entered disabled state
[68519.500430] mv88e6085 f1072004.mdio-mii:00: p2: hw VLAN 1 already used by port 3 in br-lan
[68519.537715] mv88e6085 f1072004.mdio-mii:00 lan2: failed to initialize vlan filtering on this port
[68519.558155] br-test: port 1(lan2) entered blocking state
[68519.563491] br-test: port 1(lan2) entered disabled state
[68519.570477] mv88e6085 f1072004.mdio-mii:00: p2: hw VLAN 1 already used by port 3 in br-lan
[68519.617740] mv88e6085 f1072004.mdio-mii:00 lan2: failed to initialize vlan filtering on this port
[68519.744969] mv88e6085 f1072004.mdio-mii:00 lan2: configuring for phy/gmii link mode
[68519.758069] 8021q: adding VLAN 0 to HW filter on device lan2
[68519.811676] br-test: port 1(lan2) entered blocking state
[68519.817016] br-test: port 1(lan2) entered disabled state
[68519.824131] mv88e6085 f1072004.mdio-mii:00: p2: hw VLAN 1 already used by port 3 in br-lan
[68519.847837] mv88e6085 f1072004.mdio-mii:00 lan2: failed to initialize vlan filtering on this port
[68519.868324] br-test: port 1(lan2) entered blocking state
[68519.873664] br-test: port 1(lan2) entered disabled state
[68519.880803] mv88e6085 f1072004.mdio-mii:00: p2: hw VLAN 1 already used by port 3 in br-lan
[68519.917844] mv88e6085 f1072004.mdio-mii:00 lan2: failed to initialize vlan filtering on this port
[68519.938280] br-test: port 1(lan2) entered blocking state
[68519.943616] br-test: port 1(lan2) entered disabled state
[68519.950603] mv88e6085 f1072004.mdio-mii:00: p2: hw VLAN 1 already used by port 3 in br-lan
[68519.997870] mv88e6085 f1072004.mdio-mii:00 lan2: failed to initialize vlan filtering on this port
[68520.018316] br-test: port 1(lan2) entered blocking state
[68520.023652] br-test: port 1(lan2) entered disabled state
[68520.030637] mv88e6085 f1072004.mdio-mii:00: p2: hw VLAN 1 already used by port 3 in br-lan
[68520.077896] mv88e6085 f1072004.mdio-mii:00 lan2: failed to initialize vlan filtering on this port
```
I don't really need VLANs to separate my two networks. Two bridges would work perfectly. But for the time being, the only way that works for me is one bridge with VLAN filtering enabled.

My guess is that two bridges somehow conflict with each other internally because I've seen a situation when after a reboot, the first bridge is down ("no link" in LUCI on all ports) instead of the second and vice versa. Probably, some kind of _internal VLAN_, i.e. VLAN 0, is used which then introduces a conflict in case of multiple bridges. Just a guess...

Berbe, 2022-07-25:
@neoxic Do you think this problem might be related to #7077 ?

neoxic, 2022-07-25:
@Berbe I'm quite sure it is the same problem essentially.

LucaRood, 2022-07-26:
I'm fairly sure this (at least what was described in @neoxic's reply) is also the same as #10001 (where I recently replied).

One thing I forgot to mention there, which might be relevant, is that when one bridge is created with some ports, all the other remaining ports do still work individually (at least on my WRT1200AC with 21.02.3 r16554-1d4dea6d4f). So one interface with a bridge device, plus n individual interfaces for the remaining ports works fine. It is only when another bridge device is introduced that the issue appears. So the issue definitely looks to be at the bridge level.

I figured I'd mention that, in case it helps to narrow down where to look.

BootsSiR, 2022-08-21:
I've been fighting to setup two VLAN's on my openwrt device because my ISP requires one VLAN for internet traffic and another for IPTV.  I have my internet traffic working fine, but whenever I try to setup the second VLAN for IPTV I hit the 'pX: already a member of VLAN 1' message and it won't work.  I'm guessing it's this issue here and there's no way I can use openwrt with my ISP?

neoxic, 2022-08-21:
@BootsSiR If you are ok with setting up your LAN without a bridge device (`br-lan`) and using an external switch, you can save the bridge for your WAN interface. In other words, since the bridge is going to be used for VLANs on WAN, you will end up living with only one physical LAN port, i.e. `lan1` for example, that must be connected to an external switch.

BootsSiR, 2022-08-21:
@neoxic if this is something that will be fixed I can probably just stay with my old router until a fix is in.  Think it's worth trying the latest [22.03.0-rc6](https://downloads.openwrt.org/releases/22.03.0-rc6/) release?

neoxic, 2022-08-22:
@BootsSiR I'm still on 21.02, so can't really tell. But as far as I'm concerned, this issue and #7077 are still open, so I wouldn't hold out much hope.

BootsSiR, 2022-08-22:
@neoxic yeah, same outcome on 22.03.0 rc6 as expected.  If I left my one working VLAN for my WAN connection and moved my two IPTV boxes to an external managed switch and had it handle the VLAN tagging, that should get me up and running right?  Openwrt would just pass on whatever tagging the managed switch did?

Berbe, 2022-08-22:
I do not have the expertise to decide, but if indeed 3 tickets are related to the same problem, 2 of them shall be closed.
Are :
* #9678 (this one)
* #10001

the same as #7077?
@neoxic @LucaRood

Some triage might be needed from people with expertise @pepe2k... and hopefully people to address the issue at hand of course (o:

BootsSiR, 2022-08-22:
@neoxic oh wow... I never thought of that setup.  I will try that and let you know if that does the trick for me.  Thank you for sharing that!

BootsSiR, 2022-08-23:
@neoxic Everything is running now using the setup you describe!  Is there anything I have to do differently security/firewall wise where my WAN port has been added to br-lan ?

BootsSiR, 2022-08-23:
@neoxic br-lan-n-wan coming right up! 🤣