openwrt/openwrt #9684: error: recursive dependency detected!

zfgeng, 2022-04-09:
zfgeng@LAPTOP-R6CB4TRA:~/openwrt$ make menuconfig
Collecting package info: done
tmp/.config-package.in:102171:error: recursive dependency detected!
tmp/.config-package.in:102171:  symbol PACKAGE_iptables-legacy is selected by PACKAGE_iptables-nft
tmp/.config-package.in:102978:  symbol PACKAGE_iptables-nft is selected by PACKAGE_opennds
tmp/.config-package.in:101201:  symbol PACKAGE_opennds depends on PACKAGE_iptables-legacy
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

kyoto44, 2022-04-10:
Got the same error on AX6S...

dchard, 2022-04-12:
Exactly the same on my end:

```
tmp/.config-package.in:102820:error: recursive dependency detected!
tmp/.config-package.in:102820:  symbol PACKAGE_iptables-legacy is selected by PACKAGE_iptables-nft
tmp/.config-package.in:103627:  symbol PACKAGE_iptables-nft is selected by PACKAGE_opennds
tmp/.config-package.in:101850:  symbol PACKAGE_opennds depends on PACKAGE_iptables-legacy
```

closed 2022-04-19 by zfgeng