openwrt/openwrt #9686: fw4 / https-dns-proxy non-existent rule jump target

oldium, 2022-04-09:
Git main. The combination of https-dns-proxy and fw4 creates invalid NFT rules:

```
#> ubus call service get_data
{
        "https-dns-proxy": {
                "instance1": {
                        "firewall": [
                                {
                                        "type": "redirect",
                                        "target": "DNAT",
                                        "src": "lan",
                                        "proto": "tcp udp",
                                        "src_dport": "53",
                                        "dest_port": "53",
                                        "reflection": false
                                },
                                {
                                        "type": "rule",
                                        "src": "lan",
                                        "dest": "*",
                                        "proto": "tcp udp",
                                        "dest_port": "853",
                                        "target": "REJECT"
                                }
                        ]
                }
        }
}
```

creates the following NFT rules by fw4:

```
#> fw4 print
...
        chain forward_lan {
                tcp dport 853 counter jump reject_from_lan comment "!fw4: ubus:https-dns-proxy[instance1] rule 1"
                udp dport 853 counter jump reject_from_lan comment "!fw4: ubus:https-dns-proxy[instance1] rule 1"
                jump accept_to_shared comment "!fw4: Accept lan to shared forwarding"
                jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
                ct status dnat accept comment "!fw4: Accept port forwards"
                jump accept_to_lan
        }
...
```

The issue is that `reject_from_lan` chain is not created by fw4, it is missing in `fw4 print`. I didn't want to send the full config, so if there is anything important missing, please tell me.

oldium, 2022-04-09:
CC @jow-

dave14305, 2022-04-12:
A uci rule with the same parameters parses correctly.
```
config rule
        option name 'Drop DoT'
        option src 'lan'
        option dest '*'
        option proto 'tcp udp'
        option dest_port '853'
        option target 'REJECT'
```
as:
```
# nft list ruleset | grep 853
  tcp dport 853 counter packets 0 bytes 0 jump handle_reject comment "!fw4: Drop DoT"
  udp dport 853 counter packets 0 bytes 0 jump handle_reject comment "!fw4: Drop DoT"
```
Biggest difference is the jump chain.

jow-, 2022-04-12:
I think this is a logic flaw that got carried over from fw3. If a rule uses a wildcard destination and a source zone which has logging enabled, then a wrong chain is selected. @oldium - is your `lan` zone setting `option log 1` ?

dave14305, 2022-04-12:
I was able to reproduce when lan logging was enabled, and the fix does indeed allow the chain to be created. Just FYI until oldium responds.

oldium, 2022-04-13:
> If a rule uses a wildcard destination and a source zone which has logging enabled, then a wrong chain is selected. @oldium - is your `lan` zone setting `option log 1` ?

Yes, my `lan` zone has `option log 1`.

> In any case please try the following fix:

I can confirm that the patch works.

oldium, 2022-04-15:
I see that this is fixed in the latest firewall4 Git, so we can close this as soon as the Openwrt package is updated.

oldium, 2022-04-22:
Fixed by af02a12d7ccb64a3abbd9683dd22275b538f6d50. Closing.

closed 2022-04-22 by oldium