openwrt/openwrt #9689: IPv6-PD update fails with dynamic IPv6 address

Kornelius777, 2022-04-09:
Dear all,

my provider gives me a dynamic IPv6 address (dynamic /56 prefix)
I receive it on my first router, an AVM Fritz!Box 7490.

Behind this router, I have an OpenWrt firewall (21.02.2) which receives a /62 IPv6-PD.

When my /56 prefix changes, I expected the IPv6-PD to adjust automagically.

...which it doesn't

The only change is that wan6 receives an updated /128 address for itself.
However, the /62 IPv6-PD remains unchanged.

sebastian-de, 2022-04-10:
Have you tested this with a current master build?
There was a commit in January '22 that changed the handling of removed prefixes:
https://github.com/openwrt/odhcpd/commit/83e14f45581757a4ea1418a64d80885c79db5ef0

If I understand this change correctly this handles prefixes delegated by odhcpd (downstream), but I guess it's still worth testing.

sebastian-de, 2022-04-10:
This seems to be an issue with the DHCPv6 server on the Fritz!Box (I have a 7530), as this isn't specific to OpenWrt.
I tried this on a Debain machine today:

- request a prefix from the FB with `dhclient -d -6 -P -v eth0`
- trigger a reconnect of the DSL-line

Result: The prefix received by dhclient remains unchanged.

The DHCPv6 server on the FB doesn't seem to invalidate the outdated prefix. This should therefore be reported to AVM.

Kornelius777, 2022-04-10:
OK. I just opened a Ticket with AVM.

closed 2022-04-12 by pepe2k

sebastian-de, 2022-05-03:
@Kornelius777 did you receive a response from AVM, regarding this issue?

Kornelius777, 2022-05-04:
Well...
As in many companies, it was pretty tough to pass their 1st Level Support.
Making them understand where the problem is and that their standard solution number X will not solve it took quite a while.
Eventually, they passed me to 2nd Level (despite they don't use this term - currently I just hope that this is the case) and I was asked to answer some questions.
However, after my reply, I received a standard reply saying that this new person (who I expect being a 2nd Level Guy) will not be in the office for a while - but work on the issue after his return.
That's how far I got so far.
At this time, all I can do is wait and see...

Kind regards,

Kornelius

sebastian-de, 2022-05-04:
Thank you for sharing this. Let's hope they make a serious attempt to fix this.

Regards
Sebastian

Kornelius777, 2022-05-13:
Today's answer from AVM makes me hope for a timely release of the next firmware:

They say:

"Before my absence, I had already researched this topic in-house and found out that we had already made adjustments to the current FRITZ! laboratory version with regard to the IPv6 prefix delegation.  Based on the information available to me, this should also be available for the FRITZ!Box 7490 in a timely manner.  Since the development status in the laboratory is already quite advanced, an examination of the behavior you describe in version 7.29 is not expedient."

Let's keep fingers crossed!