openwrt/openwrt #9694: Regression: usb stopped working after upgrading to OpenWrt 21.02.2 r16495-bf0c965af0

sloncho, 2022-04-10:
Device: Netgear Nighthawk X4S R7800
Openwrt: OpenWrt 21.02.2 r16495-bf0c965af0

After firmware upgrade, usb devices are not discovered, it appears that the whole usb subsystem is not working.
```
# lsusb
unable to initialize libusb: -99
```

Packages:
```
# opkg list-installed | grep usb
kmod-usb-core - 5.4.179-1
kmod-usb-dwc3 - 5.4.179-1
kmod-usb-ehci - 5.4.179-1
kmod-usb-ledtrig-usbport - 5.4.179-1
kmod-usb-ohci - 5.4.179-1
kmod-usb-printer - 5.4.179-1
kmod-usb-storage - 5.4.179-1
kmod-usb-storage-extras - 5.4.179-1
kmod-usb2 - 5.4.179-1
kmod-usb3 - 5.4.179-1
libusb-1.0-0 - 1.0.24-4
usbids - 0.354-1
usbutils - 013-2
```

Kernel logs:
```
dmesg | grep usb
[    0.050232] usbcore: registered new interface driver usbfs
[    0.050315] usbcore: registered new interface driver hub
[    0.050384] usbcore: registered new device driver usb
[   13.045633] usbcore: registered new interface driver usb-storage
[   13.046160] usbcore: registered new interface driver usblp
[   13.103191] usbcore: registered new interface driver ums-alauda
[   13.107395] usbcore: registered new interface driver ums-cypress
[   13.148482] usbcore: registered new interface driver ums-datafab
[   13.148906] usbcore: registered new interface driver ums-freecom
[   13.154024] usbcore: registered new interface driver ums-isd200
[   13.159962] usbcore: registered new interface driver ums-jumpshot
[   13.165610] usbcore: registered new interface driver ums-karma
[   13.172843] usbcore: registered new interface driver ums-sddr09
[   13.177637] usbcore: registered new interface driver ums-sddr55
[   13.183509] usbcore: registered new interface driver ums-usbat
```

No `/dev/usb` either:
```
# ls /dev/u*
/dev/ubi0         /dev/ubi0_0       /dev/ubi0_1       /dev/ubi_ctrl     /dev/ubiblock0_0  /dev/urandom
```

The only device attached is a USB printer - and it used to work with V19 and earlier.

pepe2k, 2022-04-12:
@sloncho a whole `dmesg` output might be helpful here.

sloncho, 2022-04-13:
@pepe2k here it is:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.179 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16495-bf0c965af0)) #0 SMP Wed Feb 16 20:29:10 2022
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Netgear Nighthawk X4S R7800
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 122880
[    0.000000]   Normal zone: 1080 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 474824K/491520K available (6274K kernel code, 196K rwdata, 920K rodata, 1024K init, 239K bss, 16696K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from 0xc0b00c18 with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000008] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000248] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000268] pid_max: default: 32768 minimum: 301
[    0.000469] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000487] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001542] CPU: Testing write buffer coherency: ok
[    0.002690] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002872] rcu: Hierarchical SRCU implementation.
[    0.003368] smp: Bringing up secondary CPUs ...
[    0.005277] smp: Brought up 1 node, 2 CPUs
[    0.005292] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005301] CPU: All CPU(s) started in SVC mode.
[    0.015989] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.016136] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016162] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.016298] pinctrl core: initialized pinctrl subsystem
[    0.017428] NET: Registered protocol family 16
[    0.017718] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.019088] cpuidle: using governor ladder
[    0.019154] cpuidle: using governor menu
[    0.050224] usbcore: registered new interface driver usbfs
[    0.050306] usbcore: registered new interface driver hub
[    0.050371] usbcore: registered new device driver usb
[    0.050432] pps_core: LinuxPPS API ver. 1 registered
[    0.050444] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.050484] PTP clock support registered
[    0.052328] clocksource: Switched to clocksource dg_timer
[    0.053194] thermal_sys: Registered thermal governor 'step_wise'
[    0.055619] NET: Registered protocol family 2
[    0.055788] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.056678] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.056724] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.056767] TCP bind hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.056821] TCP: Hash tables configured (established 4096 bind 4096)
[    0.056933] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.056962] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.057172] NET: Registered protocol family 1
[    0.057229] PCI: CLS 0 bytes, default 64
[    0.058547] workingset: timestamp_bits=14 max_order=17 bucket_order=3
[    0.065753] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[... 345 lines cut ...]
[   45.508206] br-lan: port 2(2-4GHz-0) entered blocking state
[   45.513474] br-lan: port 2(2-4GHz-0) entered forwarding state
[   45.523677] br-guest: port 2(2-4GHz-1) entered blocking state
[   45.524935] br-guest: port 2(2-4GHz-1) entered disabled state
[   45.530917] device 2-4GHz-1 entered promiscuous mode
[   45.541644] ath10k_pci 0001:01:00.0: rts threshold -1
[   45.545421] br-guest: port 2(2-4GHz-1) entered blocking state
[   45.546280] br-guest: port 2(2-4GHz-1) entered forwarding state
[   45.637689] IPv6: ADDRCONF(NETDEV_CHANGE): 2-4GHz-1: link becomes ready
[... cut ...]
[  125.987114] br-guest: port 3(5GHz-1) entered blocking state
[  125.987151] br-guest: port 3(5GHz-1) entered disabled state
[  125.992020] device 5GHz-1 entered promiscuous mode
[  126.003513] ath10k_pci 0000:01:00.0: rts threshold -1
[  126.006532] br-guest: port 3(5GHz-1) entered blocking state
[  126.007554] br-guest: port 3(5GHz-1) entered forwarding state
[  126.069825] IPv6: ADDRCONF(NETDEV_CHANGE): 5GHz-1: link becomes ready
[  650.686544] ath10k_pci 0001:01:00.0: htt tx: fixing invalid VHT TX rate code 0xff
```

MuchHustle, 2022-07-07:
where is the response? he outputed all that and then nothing....? wtf? I have same problem.

sloncho, 2022-08-12:
@pepe2k any update on this? Thanks.

sloncho, 2022-09-13:
Upgraded to OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a

The problem still exists. No usb devices :(

sloncho, 2023-01-23:
Wondering if someone has some insights, that renders openwrt useless for me.

MuchHustle, 2023-05-28:
So is this project abandoned @pepe2k ?

sloncho, 2023-09-26:
The problem were these missing packages:
```
kmod-phy-qcom-ipq806x-usb
kmod-usb-dwc3-qcom
```

I guess these must be added to the main image for R7800.

Once installed, I can list the usb devices.