openwrt/openwrt #9695: tools/xz: fails to build on macOS

aparcar, 2022-04-11:
I'm seeing the following error message:

```
nclude-dirs -Wstrict-aliasing -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wstrict-prototypes -Wold-style-definition -Wmissing-prototypes -Wmissing-declarations -Wmissing-noreturn -Wredundant-decls -O2 -I/Users/user/src/openwrt/openwrt/staging_dir/host/include -o xzdec xzdec-xzdec.o xzdec-tuklib_progname.o xzdec-tuklib_exit.o  -L/Users/user/src/openwrt/openwrt/staging_dir/host/lib ../../src/liblzma/.libs/liblzma.a -lpthread -pthread
ld: warning: ignoring file ../../src/liblzma/.libs/liblzma.a, building for macOS-arm64 but attempting to link with file built for macOS-arm64
Undefined symbols for architecture arm64:
  "_lzma_code", referenced from:
      _uncompress in xzdec-xzdec.o
  "_lzma_stream_decoder", referenced from:
      _uncompress in xzdec-xzdec.o
  "_lzma_version_string", referenced from:
      _version in xzdec-xzdec.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[6]: *** [Makefile:516: xzdec] Error 1
make[6]: Leaving directory '/Users/user/src/openwrt/openwrt/build_dir/host/xz-5.2.5/src/xzdec'
make[5]: *** [Makefile:422: all-recursive] Error 1
make[5]: Leaving directory '/Users/user/src/openwrt/openwrt/build_dir/host/xz-5.2.5/src'
make[4]: *** [Makefile:616: all-recursive] Error 1
make[4]: Leaving directory '/Users/user/src/openwrt/openwrt/build_dir/host/xz-5.2.5'
make[3]: *** [Makefile:483: all] Error 2
make[3]: Leaving directory '/Users/user/src/openwrt/openwrt/build_dir/host/xz-5.2.5'
make[2]: *** [Makefile:37: /Users/user/src/openwrt/openwrt/build_dir/host/xz-5.2.5/.built] Error 2
make[2]: Leaving directory '/Users/user/src/openwrt/openwrt/tools/xz'
time: tools/xz/compile#8.20#5.57#15.75
    ERROR: tools/xz failed to build.
make[1]: *** [tools/Makefile:168: tools/xz/compile] Error 1
make[1]: Leaving directory '/Users/user/src/openwrt/openwrt'
make: *** [/Users/user/src/openwrt/openwrt/include/toplevel.mk:230: tools/compile] Error 2
```

neheb, 2022-04-12:
--enable-shared=no may be the issue in tools/xz/Makefile unfortunately.

aparcar, 2022-04-13:
Works, could you please elaborate what the issue is?

neheb, 2022-04-13:
I don't know fully. But I do know macOS doesn't like static linking for some reason. Maybe the issue is some linking flag.

aparcar, 2022-04-13:
Actually now `sed`  breaks, I wonder if @nbd168 is seeing similar issues...

```
time: tools/xz/compile#0.03#0.04#0.09
make[2]: Entering directory '/Users/user/src/openwrt/openwrt/tools/sed'
make  -C /Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8 SHELL="bash"
make[3]: Entering directory '/Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8'
make  all-recursive
make[4]: Entering directory '/Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8'
Making all in po
make[5]: Entering directory '/Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8/po'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8/po'
Making all in .
make[5]: Entering directory '/Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8'
gcc    -O2 -I/Users/user/src/openwrt/openwrt/staging_dir/host/include   -L/Users/user/src/openwrt/openwrt/staging_dir/host/lib  -o sed/sed sed/sed-compile.o sed/sed-debug.o sed/sed-execute.o sed/sed-mbcs.o sed/sed-regexp.o sed/sed-sed.o sed/sed-utils.o sed/libver.a lib/libsed.a    
ld: warning: ignoring file sed/libver.a, building for macOS-arm64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
ld: warning: ignoring file lib/libsed.a, building for macOS-arm64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
Undefined symbols for architecture arm64:
  "_Version", referenced from:
      _main in sed-sed.o
  "__obstack_begin", referenced from:
      _compile_program in sed-compile.o
  "__obstack_newchunk", referenced from:
      _compile_program in sed-compile.o
      _get_openfile in sed-compile.o
  "_copy_acl", referenced from:
      _closedown in sed-execute.o
  "_dfaalloc", referenced from:
      _compile_regex_1 in sed-regexp.o
  "_dfacomp", referenced from:
      _compile_regex_1 in sed-regexp.o
  "_dfaexec", referenced from:
      _match_regex in sed-regexp.o
  "_dfafree", referenced from:
      _match_regex in sed-regexp.o
  "_dfaisfast", referenced from:
      _match_regex in sed-regexp.o
  "_dfasuperset", referenced from:
      _match_regex in sed-regexp.o
  "_dfasyntax", referenced from:
      _compile_regex_1 in sed-regexp.o
  "_fwriting", referenced from:
      _ck_fflush in sed-utils.o
  "_init_localeinfo", referenced from:
      _main in sed-sed.o
  "_locale_charset", referenced from:
      _initialize_mbcs in sed-mbcs.o
  "_memrchr", referenced from:
      _match_regex in sed-regexp.o
  "_program_name", referenced from:
      _bad_prog in sed-compile.o
      _open_next_file in sed-execute.o
      _main in sed-sed.o
      _usage in sed-sed.o
      _panic in sed-utils.o
  "_rpl_getopt_long", referenced from:
      _main in sed-sed.o
  "_rpl_lstat", referenced from:
      _follow_symlink in sed-utils.o
  "_rpl_optarg", referenced from:
      _main in sed-sed.o
  "_rpl_optind", referenced from:
      _main in sed-sed.o
  "_rpl_re_compile_pattern", referenced from:
      _compile_regex_1 in sed-regexp.o
  "_rpl_re_search", referenced from:
      _match_regex in sed-regexp.o
  "_rpl_re_set_syntax", referenced from:
      _compile_regex_1 in sed-regexp.o
[... 58 lines cut ...]
  "_xzalloc", referenced from:
      _compile_program in sed-compile.o
      _process_files in sed-execute.o
      _line_copy in sed-execute.o
      _compile_regex in sed-regexp.o
      _main in sed-sed.o
      _follow_symlink in sed-utils.o
      _init_buffer in sed-utils.o
      ...
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [Makefile:2642: sed/sed] Error 1
make[5]: Leaving directory '/Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8'
make[4]: *** [Makefile:3105: all-recursive] Error 1
make[4]: Leaving directory '/Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8'
make[3]: *** [Makefile:2321: all] Error 2
make[3]: Leaving directory '/Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8'
make[2]: *** [Makefile:49: /Users/user/src/openwrt/openwrt/build_dir/host/sed-4.8/.built] Error 2
make[2]: Leaving directory '/Users/user/src/openwrt/openwrt/tools/sed'
time: tools/sed/compile#0.17#0.18#0.47
    ERROR: tools/sed failed to build.
make[1]: *** [tools/Makefile:168: tools/sed/compile] Error 1
make[1]: Leaving directory '/Users/user/src/openwrt/openwrt'
make: *** [/Users/user/src/openwrt/openwrt/include/toplevel.mk:230: tools/sed/compile] Error 2
```

neheb, 2022-04-13:
```
ld: warning: ignoring file lib/libsed.a, building for macOS-arm64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
```

yeah... I would look and see if homebrew has a solution for this. Sounds like bad linker flags.

jonasjelonek, 2022-09-11:
Received my new Mac a few days ago and encountered the same problem when trying to build a fresh OpenWrt from latest trunk. As @neheb suggested, changing the Makefile and removing the `--enable-shared=no` helps to push compilation a bit further until sed fails.

Is there any other solution yet or is this still an unsolved problem?

jonasjelonek, 2022-09-11:
received my new M1 Mac a few days ago and tried to build OpenWrt from latest trunk. I initially encountered the same issue as @aparcar. Finally I was able to solve this. I read in another thread that binutils may be the issue, and indeed it was in my case. I had binutils installed via brew. 
After running `brew uninstall binutils`, `make clean`, compilation of the tools works just fine.

Hope this helps!

aparcar, 2022-09-11:
Excellent!

closed 2022-09-11 by aparcar