openwrt/openwrt #9701: ksmbd-server-3.4.2-2 no access on OpenWrt-21.02.2

erdoukki, 2022-04-11:
```
root@LS421DE:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="21.02.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.2"
VERSION_ID="21.02.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16495-bf0c965af0"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3-d16"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.2 r16495-bf0c965af0"
```

```
root@LS421DE:~# uname -ar
Linux LS421DE 5.4.179 #0 SMP Wed Feb 16 20:29:10 2022 armv7l GNU/Linux
```

```
root@LS421DE:~# /usr/sbin/ksmbd.mountd --version
ksmbd-tools version : 3.4.2
```

```
root@LS421DE:~# opkg status ksmbd-server
Package: ksmbd-server
Version: 3.4.2-2
Depends: libc, kmod-fs-ksmbd, libnl-core200, libnl-genl200
Status: install user installed
Architecture: arm_cortex-a9_vfpv3-d16
Conffiles:
 /etc/config/ksmbd 9cf43e11524752a960a4bc129dd251d94a34bd0acfd336e8f7e7d68f33d3bb68
 /etc/ksmbd/smb.conf.template f9b45aa86ea66fef794917c1e6ead43e5cafb5cdaba14a57c21e18f848aaa8a3
Installed-Time: 1649695050
```

```
root@LS421DE:~# /usr/sbin/ksmbd.mountd --config=/etc/ksmbd/smb.conf --nodetach
[ksmbd-worker/3281]: ERROR: Cannot resolve netlink family
[ksmbd-worker/3281]: ERROR: Cannot resolve netlink family
[ksmbd-worker/3281]: ERROR: Cannot resolve netlink family
[ksmbd-worker/3281]: ERROR: Cannot resolve netlink family
[ksmbd-worker/3281]: ERROR: Cannot resolve netlink family
[ksmbd-worker/3281]: ERROR: Cannot resolve netlink family
[ksmbd-worker/3281]: ERROR: Cannot resolve netlink family
...
```

But no access from remote !

erdoukki, 2022-04-11:
I have done some tests on a working NAS, which needed a downgraded and now do not more work !

```
root@NSA310S:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.2'
DISTRIB_REVISION='r16495-bf0c965af0'
DISTRIB_TARGET='kirkwood/generic'
DISTRIB_ARCH='arm_xscale'
DISTRIB_DESCRIPTION='OpenWrt 21.02.2 r16495-bf0c965af0'
DISTRIB_TAINTS=''
root@NSA310S:~# uname -ar
Linux NSA310S 5.4.179 #0 Wed Feb 16 20:29:10 2022 armv5tel GNU/Linux
```

The 3.4.4 was working well before my tests.
The test corrupt my init.d and I remove and downgrade with downloaded files (opkg get lost and do not apply the upgrade / reinstall) now downgraded package no more work at all !

erdoukki, 2022-04-11:
I have also some problem with the 3.4.4 on the LS421DE, but the NSA310S worked well with 3.4.4.
Badly, I have not saved the packages locally for more tests...
I suppose the older archives are lost and replaced by daily built ?

I will try from a local compilation.

tapper82, 2022-04-11:
You reverted. in the rong branch. It's still broken in master.

neheb, 2022-04-11:
He's on 21.02...

erdoukki, 2022-04-14:
Get some, (I beg your pardon, a *lot*...) of strange issues !
But mainly, I can say it is no more working !!!

I opened an issue here at OpenWrt, but it is a package issue.
I will update the next steps on a new issue there : 
[ksmbd-tools-3.4.2 not working on OpenWrt-21.02.2 #18303](https://github.com/openwrt/packages/issues/18303)

closed 2022-04-14 by erdoukki