openwrt/openwrt #9708: base-files: issues in nand/ubi sysupgrade

Lanchon, 2022-04-12:
here:

https://github.com/openwrt/openwrt/blob/bea01fa57f5c9c333138bbbc5c9f83b9d7553fb5/package/base-files/files/lib/upgrade/nand.sh#L156-L165

1) why remove ubiblock devices for volumes within ubi devices other than `$ubidev`?
2) why remove ubiblock devices for volumes other than `$kern_ubivol`, `$root_ubivol`, `$data_ubivol`?
3) fails for ubi vol numbers above 9, which can happen if `$ubidev` contains other volumes.

https://github.com/openwrt/openwrt/blob/bea01fa57f5c9c333138bbbc5c9f83b9d7553fb5/package/base-files/files/lib/upgrade/nand.sh#L145

4) should check for valid `$ubidev` after that line. a failure probably can't happen, but due to the nature of what's going on, it doesn't hurt to verify.

ptpt52, 2022-04-13:
What issue did you actually encounter?

dangowrt, 2022-04-13:
Regarding 1. and 2.: Removing **all** ubiblock devices at this point shouldn't hurt (or does it? if so: please explain why.).

Regarding 3.: Good catch. I'll replace `/dev/ubiblock*_?` with `/dev/ubiblock${ubidev:3}_*` which should then really match **all ubiblocks on that ubi device**, then filter out the partition devices (which were not supposed to be matched here, hence the use of `?` in the original statement) using an additional `case ... esac`.

Regarding 4.: I'll add a check and abort if ubidev (for which ever hard to imagine reason) still doesn't exist at this point.

closed 2022-04-13 by jow-

Lanchon, 2022-04-13:
i had already fixed all this

im working on revamping this code, i just hadn't issue a PR yet.

Lanchon, 2022-04-13:
my WIP is here:

https://github.com/Lanchon/openwrt/commits/nand-sysupgrade

Lanchon, 2022-04-13:
i am working on a safer sysupgrade. problem is... project members ask for a PR, then leave it unreviewed :(

Lanchon, 2022-04-13:
should i even work on this? the idea is that the device recovers from an interrupted sysupgrade

dangowrt, 2022-04-13:
I like the idea of writing rootfs first when doing using sysupgrade.tar and would immediately merge a PR doing **just that**.
In order to get reviewers it is important to keep your changes atomic, ie. do one thing at a time.

Lanchon, 2022-04-14:
"just that" is not enough. kernel-out-of-ubi devices are not affected then. and i want want to to provide SAFE updates:

1) load config to ram
2) rm config
3) new kernel vol written, new rootfs too, in the space allowed for config
4) atomic rename of kernel, rootfs
   now if an interrupted upgrade happens, you lost config but you didn't brick
5) recreate rootfs_data

all the while being compatible with devices without enough flash for safe upgrades

this is easy. next... allow for safe upgrades including config. which is not transparent because ubifs cannot be shrinked yet.

instead: a BL env var enables sysupgrade to reserve remaining half of the nand for root_fs data. then on next upgrade: try to get a new root_fs part there with kernel and rootfs. then switch all 3 together.

this is useful because we are getting into a stage when most devices will have enough nand to support this. and really safe upgrades is NOT too much to ask in 2022. 

in order to do this, the i have to do more than "just that". seen in the best light, the code is untidy. it is the result of people committing the least amount of lines to get their thing done. all with different styles, and sometimes without a full understanding of the codebase.

anyway; some cleanup of the code is IMHO required to improve it.

so "just that" commits wont get us far... which might be what the project wants, and that's of course fine.

dangowrt, 2022-04-14:
I'm not saying that any of this is wrong or should not be done. But it can't be all in one commit, sprayed with `sync`s and all that. You need to do it one by one, or at least in a series where each commit can be reviewed in a meaningful way.
If you want to "clean up", that's great, start if a PR/series doing that.

Lanchon, 2022-04-14:
btw, thanks getting back. im at a party right now. at my place; a friend is leaving the country so i told them all to come over and say goodbye. (i might not be making strict sense tonight.) i'll retry commits later. the problem for me is: i can't test hardware that i do not own (kernel-out-of-ubi, r/w rootfs, etc). i think the project prefers not to fix whats working. so i will only PR stuff i can fully test. thanks again!

dangowrt, 2022-04-14:
Yes, exactly this is the main problem with those existing code paths -- nobody got all the hardware to test changes.
Enjoy the party!

commit 6fe3852d (2022-04-15): base-files: more robust sysupgrade on NAND

Make sure sysupgrade on NAND also works in case of UBI volumes having
index >9. While at it, also make sure UBI device is detected and abort
in case it isn't. Use Shell built-in shorthand ':' instead of 'true'.

Fixes #9708

(cherry picked from commit 0dbca1b2baf9c26514b517a1e5860f6a5b04d5ae)