openwrt/openwrt #9711: DHCPv6 dynamic DS-lite interface not active, no IPv4 connectivity

Matthias84, 2022-04-13:
Hi, I use OpenWrt 21.02.2 on a Ubiquiti EdgeRouter X SFP with a fiber SFP module to drive my FTTH uplink. dslite package added.
I tried it using a dynamic approach that creates virtual interfaces on demand, so only the first one needs to be setuped manually:
* WAN - PPPoE, login credentials, eth5 (SFP module), WAN zone
* WAN6 - auto created DHCPv6 client interface, WAN zone
* WAN6_4 - auto created dslite RFC6333 interface

But with this setup you only get an working IPv6 upstream, but no tunneling for IPv4 packages via dslite. The WAN6_4 package counter stays at 0 / 0 and there is no config options provided and the interface symbol is on grey background. Within system log you only see repeating:
> Wed Apr 13 18:15:45 2022 daemon.warn dnsmasq-dhcp[14718]: DHCP packet received on eth5 which has no address
> Wed Apr 13 18:15:47 2022 daemon.notice netifd: Interface 'wan6_4' is now down
> Wed Apr 13 18:15:47 2022 daemon.notice netifd: Interface 'wan6_4' is setting up now
> Wed Apr 13 18:15:50 2022 daemon.notice netifd: Interface 'wan6_4' is now down
> ...

So it seems that there are DHCPv6 B4 autodiscovery messages that trigger the creation of the dslite tunnel, but it doesn't contain / fit the necessary options to setup the AFTR tunnel endpoint, or the parameters aren't passed properly? 

The link configuration seems to confirm this:

> root@OpenWrt:~# ip link
> ....
> 9: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN qlen 1000
>     link/tunnel6 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00 brd 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

If I can provide any further information to help on debugging, please let me know!
It look like related to the old #1574 ticket but there was no real solution, just a similar workaround ...
Maybe you just need to create firewall rules or do some tweaks to create DHCP forwardings?

###  Workaround

I manually created also the 2nd and 3rd interfaces and referred the corresponding devices. I needed to add the AFTR IPv6 address to the dslite interface, which I asked the ISP for. But of course setting all this fixed is not a good style ...

fda77, 2024-01-04:
I've a similar problem with an external modem and dslite connection (1+1 or vodafone in germany).
A `pppoe` interface creates an `dhcpv6` interface which creates an `dslite` interface and ipv4 comes never up.
After adding a  `unset AFTR`  into `/lib/netifd/dhcpv6.script` the automatic dslite interface is prevented. A manual configured dslite makes ipv4 working (as far as it is possivle with dslite)

fda77, 2024-01-04:
... after a litte debugging it seems to be a dns-problem. 
Adding the line `[ "$AFTR" == "ffmatr1.vodafone-ip.de" ] && AFTR="2a00:0:cff:44a0::1" ` 
replaces the aftr hostname with its ip. I'm using unbound which could have problems with ipv6-only as before dslite no ipv4 is available