openwrt/openwrt #9713: "mwifiex" can no longer be configured as an AP

eduperez, 2022-04-13:
OpenWrt 21.02.2, r16495-bf0c965af0 running on a Linksys WRT3200ACM.

Looks like the third radio on this device can no longer be configured as an AP.

```
root@router:~# opkg list-installed | fgrep mwifiex
kmod-mwifiex-sdio - 5.4.179+5.10.85-1-1
mwifiex-sdio-firmware - 20211216-1
```

This configuration worked on 19.07:

```
config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
	option hwmode '11g'
	option channel '11'
	option htmode 'HT20'
	option country 'FR'
	option disabled '1'

config wifi-iface '[REDACTED]'
	option device 'radio1'
	option network '[REDACTED]'
	option mode 'ap'
	option ssid '[REDACTED]'
	option encryption 'psk-mixed+aes'
	option key '[REDACTED]'
	option ifname '[REDACTED]'

```

However, on 21.02.2, this is shown on the logs when the interface is raised:

```
Wed Apr 13 23:10:06 2022 daemon.notice netifd: Wireless device 'radio2' is now up
Wed Apr 13 23:10:41 2022 daemon.notice netifd: Wireless device 'radio2' is now down
Wed Apr 13 23:10:42 2022 kern.info kernel: [945831.722966] mwifiex_sdio mmc0:0001:1: cannot register network device
Wed Apr 13 23:10:42 2022 daemon.notice netifd: radio2 (12718): Failed to create interface [REDACTED]
Wed Apr 13 23:10:42 2022 daemon.err hostapd: Failed to set beacon parameters
Wed Apr 13 23:10:42 2022 daemon.err hostapd: Failed to set beacon parameters
Wed Apr 13 23:10:42 2022 daemon.err hostapd: Failed to set beacon parameters
Wed Apr 13 23:10:42 2022 daemon.notice netifd: radio2 (12718): ip: SIOCGIFFLAGS: No such device
Wed Apr 13 23:10:42 2022 daemon.notice netifd: radio2 (12718): Interface 0 setup failed: IFUP_ERROR
Wed Apr 13 23:10:42 2022 daemon.notice netifd: Wireless device 'radio2' is now up
Wed Apr 13 23:10:43 2022 daemon.err hostapd: Failed to set beacon parameters
Wed Apr 13 23:10:44 2022 daemon.notice hostapd: send_auth_reply: send failed
Wed Apr 13 23:10:45 2022 daemon.err hostapd: Failed to set beacon parameters
Wed Apr 13 23:10:46 2022 daemon.notice hostapd: send_auth_reply: send failed
Wed Apr 13 23:10:47 2022 daemon.notice hostapd: handle_probe_req: send failed
Wed Apr 13 23:10:48 2022 daemon.notice hostapd: send_auth_reply: send failed
Wed Apr 13 23:10:49 2022 daemon.notice hostapd: handle_probe_req: send failed
Wed Apr 13 23:10:50 2022 daemon.notice hostapd: send_auth_reply: send failed
Wed Apr 13 23:10:52 2022 daemon.notice hostapd: handle_probe_req: send failed
Wed Apr 13 23:10:54 2022 daemon.notice hostapd: send_auth_reply: send failed

```

robimarko, 2022-04-14:
Can you try adding `option legacy_rates '1'`?

That is required for USB based 88W8997 for example

hnyman, 2022-04-14:
I feel that the whole radio2 with 88W8997 should be removed from WRT3200ACM as it easily causes trouble for mwlwifi.

eduperez, 2022-05-02:
> Can you try adding `option legacy_rates '1'`?
> 
> That is required for USB based 88W8997 for example

Thanks, but it did not work... same result.

eduperez, 2022-05-02:
> I feel that the whole radio2 with 88W8997 should be removed from WRT3200ACM as it easily causes trouble for mwlwifi.

Yes and no... it can cause trouble, and it also can be tamed; I used it for years, for the IoT devices that cannot connect to the other radio.

djaffre, 2022-11-30:
> > I feel that the whole radio2 with 88W8997 should be removed from WRT3200ACM as it easily causes trouble for mwlwifi.
> 
> Yes and no... it can cause trouble, and it also can be tamed; I used it for years, for the IoT devices that cannot connect to the other radio.

Hello,
I had the same trouble. And after testing many configuration i had success to enable the third radio in AP mode on 2.4Ghz.
I just check the checkbox enable 11b rate and the ap start. For me this workaround is suffisant because i had only 4 esp8266 from commercial product on this radio.

Bye

eduperez, 2022-12-01:
> Hello, I had the same trouble. And after testing many configuration i had success to enable the third radio in AP mode on 2.4Ghz. I just check the checkbox enable 11b rate and the ap start. For me this workaround is suffisant because i had only 4 esp8266 from commercial product on this radio.

Many, many thanks!!!

eduperez, 2022-12-02:
After a few hours, everything seems to be working as expected; however, I am seeing many messages like this one on the logs:
"mwifiex_sdio mmc0:0001:1: cmd_wait_q terminated: -512"

I am still not sure if those are harmless, or how to fix them.