openwrt/openwrt #9719: openwrt 21.02.2 - factory | dev/root changing from rw to ro

pclinuxer, 2022-04-15:
Hello:

Installed openwrt 21.02.2 to a WD-My Book Live 1Tb drive.
Uneventful, no major issues.

'dmsg' says:

[    0.907630] EXT4-fs (sda2): warning: mounting unchecked fs, running e2fsck is recommended
[    5.802598] EXT4-fs (sda1): warning: mounting unchecked fs, running e2fsck is recommended

/dev/sda2 is mounted and cannot be unmounted.
/dev/sda1 is mounted but can be unmounted.

root@OpenWrt:~# umount /dev/sda1
root@OpenWrt:~# 

root@OpenWrt:~# e2fsck /dev/sda1
e2fsck 1.45.6 (20-Mar-2020)
/dev/sda1 was not cleanly unmounted, check forced.
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/sda1: 15/16 files (0.0% non-contiguous), 4100/8192 blocks
root@OpenWrt:~#

root@OpenWrt:~# e2fsck /dev/sda1
e2fsck 1.45.6 (20-Mar-2020)
/dev/sda1: clean, 15/16 files, 4100/8192 blocks
root@OpenWrt:~# 

Cannot be mounted again as it is not in fstab:

root@OpenWrt:~# mount /dev/sda1
mount: can't find /dev/sda1 in /etc/fstab
root@OpenWrt:~# 

A reboot later:

root@OpenWrt:~# dmesg | grep -i e2fsck
[    0.876095] EXT4-fs (sda2): warning: mounting unchecked fs, running e2fsck is recommended
root@OpenWrt:~# 

root@OpenWrt:~# blkid
--- snip ---
/dev/sda2: LABEL="rootfs" UUID="ff313567-e9f1-5a5d-9895-3ba130b4a864" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="228c756a-02"
--- snip ---
root@OpenWrt:~# 

Hmm ....
Hold on a minute.
I have to change browser.

pepe2k, 2022-04-22:
@pclinuxer if I'm correct here, you are using extroot on the HDD and trying to unmount the root filesystem when it's running from the same disk? How could that ever work?

pclinuxer, 2022-04-22:
Hello:

> installed OpenWrt directly on the HDD ...
Yes.
> ... trying to unmount the root filesystem
No, I did not try.
I set Run filesystem check under Advanced Settings to enabled for that partition.
But as result, the filesystem ended up as RO.
Or so it seems to me.

In any case, it was solved by taking the drive out and reinstalling OpenWRT.
Then I followed the directions in the post below so that every time the system boots, it will check and repair the disk.

https://github.com/openwrt/openwrt/issues/9719

Thanks for your input.

Best,

PCL

closed 2022-04-22 by pclinuxer

pepe2k, 2022-04-22:
@pclinuxer 

> > ... trying to unmount the root filesystem
> No, I did not try.

Yes, you did.

> I was making the point that it could not be unmounted.

But you could re-mount it as `ro`.

> What I did was set `Run filesystem check` under `Advanced Settings` to _enabled_ for that partition.

Assuming your filesystem wasn't already corrupted at the moment when you were changing this setting, this should work.
You can always check content of the `/etc/config/fstab` for information about configured mount points and other related options (the filesystem check is also configured here, see: https://openwrt.org/docs/guide-user/storage/fstab).

> Maybe the filesystem was already corrupted at that point, hence the `ro`?.

Yes, that could be the reason, check your logs.

> In any case, this was solved by taking the drive out and reinstalling OpenWRT.

Or you could simply check and fix filesystem on that disk under desktop PC running any Linux distro.