openwrt/openwrt #9726: dnsmasq cannot start ,help

lengfwang, 2022-04-18:
daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d/: No such file or directory
daemon.crit dnsmasq[1]: FAILED to start up

root@OpenWrt:/tmp/dnsmasq.d# dnsmasq --version
Dnsmasq version 2.86  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.

root@OpenWrt:/tmp/dnsmasq.d# cat /etc/os-release 
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r19268-c3029f3652"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r19268-c3029f3652"

oldium, 2022-04-19:
Dnsmasq is running in a limited environment (`procd-ujail`), you cannot access any directory freely. The jail environment is controlled in `/etc/init.d/dnsmasq`, so you can either extend the script to accept custom directories to be part of `procd_add_jail_mount`, or remove the `procd-ujail` package, so that the secure/limited environment is not used at all.

imebeh, 2022-07-29:
remove `conf-dir=/etc/dnsmasq.d` from `/etc/dnsmasq.conf`
add `option confdir '/etc/dnsmasq.d'` in `/etc/config/dhcp` under `config dnsmasq`

tested works

clion007, 2024-05-24:
> remove `conf-dir=/etc/dnsmasq.d` from `/etc/dnsmasq.conf` add `option confdir '/etc/dnsmasq.d'` in `/etc/config/dhcp` under `config dnsmasq`
> 
> tested works

I think it's a bad idea, it will afect config in /tmp/dnsmasq.d path.

vorwd, 2024-06-14:
> Dnsmasq is running in a limited environment (`procd-ujail`), you cannot access any directory freely. The jail environment is controlled in `/etc/init.d/dnsmasq`, so you can either extend the script [...]

Thanks for this -- I added my pxe directory to the procd_add_jail_mount line and allowed for serving multiple boot images.  Unrelated to this directly, but ended up here searching for the failure to start dnsmasq due to directory issues for pxe root.