openwrt/openwrt #9727: libgcc_s.so.1: error adding symbols: DSO missing from command line

predators46, 2022-04-18:
Hy @neheb 

Can you help me. i made xmrig miner package and got error

undefined reference to symbol '_Unwind_Resume@@GCC_3.0'

libgcc_s.so.1: error adding symbols: DSO missing from command line

ynezz, 2022-04-18:
This is bug tracker, for support requests and general questions please use [forums](https://forum.openwrt.org/c/devel/8) or IRC instead.

closed 2022-04-18 by ynezz

neheb, 2022-04-18:
@predators46 static linking., probably need TARGET_LDFLAGS += $(if $(CONFIG_USE_GLIBC),-lc -lgcc_eh)

predators46, 2022-04-18:
> 

Hy 

> @predators46 static linking., probably need TARGET_LDFLAGS += $(if $(CONFIG_USE_GLIBC),-lc -lgcc_eh)

Hy @neheb  @ynezz 

Thank you. do i just copy and add this in the makefile? i use cmake

neheb, 2022-04-18:
yes.

predators46, 2022-04-18:
I've tried it but it doesn't work

[Makefile.txt](https://github.com/openwrt/openwrt/files/8504057/Makefile.txt)

predators46, 2022-04-18:
@neheb 

i wear musl