openwrt/openwrt #9728: Wireguard VPN routes are set unreliable

telmich, 2022-04-18:
On OpenWrt '21.02.1' on a Zbtlink ZBT-WG3526 with QMI LTE-Modem and an Wireguard tunnel, I experience the following behaviour on reboot:

* the default route is set through the LTE modem

The expected behaviour is:

* the default route is set through the VPN tunnel

After reboot, if I restart the wireguard tunnel it changes correctly to inside the tunnel.

My assumption is:

* Router boots, LTE is still down
* IPv6 default route is received via WAN port
* VPN is established via IPv6 - probably route is set correctly (I cannot observe this, if so it must be during boot)
* LTE is upped and replaces (?) the VPN default route

After rebooting the device, I can confirm that this hypothesis is correct: Below displayed is the route just after reboot and then a few seconds later:

```
[8:53] bridge:~% ssh root@vigir-office.place10.ungleich.ch ip route                           
default dev ungleichipv4 scope link 
192.168.1.0/24 dev br-lan scope link  src 192.168.1.1 
[8:53] bridge:~% ssh root@vigir-office.place10.ungleich.ch ip route
default via 10.73.176.34 dev wwan0  src 10.73.176.33 
10.73.176.32/30 dev wwan0 scope link  src 10.73.176.33 
192.168.1.0/24 dev br-lan scope link  src 192.168.1.1 
[8:54] bridge:~% 
```

The obvious question is:

* How can I set the priority for the VPN so that it is always chosen?

The design of this router is:

* IPv6 traffic is sent through the WAN interface
* IPv4 traffic is sent through the LTE modem
* Incoming (public IPv4 address) traffic is coming in via the VPN, which works over either IPv4 or IPv6 underlying

```
root@vigir-office:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.1'
DISTRIB_REVISION='r16325-88151b8303'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.1 r16325-88151b8303'
DISTRIB_TAINTS=''

root@vigir-office:~# cat /proc/cpuinfo                                                                                                                            [39/1972]
system type             : MediaTek MT7621 ver:1 eco:3
machine                 : Zbtlink ZBT-WG3526 (16M)
processor               : 0
cpu model               : MIPS 1004Kc V2.15
BogoMIPS                : 583.68
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp mt
Options implemented     : tlb 4kex 4k_cache prefetch mcheck ejtag llsc pindexed_dcache userlocal vint perf_cntr_intr_bit cdmm perf
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VPE                     : 0
VCED exceptions         : not available
VCEI exceptions         : not available

....

[vigir-log-post-reboot-post-wireguard-restart.txt](https://github.com/openwrt/openwrt/files/8504101/vigir-log-post-reboot-post-wireguard-restart.txt)
[vigir-log-post-reboot-post-route-change.txt](https://github.com/openwrt/openwrt/files/8504102/vigir-log-post-reboot-post-route-change.txt)
[vigir-network.txt](https://github.com/openwrt/openwrt/files/8504103/vigir-network.txt)
[vigir-firewall.txt](https://github.com/openwrt/openwrt/files/8504104/vigir-firewall.txt)

telmich, 2022-04-18:
[vigir-log-post-reboot-post-wireguard-restart.txt](https://github.com/openwrt/openwrt/files/8504115/vigir-log-post-reboot-post-wireguard-restart.txt)
[vigir-log-post-reboot-post-route-change.txt](https://github.com/openwrt/openwrt/files/8504116/vigir-log-post-reboot-post-route-change.txt)
[vigir-network.txt](https://github.com/openwrt/openwrt/files/8504117/vigir-network.txt)
[vigir-firewall.txt](https://github.com/openwrt/openwrt/files/8504118/vigir-firewall.txt)

Re added files with correct links.

pepe2k, 2022-04-22:
@telmich please, use our forum (https://forum.openwrt.org) or Wiki (https://openwrt.org) for general support/functionality question/research.

closed 2022-04-22 by pepe2k