openwrt/openwrt #9729: ramips/mt7621 nand/ubi/squashfs corruption

RussellSenior, 2022-04-18:
Observed on a batch of 7 Ubiquiti ER-X's, flashed between March and October 2021 built from from master branch, months after flashing, began seeing squashfs errors. For example, in one case device was flashed in late October (r17804-gdd0ad9b661) and in February, began seeing this error:
```
Feb  3 11:19:04 10.11.250.56 kernel: [ 5286.646845] SQUASHFS error: xz decompression failed, data probably corrupt
Feb  3 11:19:04 10.11.250.56 kernel: [ 5286.660653] SQUASHFS error: Failed to read block 0x5c69f2: -5
Feb  3 11:19:04 10.11.250.56 kernel: [ 5286.672244] SQUASHFS error: Unable to read fragment cache entry [5c69f2]
Feb  3 11:19:04 10.11.250.56 kernel: [ 5286.685716] SQUASHFS error: Unable to read page, block 5c69f2, size 22a30
Feb  3 11:19:04 10.11.250.56 kernel: [ 5286.699561] SQUASHFS error: Unable to read fragment cache entry [5c69f2]
Feb  3 11:19:04 10.11.250.56 kernel: [ 5286.713163] SQUASHFS error: Unable to read page, block 5c69f2, size 22a30
[...]
```
The error was corrected by reflashing the same image. The squashfs is written on a UBI volume, on a NAND partition. There are no apparently bad blocks on the flash, or none that appear relevant to this error.  The error only became evident where some processes happened to try to read the squashfs file after the corruption occurred. If the file is only read at boot time, then the failure might be latent until a reboot, at which time there is a fair chance (as happened in several cases) the device fails to boot fully and/or boot loops.

After seeing several cases of the "failure to boot" problem, I searched the set of 7 devices deployed and looked for squashfs errors. A traversal of all squashfs files, e.g. ```find /rom -type f | xargs sha256sum > /dev/null && dmesg | tail``` should expose any errors if they have appeared. All 7 of the devices, months or close to a year since flashing, in constant use, showed at least one such squashfs corruption.

The source of the corruption is not known, nor whether it might have been fixed since r17804-gdd0ad9b661, but people with deployed squashfs on ubi volumes, particularly mt7621, are encouraged to check for corruption and report here for more data points.

ivanich, 2022-04-22:
It seems that have the same issue, after power loss there is 50/50 chance that my Ubiquiti ER-X fails to boot, unfortunately I can't check logs and confirm, but symptoms are very similar.

ynezz, 2022-05-23:
Duplicate of #9085

closed 2022-05-23 by ynezz