openwrt/openwrt #9744: The way to fix mtk pppoe hardware offload

MeIsReallyBa, 2022-04-20:
There are two problems that broke the pppoe hw offload.

1.PPPOE can't pass the dev->type != ARPHRD_ETHER || dev->addr_len != ETH_ALEN check. It will return here. I used to mention it here.  nbd's patch solves part of the problem. https://github.com/openwrt/openwrt/issues/9531#issuecomment-1078963088

2.There is no valid resoure address with pppoe. we can see that both resource and dest address are created in linux 5.4 version but now there is only dest address.

https://github.com/openwrt/openwrt/blob/1418439da9ed2dbd4b097298c51ee70629fe10d3/target/linux/generic/pending-5.4/644-net-pppoe-support-hardware-flow-table-offload.patch#L94

https://github.com/openwrt/openwrt/blob/92add80414c2f39ba8fd0d221d0f37e75fb19951/target/linux/generic/backport-5.10/610-v5.13-15-net-ppp-resolve-forwarding-path-for-bridge-pppoe-dev.patch#L71

I mixed nod's patch and my patch to create this one and solved pppoe hw offload. The cpu usage is zero during speedtest. 

https://github.com/MeIsReallyBa/openwrt/blob/master/target/linux/mediatek/patches-5.10/999-fix-pppoe-flowoffload.patch

MeIsReallyBa, 2022-04-20:
I hope that someone could provide a better patch based on my one and make a pr.  I'm not familiar with the pr process.

MeIsReallyBa, 2022-04-29:
> so I just applied 999-fix-pppoe-flowoffload.patch, but can't see that cpu is not used (at least using htop I see the red bars), aslo it seems that the device resets itself after a few hours, erx device, but also the device lockup was introduced with the latest flow offload changes (I think ipv6 related)

This patch only supports firewall3.

oldium, 2022-05-06:
There are some changes to fix offloading specifically for PPPoE “kernel: fix flow offload issues with pppoe” - on master (kernels 5.15+5.10) and on 22.03 branch (kernel 5.10). Could it be that this fixes your issue?