openwrt/openwrt #9747: rpcd hangs on long data return

jjm2473, 2022-04-20:
POC:

1. Save this code to `/usr/libexec/rpcd/crash`:
```bash
#!/bin/bash

# rpcd hang POC
# /etc/init.d/rpcd reload && ubus call crash big

. /usr/share/libubox/jshn.sh

sleep0() {
    sleep 31
    echo "{}"
    return 0
}

big() {
    local i
    echo "{ \"result\":[ "
    for (( i=1; i<=3000; i++ ))
    do
        cat <<-EOF
                {
                        "layer3": "ipv4",
                        "timeout": 7413,
                        "zone": "0",
                        "dst": "129.211.129.109",
                        "layer4": "tcp",
                        "src": "192.168.31.104",
                        "sport": "58488",
                        "mark": "0",
                        "packets": 0,
                        "dport": "80",
                        "bytes": 0,
                        "use": "2"
                },
                {
                        "layer3": "ipv4",
                        "timeout": 7413,
                        "zone": "0",
                        "dst": "129.211.129.109",
                        "layer4": "tcp",
                        "src": "192.168.31.104",
                        "sport": "58488",
                        "mark": "0",
                        "packets": 0,
                        "dport": "80",
                        "bytes": 0,
                        "use": "2"
                },
EOF
    done
    echo "{} ] }"
    return 0
}

main () {
    case "$1" in
        list)
            json_init
            json_add_object "sleep"
            json_close_object
            json_add_object "big"
            json_close_object
            json_dump
            ;;
        call)
            case "$2" in
                sleep)
                    sleep0
                    ;;
                big)
                    big
                    ;;
            esac
            ;;
    esac
}

main "$@"

```

2. Call `/etc/init.d/rpcd reload ;  ubus list | grep crash`, make sure `crash` was registered
3. Call `ubus call crash big >/dev/null`, will hangs for a long time (about 30s to 60s)
4. Call `ubus list`, show that all ubus objects registered by `rpcd` disappeared

ptpt52, 2022-04-20:
I also notice this issue!!

jow-, 2022-04-20:
The root cause is ubusd kicking rpcd's connection when a too long message is sent to ubusd. The rpcd service will keep running but it won't be connected to ubus anymore. I'll amend rpcd later to reconnect on dropped ubus connections but that will not fix large conntrack lists. For that a change in the message format will be needed to send the reply chunked into multiple parts.

jjm2473, 2022-04-21:
@jow-  can we just make this call fail, don't unregister other ubus object? If `uci` unregistered, we can not login luci anymore.

jow-, 2022-04-21:
Not easily, since libubus will silently drop the connection. Error handling and reconnect logic has to be added to rpcd

ynezz, 2022-06-09:
Sorry that I couldn't currently help more, but I wanted to try [following patch](https://gitlab.com/prpl-foundation/prplos/prplos/-/commit/7150f631ae2ed938138705aa1f456c4cdf45201f.patch) on this issue as well, since it fixes likely similar issue. Please note, that it shouldn't be probably applied as it is, since Coverity reports following UAF:

[image]

IIRC I've tried to check that with clang's static code analyzer already and it didn't complained, so might be as well false positive, similar to the [current one](https://openwrt.gitlab.io/-/project/ubus/-/jobs/2563471391/artifacts/build/scan/2022-06-08-150320-72-1/report-9335a4.html#EndPath) related to the linked list handling as well.

jjm2473, 2022-06-10:
> @jow- @jjm2473 any progress about this issue? i seems still issue on lastest code

I have limit conntrack records returned to 1000, workaround this issue. 😆

ProKn1fe, 2023-01-06:
I found why getConntrackList crash when you have >5k connections in luci, it's because UBUS_MAX_MSGLEN have only 1MB max length. I increase it to 4mb and now it works with 10-20k connections.
https://git.openwrt.org/?p=project/ubus.git;a=blob_plain;f=CMakeLists.txt;hb=refs/heads/master
@jow- , @ynezz can we get this parameter into config files?

chummumm, 2023-04-26:
> I found why getConntrackList crash when you have >5k connections in luci, it's because UBUS_MAX_MSGLEN have only 1MB max length. I increase it to 4mb and now it works with 10-20k connections. [...]

How can I modify this at compile time?

ProKn1fe, 2023-04-26:
I'm use method of creating patch.

`openwrt\package\system\ubus\patches\0001-fix-crash.patch`

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index a65ccbb..a17e924 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -14,7 +14,7 @@ OPTION(BUILD_EXAMPLES "build examples" ON)
 
 SET(CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "")
 SET(UBUS_UNIX_SOCKET "/var/run/ubus/ubus.sock")
-SET(UBUS_MAX_MSGLEN 1048576)
+SET(UBUS_MAX_MSGLEN 8388608)
 
 ADD_DEFINITIONS( -DUBUS_UNIX_SOCKET="${UBUS_UNIX_SOCKET}")
 ADD_DEFINITIONS( -DUBUS_MAX_MSGLEN=${UBUS_MAX_MSGLEN})

```

chummumm, 2023-04-27:
> I'm use method of creating patch.
>
> `openwrt\package\system\ubus\patches\0001-fix-crash.patch`
>
> ```diff
> diff --git a/CMakeLists.txt b/CMakeLists.txt
> index a65ccbb..a17e924 100644
> --- a/CMakeLists.txt [...]

ok, thank you very much

jjm2473, 2025-05-28:
In 24.10.1, `ubus call crash big >/dev/null` will still hang, but the objects registered by `rpcd` will not disappear from ubus, but it will still cause the user to lose the login status