openwrt/openwrt #9750: ZTE MF286D modem randomly crashes

WereCatf, 2022-04-21:
Every now and then, the modem on my ZTE MF286D just randomly stops working. ModemManager thinks it's still connected, but no traffic actually moves through nor does the modem respond to e.g. attempts at resetting it; only a full reboot of the entire router works.

I have not found a way of reliably triggering this: sometimes it happens within minutes, sometimes hours.

Logs do not  really say much of anything useful. When this issue occurs, the logs spit out messages similar as to the following,  but nothing else is out of the ordinary:
```
Thu Apr 21 09:33:46 2022 daemon.debug [2126]: transaction 0xa aborted, but message is not abortable
Thu Apr 21 09:33:46 2022 daemon.warn [2126]: <warn>  [modem0/bearer1] reloading stats failed: QMI operation failed: Transaction timed out

```
There are no other messages related to the modem when the issue happens. I do not know what other information would be useful here, so do tell, if you need something more.

Tested on OpenWRT-trunk from commit 0d2d52df69e91fe2f6e3e65750c0728fc45f5ab2, compiled on 21.04.2022.

mrkiko, 2022-04-21:
Seems a modem firmware related issue, so not sure how we could proceed.
What do you think @aleksander0m ?

aleksander0m, 2022-04-21:
Can't say anything without looking at the logs

WereCatf, 2022-04-21:
> Can't say anything without looking at the logs

Well, maybe you'll see something useful there, but I certainly don't. The log: https://www.dropbox.com/s/v2w4xjbq5w0a4rd/log.txt?dl=0

aleksander0m, 2022-04-21:
Yeah, not much info with the default verbosity.

There's a call incoming at the same time of your connection, was that by chance?
```
[modem0/call0] call state changed: unknown -> ringing-in (incoming-new)
```

I wonder if that somehow messed up the modem internal stack.

WereCatf, 2022-04-21:
> There's a call incoming at the same time of your connection, was that by chance?
I noticed that there, but I didn't figure it'd have anything to do with the issue. I did not initiate it and I have no idea why there would be a call incoming to it.

The SIM-card is one of my husband's, so I guess I could ask him what the number is and try calling it, then seeing if the modem goes wonky again, or did you have some other ideas?

aleksander0m, 2022-04-21:
That's something to try I guess. Otherwise, there's not much we can do really, looks like the firmware gets stuck.

WereCatf, 2022-04-21:
Calling the number doesn't seem to be the cause. At least I can't get it to get stuck by calling the SIM-card's number, which means that I need to flash back to the stock firmware, because the router isn't very useful like this.

Leo-PL, 2022-04-23:
I recommend upgrading to the latest stock firmware before flashing OpenWrt, so modem gets upgraded with it as well, then proceeding with OpenWrt.
For me it just works, but I'm using the default uqmi setup with autoconnect enabled.

WereCatf, 2022-04-23:
As far as I know, it was on the latest available firmware. mmcli says the modem's current firmware is BD_ELISAFIMF286DMODV1.0.1B05 -- what's yours?

Leo-PL, 2022-04-23:
Mine came from Nordic, firmware is BD_TELIAMF286DV1.0.0B10, though a week or so ago, .B11 was released, haven't flashed back yet to attempt upgrade. It may be possible to update firmware on the modem only though, if the delta update for it gets extracted from main package, and pushed over ADB to the right place in the modem's filesystem.

WereCatf, 2022-04-24:
I don't think the Elisa firmware-version corresponds with the modem firmware. I just flashed the original firmware back to my MF286D and the router's web-UI claims a firmware-version of B11, whereas mmcli on OpenWRT claims the odd 1.0.1B05 -- googling around, I didn't come across any mention of the version starting with "1.0.1", only "1.0.0"

The router also claims there is no newer firmware available. Looking around the filesystem, logs and everything, I'm only seeing mentions of the year 2019, so I don't know what to think.

I'm pretty much out of my depth here, but it'd be nice to figure out how get things to work nice and stable with modern OpenWRT.

Leo-PL, 2022-04-24:
Other than that, I've seen folks at the mentioned polish forum cross-flash the modems with "ZTE Terminal Software Update Framework". OTOH, have you tried using just uqmi/luci-proto-qmi? Because it seems that ModemManager seems to be irritated somehow by voice support of this modem. Maybe to avoid the problem as well, you might consider switching it to register only in PS domain, as opposed to default CS+PS? This is done by `mmcli -m <modem> --3gpp-set-eps-ue-mode-operation=ps-1`
OTOH I think that this will prevent it from receiving SMS messages if you need them.

For me, uqmi works even with a hack added to support IPv4+IPv6 dual stack on separate APNs. Unlike on other modems I was able to set "autoconnect" option to 1 in that scenario and the modem keeps the session nicely.

WereCatf, 2022-04-24:
Well, I could try both, but I suppose I'll have to try and come up with a reliable way of triggering the crash, so I can actually test if either (or both) of those actually help. It just seems so rare and random that I'm not sure it's even possible to do it reliably.

Wouldn't hurt to mention the possibility of the modem crashing and what one might try to work around it on the wiki.

CHKDSK88, 2022-04-26:
https://gist.github.com/CHKDSK88/949df9b981d977000b6d413f6b6f3bc3 -> here's decompiled dts from OFW. Maybe it could help.

Leo-PL, 2022-04-26:
Especially this part: 
```
				usb2-susphy-quirk;
				usb2-host-discon-quirk;
				usb2-host-discon-phy-misc-reg = <0x24>;
				usb2-host-discon-mask = <0x100>;
```
in both `snps,dwc3` nodes.

Leo-PL, 2024-03-02:
@WereCatf I looked on some other instability issues with MF286D that have popped up on the internet, and it seems the PSU might be at fault, too - with some users even observing uncontrolled restarts. Is it still happening? Maybe you could try with a different, but known good one. During the transmission, the modem seems to create substantial spikes in current draw.