openwrt/openwrt #9762: opkg --force-downgrade not working

jjm2473, 2022-04-23:
Openwrt: 21.02.2

1. I have install a `libmbedtls12` from other source, version `2.16.12-2`.
2. Run `opkg update; opkg --force-downgrade install libmbedtls12` output that:
    ```
    Installing libmbedtls12 (2.16.12-2) to root...
    Collected errors:
     * opkg_download_pkg: Package libmbedtls12 is not available from any configured src.
     * opkg_install_pkg: Failed to download libmbedtls12. Perhaps you need to run 'opkg update'?
     * opkg_install_cmd: Cannot install package libmbedtls12.
    ```
The version of `libmbedtls12` in official repo is `2.16.12-1`, opkg can not downgrade to it.

pepe2k, 2022-05-04:
@jjm2473 
> I have installed a libmbedtls12 from other source [...]

What source? What platform?

jjm2473, 2022-05-05:
@pepe2k  All source based on official release 21.02.2, including opkg. Platform is aarch64-cortex-a53.
I think it is not related on platform. I have tested on x86_64 21.02.1, newer libmbedtls12 (`2.16.12-2`) was installed on system, then run `opkg update; opkg --force-downgrade --force-reinstall install libmbedtls12`, show that:
```
Removing package libmbedtls12 from root...
Installing libmbedtls12 (2.16.12-2) to root...
Collected errors:
 * opkg_download_pkg: Package libmbedtls12 is not available from any configured src.
 * opkg_install_pkg: Failed to download libmbedtls12. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package libmbedtls12.
```
online libmbedtls12 version is `2.16.12-1`.

pepe2k, 2022-05-05:
@jjm2473 could you please share your `2.16.12-2` version of libmbedtls12 and show output from commands `opkg -V info libmbedtls12` and `opkg -V status libmbedtls12`

jjm2473, 2022-05-05:
@pepe2k Actually, I just edit `/usr/lib/opkg/status` and `/usr/lib/opkg/info/libmbedtls12.control` to change its version.

Reproduce steps:
1. `opkg update; opkg install libmbedtls12`
2. Edit `/usr/lib/opkg/status` and `/usr/lib/opkg/info/libmbedtls12.control` to fake a newer version
3. `opkg --force-downgrade --force-reinstall install libmbedtls12`

silentred, 2025-07-16:
wired behavior too. It seems when a higher version is installed, the version info is cached in opkg pkg management.  It fails to install a lower version pkg.
```
opkg update

# to make sure a lower version exist (2.6.10)
opkg list | grep ruby 

# to remove installed higher version (3.1.2), it is successfully removed 
opkg remove ruby

opkg info ruby
# output shows info of ruby(2.6.10), there is some `status` of ruby(3.1.2) in the end. 

opkg install ruby --force-downgrade
# output shows that there is no ruby(3.1.2) to install.
```
I have to download archieve ipk files for installing.

jjm2473, 2025-07-16:
@silentred In fact, you can directly edit `/usr/lib/opkg/status`, delete the package definition, or change the version number to 0.0.0, and then you can install it normally.