openwrt/openwrt #9763: RB760iGS fails to boot after sysupgrade

cmonty14, 2022-04-23:
Hello,
after booting this router RB760iGS with initramfs via TFTP I started to install squashfs, however the router fails to boot correctly, means theres' no connection to router's defailt IP 192.168.1.1 using ethernet ports 2-5.

I have tried to flash the following files w/o success:
openwrt-21.02.3-ramips-mt7621-mikrotik_routerboard-760igs-squashfs-sysupgrade.bin
openwrt-ramips-mt7621-mikrotik_routerboard-760igs-squashfs-sysupgrade.bin

I can observe that the router's SFP LED is flashing, so I assume the router is in a kind of boot loop.

This issue is well described [here](https://forum.openwrt.org/t/mikrotik-hex-s-rb760igs-installation-via-sysupgrade-no-connection/75513) in the forum.
But there's no solution available.

What's worse in my case is that I cannot install RouterOS using netinstall as documented [here](https://help.mikrotik.com/docs/display/ROS/Netinstall).

Before I booted OpenWRT initramfs via TFTP I upgraded router's firmware and packages to version 7.2.1.

cmonty14, 2022-04-24:
Issue fixed by downgrading RouterOS FW and software packages to version 6.46.4.

closed 2022-04-24 by cmonty14

walterav1984, 2022-04-26:
Was downgrading OS and Packages enough or did you also need to downgrade bootloader as in? 

`> System > RouterBOARD > Current Firmware / Upgrade Firmware`

Have you tried current 6.48.6 Long-term or 6.49.6 Stable besides 6.46.4 or are these also affected by current 7.2.1 flash durability updates or improvent upgrade/downgrade mechanisms?

`*) filesystem - improved long-term filesystem stability and data integrity;`

cmonty14, 2022-04-27:
I downgraded everything on the router device incl. firmware.
And I didn't try any other version.

walterav1984, 2022-04-27:
Thanks for the details, might hint the device [wiki page](https://openwrt.org/toh/mikrotik/rb760igs).

RuStrannik, 2022-06-12:
Just wanted to add confirmation of ability to boot after sysupgrade from '6.48 (Stable)'.

walterav1984, 2022-06-13:
@RuStrannik as in 6.48.5,  6.48.6 (only 2 available for download) or just 6.48 also do you remember what version the bootloader was on before installing OpenWrt?

RuStrannik, 2022-06-13:
@[walterav1984](https://github.com/walterav1984), I'm afraid the patch number was not specified, the version number was mentioned exactly as quoted. I did't play with RouterOS much, just went straight to flashing OpenWRT as soon as I got the router board, however, I did dumped and saved all 6 original partitions, so if you can tell me how to extract version numbers from those, I will do that.

Mihai-B, 2022-07-21:
Is this still a problem?
I cannot downgrade to version 6.46.4. as my factory image was higher than thios and you can only downgrade to the same version as the one it came from factory.

muriloglix, 2022-12-18:
Confirming that version 6.48.6 (packages and firmware) works perfectly.

kohend, 2023-05-09:
I also managed to use 6.48.6, I had some issues but they were due to a faulty USB network adapter.

jannoke, 2025-01-08:
Is there something specific that needs to be done to downgrade. I have 7.12 installed and factory one is 6.48.6. I uploaded the 6.48.6 to flash and used downgrade command, but it would still boot as 7.12.

rohman21, 2025-04-19:
hello, does anyone have full dump firmware of rb760igs ? I really need it, I corrupted my bios and now it won't boot up or detected by netinstall. Thanks in advance

trenchcoatjedi, 2025-07-08:
Can confirm that OpenWRT will boot and flash sysupgrade correctly on bootloader 6.49.8