openwrt/openwrt #9764: firewall4: negative netmask not working

janh, 2022-04-23:
firewall3 allowed to use negative netmasks, as in this example:

```
config rule
	option target 'ACCEPT'
	option src 'wan'
	option dest 'lan'
	option family 'ipv6'
	option proto 'tcp'
	option dest_ip '::1:0123:4567:89ab:cdef/-56'
	option dest_port '443'
```

This no longer works with firewall4. `/etc/init.d/firewall reload` prints the following error:
```
/proc/self/fd/0:165:13-43: Error: Could not resolve hostname: Try again
```

The same rule works fine when the full netmask is specified explicitly (`::1:0123:4567:89ab:cdef/::ff:ffff:ffff:ffff:ffff`).

(Tested with OpenWrt 22.03.0-rc1)

ptpt52, 2022-04-23:
mark

janh, 2022-05-22:
@jow: It looks like [your recent commit](https://git.openwrt.org/?p=project/firewall4.git;a=commit;h=c22eeeff1ef0884fd3c76f4ff2c72caa9de82fd5) was intended to fix this, but it doesn't work properly.

`/etc/init.d/firewall reload` now prints a different error:
```
/proc/self/fd/0:107:61-63: Error: syntax error, unexpected tcp, expecting comma
```
The corresponding line in `fw4 print` is this:
```
ip6 daddr ::1:123:4567:89ab:cdef/::ff:ffff:ffff:ffff:ffff tcp dport 443 counter jump accept_to_lan comment "!fw4: @rule[10]"
```

When the full netmask is specified the rule is slightly different and there is no error:
```
ip6 daddr & ::ff:ffff:ffff:ffff:ffff == ::1:123:4567:89ab:cdef tcp dport 443 counter jump accept_to_lan comment "!fw4: @rule[10]"
```

notgood, 2022-05-24:
As this issue breaks existing configs, can it be labeled for openwrt-22.03 please?

jow-, 2022-05-31:
Please test the firewall bump commit at https://git.openwrt.org/49e6daa8

commit a7ddef6e (2022-05-31): firewall4: update to latest Git HEAD

210991d fw4: prefer /dev/stdin if available
4e5e322 fw4: make `fw4 restart` behavior more robust
221040e ruleset: emit time ranges when both start and stop times are specified
30a7d47 fw4: fix datetime parsing
fb9a6b2 ruleset: correct mangle_output chain type
6dd2617 fw4: fix logic flaw in testing hw flow offloading support
c7c9c84 fw4: ensure that negative bitcounts are properly translated
c4a78ed fw4: fix typo in emitted set types

Fixes: #9764, #9923, #9927, #9935, #9955
commit dbd4c345 (2022-06-01): firewall4: update to latest Git HEAD

210991d fw4: prefer /dev/stdin if available
4e5e322 fw4: make `fw4 restart` behavior more robust
221040e ruleset: emit time ranges when both start and stop times are specified
30a7d47 fw4: fix datetime parsing
fb9a6b2 ruleset: correct mangle_output chain type
6dd2617 fw4: fix logic flaw in testing hw flow offloading support
c7c9c84 fw4: ensure that negative bitcounts are properly translated
c4a78ed fw4: fix typo in emitted set types

Fixes: #9764, #9923, #9927, #9935, #9955

(cherry picked from commit a7ddef6ef122a935d6bdf38d7ead686b12738007)
closed 2022-05-31 by jow-