openwrt/openwrt #9765: firewall4: ICMP rule without specified family is only applied for IPv4

janh, 2022-04-23:
With firewall3, the following rule applied to both IPv4 and IPv6:

```
config rule
	option target 'ACCEPT'
	option src 'guest'
	option proto 'icmp'
```

With firewall4, it only applies to IPv4 (this is also the case when `option family 'any'` is explicitly specified).

To get the same result as before, two separate rules for IPv4 and IPv6 are now required:

```
config rule
	option target 'ACCEPT'
	option src 'guest'
	option proto 'icmp'
	option family 'ipv4'

config rule
	option target 'ACCEPT'
	option src 'guest'
	option proto 'icmp'
	option family 'ipv6'
```

(Tested with OpenWrt 22.03.0-rc1)

ptpt52, 2022-04-23:
mark

wongsyrone, 2022-04-26:
Analysis:

It infers the family based on
```
ipset, "set match",
[2398](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2398)                                 sip, "source IP",
[2399](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2399)                                 dip, "destination IP",
[2400](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2400)                                 rule.src?.zone, "source zone",
[2401](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2401)                                 rule.dest?.zone, "destination zone",
[2402](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2402)                                 rule.helper, "helper match",
[2403](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2403)                                 rule.set_helper, "helper to set"
```

Your rule has none of them, it will get family 0, a.k.a AF agnostic rule

```
[2424](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2424)                         /* check if there's no AF specific bits, in this case we can do an AF agnostic rule */
[2425](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2425)                         if (!family && rule.target != "dscp" && !has_ipv4_specifics && !has_ipv6_specifics) {
[2426](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2426)                                 add_rule(0, proto, [], [], sports, dports, null, null, null, rule);
[2427](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l2427)                         }
```
Eventually, your rule is pushed onto `this.state.rules`.

Finally, the rule is rendered to nft rules via `/usr/share/firewall4/templates/rule.uc`, the family attribute is handled by `fw4.nfproto`. 

```
1 {%+ if (rule.family && !rule.has_addrs): -%}
[   2](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/firewall4/templates/rule.uc;h=417bef7d2d995fd9ac124f2d29525448bbfcd015;hb=HEAD#l2)         meta nfproto {{ fw4.nfproto(rule.family) }} {%+ endif -%}
```

The rendered rule doesn't specify `inet` as nfproto based on code of `fw4.nfproto`

```
[1606](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1606)         nfproto: function(family, human_readable) {
[1607](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1607)                 switch (family) {
[1608](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1608)                 case 4:
[1609](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1609)                         return human_readable ? "IPv4" : "ipv4";
[1610](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1610) 
[1611](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1611)                 case 6:
[1612](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1612)                         return human_readable ? "IPv6" : "ipv6";
[1613](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1613) 
[1614](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1614)                 default:
[1615](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1615)                         return human_readable ? "IPv4/IPv6" : null;
[1616](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1616)                 }
[1617](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=HEAD#l1617)         },
```

You will get `null` instead of `inet` for IPv4/IPv6 since your family is 0, which goes the default branch.

That's the reason why you have to create two rule entries to bypass the family parser, and ultimately push two rules to `this.state.rules`, and the generated rule specify `nfproto ipv4` and `nfproto ipv6`.

wongsyrone, 2022-04-26:
EDIT:

There's something wrong with my analysis.

Just see you specify `src` in your rule. it will infer family based on `rule.src?.zone`( the source zone). I have no knowledge of your zone and cannot analyze further.

To check the rule generated by fw4, you can execute `fw4 print` to see what went wrong.

janh, 2022-04-26:
The issue can also be reproduced with `src 'lan'` (of course the rule doesn't make any practical sense then, as the zone has `option input 'ACCEPT'` by default, but it demonstrates the issue). So the full rule is then:
```
config rule
	option target 'ACCEPT'
	option src 'lan'
	option proto 'icmp'
```

On firewall3:
```
# fw3 -4 print
iptables -t filter -A zone_lan_input -p icmp -m comment --comment "!fw3: @rule[9]" -j ACCEPT
# fw3 -6 print
ip6tables -t filter -A zone_lan_input -p icmpv6 -m comment --comment "!fw3: @rule[9]" -j ACCEPT
```

On firewall4:
```
# fw4 print
	chain input_lan {
		meta l4proto icmp counter accept comment "!fw4: @rule[9]"
		jump accept_from_lan
	}
```

On firewall4 with two family-specific rules:
```
# fw4 print
	chain input_lan {
		meta nfproto ipv4 meta l4proto icmp counter accept comment "!fw4: @rule[9]"
		meta nfproto ipv6 meta l4proto ipv6-icmp counter accept comment "!fw4: @rule[10]"
		jump accept_from_lan
	}
```

janh, 2022-04-26:
After taking a closer look, it looks like the problem is that the rule ends up being handled in [this section](https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=b725459c77d22d0a4f8ba924661a07c2239ca124;hb=refs/heads/master#l2424):
```
/* check if there's no AF specific bits, in this case we can do an AF agnostic rule */
if (!family && rule.target != "dscp" && !has_ipv4_specifics && !has_ipv6_specifics) {
        add_rule(0, proto, [], [], sports, dports, null, null, null, rule);
}
```

Adding `proto.name != "icmp"` to this check seems to fix the issue.

jow-, 2022-04-26:
https://git.openwrt.org/?p=project/firewall4.git;a=commitdiff;h=370f9b5188fb013a10de1703b524e414ba636e66;hp=2379c3d117ea6f8651957652672e758d7aa240ec

closed 2022-05-04 by pepe2k

jow-, 2022-05-20:
Fixed with 2df17604a4f891447beb66988e3d83e23ab3a3b0