openwrt/openwrt #9766: HiLink HLK7621 random mac address

proffust, 2022-04-23:
Hello!
after install snapshot release on https://www.hlktech.com/en/Goods-51.html all working, but after reboot mac address generate randomly. with stork firmware on every plate different macs, which get from factory md patrition

root@OpenWrt:/# . /lib/functions/system.sh
root@OpenWrt:/# mtd_get_mac_binary factory 0xe000
40:d6:3c:fb:02:19 - LAN mac
root@OpenWrt:/# mtd_get_mac_binary factory 0xe006
40:d6:3c:fb:02:1a - WAN mac

namidairo, 2022-04-24:
```
Since it's an evaluation board and there is no solid information
about the MAC address assignment, the ethernet MAC address is left random.
```

Is what the adding commit says, which apparently isn't true in your case.

You should be able to add it to the DTS if you are comfortable.

Something like...

Changing this:
```
		port@4 {
			status = "okay";
			label = "wan";
		};
```

To this.

```
		port@4 {
			status = "okay";
			label = "wan";
			nvmem-cells = <&macaddr_factory_e006>;
			nvmem-cell-names = "mac-address";
		};
```

And adding the below as well...

```
&factory {
	compatible = "nvmem-cells";
	#address-cells = <1>;
	#size-cells = <1>;

	macaddr_factory_e000: macaddr@e000 {
		reg = <0xe000 0x6>;
	};

	macaddr_factory_e006: macaddr@e006 {
		reg = <0xe006 0x6>;
	};
};

&gmac0 {
	nvmem-cells = <&macaddr_factory_e000>;
	nvmem-cell-names = "mac-address";
};
```

rma-x, 2022-05-04:
I already submitted a patch for this, but it didn't get any attention so far on the mailing list or in Patchwork:
http://lists.openwrt.org/pipermail/openwrt-devel/2022-April/038456.html

I only applied the MAC address found at 0xe000 to the CPU port of the switch, but didn't designate any port as WAN, because this device is so generic and has no WAN marking on the board. Instead I sent another patch that renames the switch ports to p0..p4 as they are named in the data sheet which also matches the board markings for the respective LEDs:
http://lists.openwrt.org/pipermail/openwrt-devel/2022-April/038457.html

NilsRo, 2022-09-15:
> I already submitted a patch for this, but it didn't get any attention so far on the mailing list or in Patchwork:
>
> http://lists.openwrt.org/pipermail/openwrt-devel/2022-April/038456.html
>
>
>
> I only [...]

Same for Netgear R7800. Is this a fix for all devices? Does not looks like this...

rma-x, 2022-09-15:
> Same for Netgear R7800. Is this a fix for all devices?

My fix is specific to the HLK-7621 board.

There is no generic way where and how the MAC addresses are stored in a router. Even boards that use the same SoC might not do it the same way. And given that we look at at two totally different devices here (different SoC vendors, different board vendors), I wouldn't expect any similarities, even though the symptoms might be the same.

LXGMAX, 2023-01-11:
I fix it by change dts
```c
&gmac0 {
  mtd-mac-address = <&factory 0xe000>;
};
```
and 
```c
&switch0 {
  ports {
    port@0 {
      status = "okay";
      label = "lan1";
    };

    port@1 {
      status = "okay";
      label = "lan2";
    };

    port@2 {
      status = "okay";
       label = "lan3";
    };

    port@3 {
      status = "okay";
      label = "lan4";
    };

    port@4 {
      status = "okay";
      label = "wan";
      mtd-mac-address = <&factory 0xe000>;
      mtd-mac-address-increment = <1>;
    };
  };
};
```

rma-x, 2023-01-11:
> ```c
>       mtd-mac-address = <&factory 0xe000>;
>       mtd-mac-address-increment = <1>;
> ```

AFAIK this increments the MAC address found at 0xe000 by 1 instead of using the MAC address found at 0xe006 as shown in the initial comment of this issue. In most cases this might not matter, because by default the two seem to be the same, but we should use the two addresses as the manufacturer intended them, because on a customized board they might contain entirely different MAC addresses.

LXGMAX, 2023-01-12:
> AFAIK this increments the MAC address found at 0xe000 by 1 instead of using the MAC address found at 0xe006 as shown in the initial comment of this issue. In most cases this might not matter, because by default the two seem to be the same (`*0xe006 == (*0xe000)+1`), but we should use the values at these two addresses as the manufacturer intended them, because on a customized board they might contain MAC addresses that don't follow this rule.

I've tested remove `mtd-mac-address-increment = <1>;` and set address to `0xe006` worked, WAN port read the correct MAC address offset. Now the WAN and LAN got factory address normally.
```c
    port@4 {
      status = "okay";
      label = "wan";
      mtd-mac-address = <&factory 0xe006>;
    };
```