openwrt/openwrt #9767: 802.11r Fast Transition, speed drops and disconnections

ondoteam, 2022-04-23:
# Expected Behavior

Stability of connection and speed.

# Current Behavior

After activate 802.11r Fast Transition under wlan1 (5 Ghz.) I got multiple random reconnections (many times it goes from 5 Ghz to 2.4 Ghz.) and a drop in speed from about 700 Mbps to 100 Mbps (until I manually disconnect, wait 15 seconds and connect).

With the workaround, I continue to have disconnections but the speed is maintainded as long as the device changes back to 5 Ghz band.

[image]

# Failure Information

802.11r protocol it is working fine in every access point with OpenWRT (WiFi 5) that I own but TOTOLINK X5009R (WiFi 6)

## Steps to Reproduce

1. Install OpenWRT (from 21.02.2 to lastest SNAPSHOT) in the TOTOLINK X5000R.
2. Configure some SSIDs.
3. Activate 802.11r Fast Transition.
4. Set a NAS ID (Ex.: TOTOLINK).
5. Set a Mobility Domain (Ex.: 2222).
6. All other related options to default values.
7. Save and apply.
8. Connect to your 5 Ghz SSID.
9. Go to fast.com and do some tests until you see a speed drop (maybe some reconnection). Make sure that your device still on 5 Ghz. band while testing. 

## Context

* Firmware Version: OpenWrt SNAPSHOT r19508-0d2d52df69 / LuCI Master git-22.089.43958-7110635
* Hardware: TOTOLINK X5000R
* Arch.: MediaTek MT7621 ver:1 eco:3
* WiFi config:

```
config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
	option band '5g'
	option country 'ES'
	option cell_density '0'
	option htmode 'HE80'
	option channel '36'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'test'
	option encryption 'psk2'
	option key 'test1!test1!'
	option ieee80211r '1'
	option nasid 'NASID'
	option mobility_domain '2222'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option disassoc_low_ack '0'

```

## Workaround
Activating **802.11w Management Frame Protection** as "**Optional**" and **Enable key reinstallation (KRACK) countermeasures** seems to keep speed but reconnections still occurs.

```
	option ieee80211w '1'
	option wpa_disable_eapol_key_retries '1'
```

[image]

## Failure Logs

AX Client
```
Fri Apr 22 20:01:27 2022 daemon.info hostapd: wlan1: STA yy:yy:yy:yy:yy:yy IEEE 802.11: authenticated
Fri Apr 22 20:01:27 2022 daemon.info hostapd: wlan1: STA yy:yy:yy:yy:yy:yy IEEE 802.11: associated (aid 2)
Fri Apr 22 20:01:27 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED yy:yy:yy:yy:yy:yy
Fri Apr 22 20:01:27 2022 daemon.info hostapd: wlan1: STA yy:yy:yy:yy:yy:yy WPA: pairwise key handshake completed (RSN)
Fri Apr 22 20:01:27 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED yy:yy:yy:yy:yy:yy
Fri Apr 22 20:02:20 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED yy:yy:yy:yy:yy:yy  <<<< HERE the speed drop
Fri Apr 22 20:02:20 2022 daemon.err hostapd: nl80211: kernel reports: key addition failed
Fri Apr 22 20:02:20 2022 daemon.info hostapd: wlan1: STA yy:yy:yy:yy:yy:yy IEEE 802.11: associated (aid 2)
Fri Apr 22 20:02:20 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED yy:yy:yy:yy:yy:yy
```
AC client
```
Fri Apr 22 19:27:10 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Fri Apr 22 19:27:10 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 10)
Fri Apr 22 19:27:10 2022 daemon.notice hostapd: wlan1: Prune association for xx:xx:xx:xx:xx:xx
Fri Apr 22 19:27:10 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx <<<< HERE the speed drop
Fri Apr 22 19:27:12 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
Fri Apr 22 19:27:12 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Fri Apr 22 19:27:12 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Fri Apr 22 19:27:13 2022 authpriv.info dropbear[8717]: Child connection from 192.168.1.200:56775
Fri Apr 22 19:27:21 2022 authpriv.notice dropbear[8717]: Password auth succeeded for 'root' from 192.168.1.200:56775
Fri Apr 22 19:27:40 2022 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

oldium, 2022-04-24:
If you are seeing inactivity deatuthentications, you can try `option skip_inactivity_poll '1'` in your `wifi-iface` config section.

ondoteam, 2022-04-24:
> If you are seeing inactivity deatuthentications, you can try `option skip_inactivity_poll '1'` in your `wifi-iface` config section.

Thank you! Not much testing, but no more disconnections for a while. The only odd behaviour with workaround:

```
option ieee80211w '1'
option wpa_disable_eapol_key_retries '1'
option skip_inactivity_poll '1' 
```

It is that sometimes, while roaming, the speed stucks on 100 Mbps with a 1200 / 1200 connection on 5 Ghz. 

Do you have and idea of what's happening with this router? Cons of using those options?

oldium, 2022-04-24:
Inactivity poll is a feature to disconnect devices that are out of range. Maybe your client's wifi has some low-power mode feature and switches off for a while, so it does not respond to polls and gets disconnected. Maybe check the settings of the low-power mode on your client - if it has anything to do with the slowdown?

I disabled the inactivity poll because of WoWLAN (wake-over-WiFi) not working on my notebook - when it was sleeping while being connected to WiFi, it was able to do the GTK rekeying, but it did not respond to the poll, so it got disconnected. I hoped that it helps you and I was correct 😁

ondoteam, 2022-04-24:
The main client for testing is an OPPO Find X3 Pro and it is awake during the tests (screen on and me hitting the refresh button of fast.com). The speed drop is happening within a time windows of 30 / 120 seconds after WiFi connection and only in my two TOTOLINK X5000R. Same client do not have this problem on four MAP-AC2200 and one GL-B1300. 

One more time, thank you very much for your time!

oldium, 2022-04-25:
You are welcome, but I ran out of advices - I do not know what could be the reason for the slowdown after reconnection.

parthkarnawat, 2023-01-12:
@ondoteam  Did you manage to make it work for the ax client? I am also getting the "key installation failed" error and it would be nice to get rid of it, it happens when I enable 802.11w