openwrt/openwrt #9770: kmod-mt76x2u snapshot package not support netgear A6210 USB wifi

youkechat, 2022-04-24:
detail was here, https://forum.openwrt.org/t/openwrt-not-support-netgear-a6210/126000/7, but I think this was driver bug or compatible problem, So I made a issue here

openwrt version: OpenWrt SNAPSHOT r19519-39799974a3 / LuCI Master git-22.089.43958-7110635
model: friendlyelec nanopi r4s
architecture: armv8 processor rev4
kernel version:5.10.112-1-d6ca1fc39c57d2c317b777327099b930
kmod-mt76x2u version: 5.10.112+2022-04-23-a666d563-4

Boos4721, 2022-04-25:
[image]

youkechat, 2022-04-28:
after install wpad-basic-wolfssl, A6210 ap mode enabled

closed 2022-04-28 by youkechat